project-chip/connectedhomeip #28183: [ReadHandler] Allow custom ReportScheduler Injection

lpbeliveau-silabs, 2023-07-21:


bzbarsky-apple, 2023-07-21:
> We could allow users to define their own scheduler by allowing injection from the app layer

Uh... that is in fact what you have currently implemented, no?  I am very unclear as to what this issue is about @lpbeliveau-silabs .

lpbeliveau-silabs, 2023-08-03:
While it was the case at some point during the merge of integration, it was removed on a request in the integration PR: https://github.com/project-chip/connectedhomeip/pull/28104#discussion_r1269825651

We currently offer the option between the Synchronized Scheduler and the default one from config, however it is not in _ServerInitParams_ anymore.

A user would need to edit Server.cpp/.h to be able to change the scheduling logic.

closed 2023-08-04 by mergify[bot]