project-chip/connectedhomeip #28192: [Feature] force tcp connections in Chip-Tool controller

Apollon77, 2023-07-22:
### Feature description

The matter specs also define TCP additionally to UDP. In fact no current controller (incl Chip tool) use TCP irrelevant if the flag is set in the announcement. 
I also got into that cert tests do not really include TCP.

But still if I like to implement it it would be cool to be able to force Chip-Tool as controller to use tcp over UDP if supported. 

### Platform

all

bzbarsky-apple, 2023-07-22:
> The matter specs also define TCP additionally to UDP

Note that this part is provisional.  But yes, it would be good to make it actually work, have test plans, etc.

> if the flag is set in the announcement.

That's one of the things that needs to change: currently nodes set that flag without actually supporting TCP (known issue; see https://github.com/project-chip/connectedhomeip/issues/23028.

So the spec needs changes to define a different "yes, really, I do support TCP" flag.  Then https://github.com/project-chip/connectedhomeip/issues/28141 needs to be fixed.  Then the setup described above might work.

bzbarsky-apple, 2023-07-26:
> then the 1.1 standard misses to have this listed in chapter 2.12 "Provisional items", or?!

Yes, it should be listed there and isn't.

Apollon77, 2024-06-19:
TCP landed in 1.3 basically

closed 2024-06-19 by Apollon77