project-chip/connectedhomeip #28216: [CERT-TEST-FAILURE] - TC-LWM- 2.1- test failed

Ashwinigrl, 2023-07-24:
### Feature Area

Area:Dishwashers

### Test Case

TC-LWM- 2.1

### Reproduction steps

**Summary**
For Generic failure code, It implies that switching to the mode indicated by the NewMode field is not allowed or not possible for some unspecified reason. If we try switching the modes within any supported mode it end with success

Not able to verify generic failure response in RVCCLEANM like we have a scenario in RVCRUNM " with Status:2 and statusText: Change to the mapping mode is only allowed from idle
**Test Step Screen shot :**
[image]

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

c1ec2d777456924dcaa59b53351b00d73caf378f

### Platform

raspi

rbultman, 2023-07-27:
Should be fixed by https://github.com/CHIP-Specifications/chip-test-plans/pull/3164

Ashwinigrl, 2023-08-08:
As the step can be skipped using PICS , Hence closing this issue

closed 2023-08-08 by Ashwinigrl