project-chip/connectedhomeip #28249: [Android][Build] Android studio build too slow

Allen-Guof, 2023-07-25:
### Build issue(s)

I want to build source code with Android Studio. And set **_matterSdkSourceBuild_**=true. 

However, Android Studio often keeps showing **"updating symbols**" all the time. After a long time, Android Studio is very likely to freeze. Is there any way to improve this? I'm using a 2018 Mac mini, 3.2 GHz 6-Core Intel Core i7, 16 GB 2667 MHz DDR4.
[image]

### Platform

android

Allen-Guof, 2023-10-17:
@yunhanw-google ,Hi, I'd like to inquire about the minimum hardware requirements for this project. I'm considering whether my computer's performance is sufficient to run this project. If you could provide information regarding the minimum requirements for CPU, memory... I would greatly appreciate it.

CaoJieNan, 2024-02-26:
I have also encountered the same problem recently. Is there a way to solve it.