project-chip/connectedhomeip #28253: Examples/chef not installing properly

sahityadubey, 2023-07-25:
### Reproduction steps

When I am running the below command, installation stop here with error "ninja: Entering directory `out'"
How can we fix this error?

`$ ./chef.py -zbr -d rootnode_dimmablelight_bCwGYSDpoe -t linux
`

>**Logs:**

 Downloading packages into local source directory:

  Setting up CIPD package manager...done (15.3s)
  Setting up Project actions........skipped (0.1s)
  Setting up Python environment.....[|]
[notice] A new release of pip is available: 23.2 -> 23.2.1
[notice] To update, run: pip install --upgrade pip
done (1m17.4s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20230725 10:20:43 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Installing pip requirements for all...

[notice] A new release of pip is available: 23.2 -> 23.2.1
[notice] To update, run: pip install --upgrade pip
Running ZAP script to generate artifacts
Searching for zcl file from /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/devices/rootnode_dimmablelight_bCwGYSDpoe.zap
🔧 Using temporary state directory: /var/folders/q9/684jnz8j00b9187_lwt93f7r0000gp/T/zap.8Fo7we
🤖 ZAP generation started: 
    🔍 input files: /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/devices/rootnode_dimmablelight_bCwGYSDpoe.zap
    🔍 output pattern: /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/out/rootnode_dimmablelight_bCwGYSDpoe/zap-generated/
    🔍 using templates: /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/src/app/zap-templates/matter-idl-server.json
    🔍 using zcl data: /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/src/app/zap-templates/zcl/zcl.json
    🔍 zap version: ver. 2023.7.20, featureLevel 97, commit: b57cef1fc261cedcbee418566a3d1e83b2d200d6 from 2023-07-20T18:42:00.000Z, mode: binary, exe: zap-cli
🕐 Setup time: 820ms 
👉 using input file: /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/devices/rootnode_dimmablelight_bCwGYSDpoe.zap
👉 using output destination: /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/out/rootnode_dimmablelight_bCwGYSDpoe/zap-generated/
🕐 File loading time: 87ms
🤖 Generating files:
    ✍  /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/out/rootnode_dimmablelight_bCwGYSDpoe/zap-generated/Clusters.matter
🕐 Generation time: 356ms 
Files generated in: /Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/out/rootnode_dimmablelight_bCwGYSDpoe/zap-generated/
RPC PW enabled
RPC PW on linux not supported
Software Version String: ""
Product ID 0x8000 / Vendor ID 0xFFF1
Building...
[... 30 lines cut ...]
                                      ^
1 error generated.
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/./chef.py", line 934, in <module>
    sys.exit(main())
  File "/Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/./chef.py", line 802, in main
    shell.run_cmd("ninja -C out")
  File "/Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/chef/stateful_shell.py", line 139, in run_cmd
    raise RuntimeError(
RuntimeError: Error. Nonzero return code.
Returncode: 1
Cmd: ninja -C out`

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

master

### Platform

android

sahityadubey, 2023-07-26:
@andy31415 @andreilitvin 
Do you have any update here?

andy31415, 2023-07-26:
Tried in `ghcr.io/project-chip/chip-build-vscode:1` and it seems to work fine:

```
> ./chef.py -zbr -d rootnode_dimmablelight_bCwGYSDpoe -t linux
....
[1176/1176] stamp obj/default.stamp
Done
```

I am unclear where you compile. Your `Platform` says "android" yet your paths are `/Users/sdubey/Sahitya/Projects/ChipTool/connectedhomeip/examples/` which look like a mac.

The error seems environment-specific, guessing a different compiler version than what chef on linux is using (maybe clang instead of gcc, which may be more strict).

sahityadubey, 2023-07-26:
I am trying to run on mac

andy31415, 2023-07-26:
I don't think anyone tried to use mac before (otherwise they would have found the same error). I am guessing you get mac which uses clang which is more strict on errors. I do not have a mac to try things out, but guessing some `Wsign-conversion` exclusion is needed somewhere or a patch to pigweed is required.

andy31415, 2023-07-26:
You can probably adjust the flags in `build/config/compiler/BUILD.gn` however I would not do that by default on the entire project since it seems we want this king of warnings.

andy31415, 2023-07-26:
It may also be that `is_clang` is not correctly set since on mac even `gcc` is clang.

andy31415, 2023-07-26:
If you run this in the linux container it will probably just work, but you will get a linux app instead of a mac app.

sahityadubey, 2023-07-27:
On Linux container, I am getting two separate error.

1. Virtual device container not working 
**Log-** 
```
sh-5.0$  cd ~/matter-virtual-device-gui/
sh-5.0$ electron main.js --s=localhost:33000 --no-sandbox
qemu: uncaught target signal 5 (Trace/breakpoint trap) - core dumped
qemu: uncaught target signal 5 (Trace/breakpoint trap) - core dumped
Connecting to Socket Target: localhost:33000
[443:0727/095542.589014:ERROR:ozone_platform_x11.cc(240)] Missing X server or $DISPLAY
[443:0727/095542.591546:ERROR:env.cc(255)] The platform failed to initialize.  Exiting.
```

2. After successfully running below command I am not able to commission the device in my mobile phone GHA.
Is there anything I am missing ?

**Command-**
`./linux/out/rootnode_dimmablelight_bCwGYSDpoe`

**Log-**
sh-5.0$ ./linux/out/rootnode_dimmablelight_bCwGYSDpoe
DBG  [1690445022.366123][357:357] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
DBG  [1690445022.418726][357:357] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
ERR  [1690445022.420842][357:357] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
DBG  [1690445022.426871][357:357] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
DBG  [1690445022.432072][357:357] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
DBG  [1690445022.432669][357:357] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
INF  [1690445022.437257][357:357] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-VisTje)
ERR  [1690445022.438500][357:357] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
INF  [1690445022.438693][357:357] CHIP:DL: NVS set: chip-counters/reboot-count = 3 (0x3)
ERR  [1690445022.441402][357:357] CHIP:DL: Failed to get Ethernet interface
ERR  [1690445022.441733][357:357] CHIP:DL: Failed to reset Ethernet statistic counts
ERR  [1690445022.442027][357:357] CHIP:DL: Failed to get WiFi interface
ERR  [1690445022.442216][357:357] CHIP:DL: Failed to reset WiFi statistic counts
ERR  [1690445022.442821][357:357] CHIP:SPT: *** WARNING: Using temporary passcode 20202021 due to no neither --passcode or --spake2p-ver
ERR  [1690445022.443275][357:357] CHIP:SPT: *** WARNING: Using temporary test discriminator 3840 due to --discriminator not given on com
ERR  [1690445022.443431][357:357] CHIP:SPT: PASE PBKDF iterations set to 1000
INF  [1690445022.444003][357:357] CHIP:SPT: LinuxCommissionableDataProvider didn't get a PASE salt, generating one.
INF  [1690445022.491203][357:357] CHIP:DL: Device Configuration:
INF  [1690445022.491501][357:357] CHIP:DL:   Serial Number: TEST_SN
INF  [1690445022.491631][357:357] CHIP:DL:   Vendor Id: 65521 (0xFFF1)
INF  [1690445022.491742][357:357] CHIP:DL:   Product Id: 32768 (0x8000)
INF  [1690445022.491916][357:357] CHIP:DL:   Hardware Version: 0
INF  [1690445022.492018][357:357] CHIP:DL:   Setup Pin Code (0 for UNKNOWN/ERROR): 20202021
INF  [1690445022.492117][357:357] CHIP:DL:   Setup Discriminator (0xFFFF for UNKNOWN/ERROR): 3840 (0xF00)
INF  [1690445022.492302][357:357] CHIP:DL:   Manufacturing Date: (not set)
INF  [1690445022.492408][357:357] CHIP:DL:   Device Type: 65535 (0xFFFF)
INF  [1690445022.492487][357:357] CHIP:-: ==== Onboarding payload for Standard Commissioning Flow ====
INF  [1690445022.493705][357:357] CHIP:SVR: SetupQRCode: [MT:Y.K90AFN00KA0648G00]
INF  [1690445022.494093][357:357] CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
INF  [1690445022.494178][357:357] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3AY.K90AFN00KA0648G00
INF  [1690445022.494919][357:357] CHIP:SVR: Manual pairing code: [34970112332]
INF  [1690445022.495474][357:357] CHIP:-: PW_RPC initialized.
INF  Starting pw_rpc server on port 33000
> INF  [1690445022.501459][357:357] CHIP:SVR: Server initializing...
DBG  [1690445022.501787][357:357] CHIP:FP: Initializing FabricTable from persistent storage
INF  [1690445022.504827][357:357] CHIP:TS: Last Known Good Time: 2023-07-26T17:37:30
INF  [1690445022.505306][357:357] CHIP:DMG: AccessControl: initializing
INF  [1690445022.505380][357:357] CHIP:DMG: Examples::AccessControlDelegate::Init
INF  [1690445022.505597][357:357] CHIP:DMG: AccessControl: setting
INF  [1690445022.505867][357:357] CHIP:DMG: DefaultAclStorage: initializing
INF  [1690445022.505951][357:357] CHIP:DMG: DefaultAclStorage: 0 entries loaded
DBG  [1690445022.506593][357:357] CHIP:IN: UDP::Init bind&listen port=5540
DBG  [1690445022.507292][357:357] CHIP:IN: UDP::Init bound to port=5540
DBG  [1690445022.507423][357:357] CHIP:IN: TransportMgr initialized
INF  [1690445022.508553][357:357] CHIP:DL: writing settings to file (/tmp/chip_kvs-GYrpVb)
ERR  [1690445022.508816][357:357] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
INF  [1690445022.509073][357:357] CHIP:DL: writing settings to file (/tmp/chip_kvs-9rfcHc)
ERR  [1690445022.509202][357:357] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
[... 51 lines cut ...]
INF  [1690445022.549251][357:357] CHIP:DIS: mDNS service published: _matterc._udp
INF  [1690445022.549738][357:357] CHIP:IN: CASE Server enabling CASE session setups
DBG  [1690445022.550082][357:357] CHIP:IN: SecureSession[0x40002902f0]: Allocated Type:2 LSID:47119
DBG  [1690445022.550462][357:357] CHIP:SC: Allocated SecureSession (0x40002902f0) - waiting for Sigma1 msg
[... cut ...]
INF  [1690445022.552059][357:357] CHIP:SVR: https://project-chip.github.io/connectedhomeip/qrcode.html?data=MT%3AY.K90AFN00KA0648G00
INF  [1690445022.552126][357:357] CHIP:SVR: Manual pairing code: [34970112332]
INF  [1690445922.659101][357:357] CHIP:SVR: Closing pairing window
DBG  [1690445922.662939][357:357] CHIP:IN: SecureSession[0x4000290ef0]: Released - Type:1 LSID:47118
INF  [1690445922.664552][357:357] CHIP:DIS: Updating services using commissioning mode 0
INF  [1690445922.668680][357:357] CHIP:DIS: CHIP minimal mDNS started advertising.

petetnt, 2023-08-30:
FWIW, I got the example to build on MacOS by removing `-Wconversion` from `build/config/compiler/BUILD.gn` and `src/platform/Darwin/Build.gn` 

```diff
diff --git a/build/config/compiler/BUILD.gn b/build/config/compiler/BUILD.gn
index 31f396a457..3e636bd0f6 100644
--- a/build/config/compiler/BUILD.gn
+++ b/build/config/compiler/BUILD.gn
@@ -250,7 +250,7 @@ config("strict_warnings") {
   }
 
   if (current_os == "mac" || current_os == "ios") {
-    cflags += [ "-Wconversion" ]
+    # cflags += [ "-Wconversion" ]
   }
 
   if (matter_enable_java_compilation) {
diff --git a/src/platform/Darwin/BUILD.gn b/src/platform/Darwin/BUILD.gn
index 1769d4a29f..22e2dc7374 100644
--- a/src/platform/Darwin/BUILD.gn
+++ b/src/platform/Darwin/BUILD.gn
@@ -38,7 +38,7 @@ config("darwin_config") {
 
   cflags = [
     "-fobjc-arc",
-    "-Wconversion",
+    # "-Wconversion",
   ]
 }
```

As far as I understand the root issue is that `pigweed` does not build with `-Wconversion` enabled (https://issues.pigweed.dev/issues/259746255)  and it's explicitly enabled on `mac` builds.

petetnt, 2023-08-30:
That however ends up with the software dying when trying to read the attributes:

Device logs:

```
Connected to rpc.
[RPC] Send rpc 'chip.rpc.Device.GetDeviceInfo'.
Starting chip.rpc.Device.GetDeviceInfo on channel 1
Received data which length is 173
[RPC] Send rpc 'chip.rpc.Descriptor.DeviceTypeList'.
Starting chip.rpc.Descriptor.DeviceTypeList on channel 1
Received data which length is 49
[RPC] Send rpc 'chip.rpc.Descriptor.PartsList'.
Starting chip.rpc.Descriptor.PartsList on channel 1
Received data which length is 48
[RPC] Send rpc 'chip.rpc.Descriptor.ServerList'.
Starting chip.rpc.Descriptor.ServerList on channel 1
Received data which length is 158
[RPC] Send rpc 'chip.rpc.Attributes.Read'.
Starting chip.rpc.Attributes.Read on channel 1
Connection closed
```

Chef logs:

```
[1693426243808] [20996:854395] [DIS] Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
[1693426243808] [20996:854395] [DIS] Registering service 440B92AF5F070CAA on host 000000000000.local. with port 5540 and type: _matterc._udp,_V65521,_S15,_L3840,_CM on interface id: 0
[1693426243810] [20996:854395] [IM] No subscriptions to resume
[1693426244533] [20996:854395] [DIS] Mdns: OnRegister name: 440B92AF5F070CAA, type: _matterc._udp., domain: local., flags: 2
[1693426244534] [20996:854395] [DIS] mDNS service published: _matterc._udp; instance name: 440B92AF5F070CAA
[1693426244542] [20996:854395] [DIS] Status: Satisfied
[1693426244542] [20996:854395] [DIS]    lo0 (1)
[1693426244543] [20996:854395] [DIS]            * ipv6: ::1
[1693426244543] [20996:854395] [DIS]            * ipv6: fe80::1
[1693426244544] [20996:854395] [DIS]    en0 (12 / WiFi)
[1693426244544] [20996:854395] [DIS]            * ipv6: fe80::c06:b969:469d:e586
[1693426244545] [20996:854395] [DIS] Mdns: OnRegisterRecord flags: 1
[1693426244545] [20996:854395] [DIS] Mdns: OnRegisterRecord flags: 1
[1693426244545] [20996:854395] [DIS] Mdns: OnRegisterRecord flags: 0
[1693426250702] [20996:854397] [DMG] Reading attribute: Cluster=0x0000_0008 Endpoint=1 AttributeId=0x0000_0002 (expanded=0)
[1693426250702] [20996:854397] [DL] Chip stack locking error at 'src/app/util/attribute-storage.cpp:572'. Code is unsafe/racy
[1693426250702] [20996:854397] [-] chipDie chipDie chipDie
[1]    20996 abort      ./linux/out/rootnode_dimmablelight_bCwGYSDpoe
```

Possibly related to https://github.com/project-chip/connectedhomeip/issues/7557 ?

andy31415, 2023-08-30:
Mac does not support chip stack locking, so if chef tries to use that in any way it probably fails. Chef was never tested to run on mac.

petetnt, 2023-08-30:
I see, thanks for clarification

closed 2023-09-26 by andy31415