project-chip/connectedhomeip #28254: [Android][Build] error: no suitable method found for toString(Optional<byte[]>)

Allen-Guof, 2023-07-25:
### Build issue(s)

When there is a byte[] type in the Struct structure, the generated Android code cannot run. When executing the command **./scripts/build/build_examples.py --target android-arm64-chip-tool** build, there will be an error:
...
2023-07-25 14:40:51 INFO    python3 ../../build/chip/java/javac_runner.py --classdir obj/src/controller/java/classes --outfile gen/src/controller/java/java.classlist --build-config gen/src/controller/java/java.json -- -d obj/src/controller/java/classes @gen/src/controller/java/java.sources -Werror -Xlint:all -Xlint:deprecation -parameters -Xlint:-options -source 8 -target 8
2023-07-25 14:40:51 INFO    ../../src/controller/java/zap-generated/chip/devicecontroller/ChipStructs.java:2245: error: no suitable method found for toString(Optional<byte[]>)
2023-07-25 14:40:51 INFO          output.append(Arrays.toString(data));
2023-07-25 14:40:51 INFO                              ^
2023-07-25 14:40:51 INFO        method Arrays.toString(long[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to long[])
2023-07-25 14:40:51 INFO        method Arrays.toString(int[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to int[])
2023-07-25 14:40:51 INFO        method Arrays.toString(short[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to short[])
2023-07-25 14:40:51 INFO        method Arrays.toString(char[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to char[])
2023-07-25 14:40:51 INFO        method Arrays.toString(byte[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to byte[])
2023-07-25 14:40:51 INFO        method Arrays.toString(boolean[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to boolean[])
2023-07-25 14:40:51 INFO        method Arrays.toString(float[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to float[])
2023-07-25 14:40:51 INFO        method Arrays.toString(double[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to double[])
2023-07-25 14:40:51 INFO        method Arrays.toString(Object[]) is not applicable
2023-07-25 14:40:51 INFO          (argument mismatch; Optional<byte[]> cannot be converted to Object[])
2023-07-25 14:40:51 INFO    Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output

The generated struct code:
```
  public static class YeelightEnhancedColorEffectConfigClusterEffectStruct {
    ...
    public Optional<byte[]> data;
    ....

    @Override
    public String toString() {
      StringBuilder output = new StringBuilder();
      ...
      output.append("\tdata: ");
      output.append(Arrays.toString(data));
      ....
      return output.toString();
    }
  }
```

### Platform

android

yunhanw-google, 2023-07-27:
We have decoupled ChipStructs.java from android chip-tool, how do you compile the android chip-tool?

yunhanw-google, 2023-07-27:
@Allen-Guof @joonhaengHeo

Allen-Guof, 2023-07-27:
@yunhanw-google Sorry, I didn't understand your meaning.
1. I defined a custom Cluster class, and inside it contains an EffectStruct member.
2. source scripts/activate.sh
3. ./scripts/tools/zap_regen_all.py
4. ./scripts/build/build_examples.py --target android-arm64-chip-tool build

joonhaengHeo, 2023-07-28:
@yunhanw-google 
Not yet removed from Android Chip-tool. Optionally removed only for java-matter-controller.
Because the ClusterReadMapping and ClusterWriteMapping class are still being used in android chip-tool, this part of the work has not been completed yet.

@Allen-Guof 
When I checked ChipStructs-java.zapt, the optional parameters are not considered. If you want to generate `Optional<byte[]>` , you may need to add consider Optional to this file. And, as yunhanw said above, the Generated File is being deleted. If you want to add a custom cluster, it is recommended to use the read / writer / invoke API implemented in ChipDeviceController.

Allen-Guof, 2023-07-28:
Ok, I got it. Thank your support. @joonhaengHeo

closed 2023-07-28 by Allen-Guof