project-chip/connectedhomeip #28255: [1.1] - Commissioning window not opening in the Node using the command.

smishra-volansys, 2023-07-25:
### Reproduction steps

We are using the Chip-tool command to open the Commissioning window on an end device Node to pair it with another fabric.
I am performing the following steps:

1) Pairing the End device with one Fabric using the chip-tool pairing command. The device is commissioned successfully. 

2) Trying to open the commissioning window using the following command:

**chip-tool pairing open-commissioning-window <node_id> <option> <window_timeout> <iteration> <discriminator>**

3)After the Command is fired,  I get the new Manual Code and the QR Payload in the logs, as shown: 

[image]

4) Yet, While I try to pair the Node into a new Fabric, the device is not listed in the scanning, I am using the NRF connect application to check if the device is advertising or not. It is not shown there as well. 

5) Also, to pair it using the chip tool I use the following command syntax from the New chip-tool instance of  the new fabric with values replaced as asked:
**chip-tool pairing code <payload> <node_id> --commissioner-name <commissioner_name>**

### Bug prevalence

Frequent

### GitHub hash of the SDK that was being used

v1.1

### Platform

other

### Type

Test Improvement, Platform Issue, Core SDK Performance Improvement

smishra-volansys, 2023-07-26:
[Uploading Chiptool_Commissioningwindow.log…]()
@bzbarsky-apple PFA the Logs from the Chip-tool.

the device is an Eve Smart Plug, Unable to get the logs for the same.

smishra-volansys, 2023-07-26:
[Chiptool_Commissioningwindow.log](https://github.com/project-chip/connectedhomeip/files/12172051/Chiptool_Commissioningwindow.log)
PFA the Logs, @bzbarsky-apple

bzbarsky-apple, 2023-07-26:
OK.  So after running that, what does the log from chip-tool look like if you do:
```
chip-tool pairing code 25 MT:4K9U4GBQ15NNPJ3GY10 --identity beta
```

Alternately, what does a browse for `_matterc._udp` show?

But fundamentally, if the device responds with success to "open commissioning window" but is not advertising commissionable, that's a bug in the device and should be reported to Eve in this case.

smishra-volansys, 2023-07-27:
Hi @bzbarsky-apple, 
For the Validation with out matter controller I tried running the Lighting-app on one device and commissioned it by running Chip tool on another controller. Once the successful pairing was done, I opened the commissioning window for the Light app and paired it with a 3rd controller. Both of there controller are able to send on/off cluster commands and the response is as expected.

Hence for Eve, I'll get in touch with them and get the updates on the same if any. 

Thankyou !

closed 2023-08-29 by smishra-volansys