project-chip/connectedhomeip #28256: Windowcovering works fine with chip-tool but show "no response" on Apple HomePod

nmtoan91, 2023-07-25:
### Reproduction steps

Hi,
My bridged app works fine with chip-tool with several kinds of devices such as light, onoff lights switches, window coverings.

But the window-covering device show "no response" on my Apple HomePod.
Here is the log of the bridged-app when I try to tap the windowcovering icon on my Apple home app.

```cpp
[1690269074.430322][14397:14397] CHIP:DMG: Refresh subscribe timer sync after 2 seconds
[1690269074.430443][14397:14397] CHIP:DMG: Building Reports for ReadHandler with LastReportGeneration = 29 DirtyGeneration = 0
[1690269074.431536][14397:14397] CHIP:DMG: Fetched 0 events
[1690269074.431568][14397:14397] CHIP:DMG: <RE> Sending report (payload has 11 bytes)...
[1690269074.431781][14397:14397] CHIP:EM: <<< [E:8573i S:55896 M:266995597] (S) Msg TX to 5:000000008467109D [A32E] --- Type 0001:05 (IM:ReportData)
[1690269074.431844][14397:14397] CHIP:IN: (S) Sending msg 266995597 on secure session with LSID: 55896
[1690269074.432014][14397:14397] CHIP:DMG: IM RH moving to [AwaitingReportResponse]
[1690269074.432044][14397:14397] CHIP:DMG: Refresh Subscribe Sync Timer with min 0 seconds and max 2 seconds
[1690269074.432069][14397:14397] CHIP:DMG: <RE> ReportsInFlight = 1 with readHandler 0, RE has no more messages
[1690269074.432090][14397:14397] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1690269074.432143][14397:14397] CHIP:DMG: Unblock report hold after min 0 seconds
[1690269074.555071][14397:14397] CHIP:EM: >>> [E:8573i S:55896 M:56270493 (Ack:266995597)] (S) Msg RX from 5:000000008467109D [A32E] --- Type 0001:01 (IM:StatusResponse)
[1690269074.555152][14397:14397] CHIP:EM: Found matching exchange: 8573i, Delegate: 0x557f91ae6b38
[1690269074.555203][14397:14397] CHIP:EM: Rxd Ack; Removing MessageCounter:266995597 from Retrans Table on exchange 8573i
[1690269074.555281][14397:14397] CHIP:DMG: StatusResponseMessage =
[1690269074.555315][14397:14397] CHIP:DMG: {
[1690269074.555350][14397:14397] CHIP:DMG: 	Status = 0x00 (SUCCESS),
[1690269074.555380][14397:14397] CHIP:DMG: 	InteractionModelRevision = 1
[1690269074.555407][14397:14397] CHIP:DMG: }
[1690269074.555437][14397:14397] CHIP:IM: Received status response, status is 0x00 (SUCCESS)
[1690269074.555475][14397:14397] CHIP:DMG: <RE> OnReportConfirm: NumReports = 0
[1690269074.555500][14397:14397] CHIP:DMG: IM RH moving to [GeneratingReports]
[1690269074.555601][14397:14397] CHIP:EM: <<< [E:8573i S:55896 M:266995598 (Ack:56270493)] (S) Msg TX to 5:000000008467109D [A32E] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1690269074.555652][14397:14397] CHIP:IN: (S) Sending msg 266995598 on secure session with LSID: 55896
[1690269074.555829][14397:14397] CHIP:EM: Flushed pending ack for MessageCounter:56270493 on exchange 8573i
```

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

latest

### Platform

darwin

plan44, 2023-08-06:
I am also working on a bridge app, and have added Window Covering last week. It works fine with Apple Home, so I'm wondering what might be the difference.

How did you implement the delegate for window-covering?

I found that giving feedback of the actual jalousie/roller position is important, the cluster derives "complete" status from target==current, and only if current is updated last.

My bridge is GLPv3 opensource, you may have a [look](/plan44/p44mbrd), the action for window covering is `P44_WindowCoveringImpl::startMovement` [in this file](https://github.com/plan44/p44mbrd/blob/main/src/adapters/p44/p44devices.cpp#L587).

BTW, as a bridge developer, maybe you'd be interested in #28372 and maybe comment on it?

bzbarsky-apple, 2023-09-01:
@nmtoan91 For issues with Apple Home, please file a feedback assistant ticket and report the ticket ID here.  That will provide the Home-side logs, etc.

That said, I believe there are some restrictions on window covering in Apple Home so far; in particular it may not support position-unaware window coverings yet.  

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.