project-chip/connectedhomeip #28274: [BUG] Failed to post event to CHIP Platform event queue error when trying to commission M5/ESP32 all clusters or lighting app

kean-apple, 2023-07-25:
### Reproduction steps

Observed Failed to post event to CHIP Platform event queue error when trying to commission M5/ESP32 all clusters or lighting app

SHA: 3efdd368f71ebad966f158f38384c897c53db5b4
1. Build all clusters or lighting app for M5 board
2. Commission M5 board using Apple Home app on iPhone
3. After receiving Sigma3 message when setting up CASE session, observe Failed to post event to CHIP Platform event queue errors

Note: 
- this often causes commissioning to fail. Also, seen in other non commissioning flows as well
- I've seen this happen occasionally on older SHAs but latest SHA seems to happen nearly 100%

Log snippet

I (47202) chip[SC]: Received Sigma3 msg
E (47212) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
I (47372) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
E (47462) chip[DL]: Failed to post event to CHIP Platform event queue
E (47462) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (47552) chip[DL]: Failed to post event to CHIP Platform event queue
E (47552) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (47572) chip[DL]: Failed to post event to CHIP Platform event queue
E (47572) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
E (47742) chip[CR]: mbedTLS error: BIGNUM - Memory allocation failed
E (47752) chip[DL]: Failed to post event to CHIP Platform event queue
E (47752) chip[DL]: Failed to schedule work: 1000000
E (47752) chip[SC]: Failed to Schedule the AfterWorkCallback 

### Bug prevalence

Nearly always when commissioning

### GitHub hash of the SDK that was being used

3efdd368f71ebad966f158f38384c897c53db5b4

### Platform

esp32

### Platform Version(s)

v4.4.3

### Anything else?

[out-memory-lighting-app-M5.txt](https://github.com/project-chip/connectedhomeip/files/12165265/out-memory-lighting-app-M5.txt)
[out-memory-error-all-cluster-app-M5.txt](https://github.com/project-chip/connectedhomeip/files/12165266/out-memory-error-all-cluster-app-M5.txt)

bzbarsky-apple, 2023-07-25:
@dhrishi @shubhamdp

dhrishi, 2024-10-25:
The lighting app configuration was optimized and the queue size was changed to 40. This should not happen now. Please re-open  if you still encounter this.

closed 2024-10-25 by dhrishi