project-chip/connectedhomeip #28277: [ReadHandler] Reduce RAM usage

lpbeliveau-silabs, 2023-07-25:


lpbeliveau-silabs, 2023-08-03:
Even wort, timestamps are uint64_t, so each one is taking 8 bytes, so 24bytes worth of timestamp.

There is some logic that validates if max is greater or smaller than min and vice versa so storing the 16-bit offset might be a bit complicated, that could be tackled in either https://github.com/project-chip/connectedhomeip/issues/27996 or https://github.com/project-chip/connectedhomeip/issues/27997 though.

For the sync timestamp, I am hoping to trade the sync timestamp for a uint8_t flag that would hold mEngineRunScheduled  flag and a new Sync flag. That would require some rework on the sync logic, which is already needed in https://github.com/project-chip/connectedhomeip/issues/28083.

Regarding the size, given that the pool size is dictated by the maximum number of read handler:
`CHIP_IM_MAX_NUM_READS + CHIP_IM_MAX_NUM_SUBSCRIPTIONS` , which defaults to 64, reducing the size of nodes has a huge impact.

bzbarsky-apple, 2023-08-05:
>  timestamps are uint64_t

Ah, right, we store TimeStamp, not Timeout.

> which defaults to 64

Wait, why does that default to 64?  I would expect it, on any sort of constrained platform, to be 5 reads and 15 subscriptions, total of 20.  Still pretty large if we have 24 bytes just for the timestamps...

lpbeliveau-silabs, 2023-08-17:
Good news is with https://github.com/project-chip/connectedhomeip/pull/28733, we will only have 2 timestamps.

> Wait, why does that default to 64? 

It boils down to CHIP_CONFIG_MAX_FABRICS + CHIP_CONFIG_MAX_FABRICS * 3, which is 16 if not defined.

It seems a lot of platform do define it to 5 but this is on a per platform basis.

bzbarsky-apple, 2023-08-28:
> which is 16 + 16 * 3 from our default define in CHIPConfig.h

None of the constrained devices use that.  What's the actual RAM use on an actual device, not linux/mac, where no one is going to care about a few KB of RAM?