project-chip/connectedhomeip #28281: [Feature] Allow the device name to be stored during runtime

ajoletta, 2023-07-25:
### Feature description

...
Allow for the device name to be stored during runtime

Problem:
Currently the device name is hard coded and does not allow for the device name to be updated during runtime.

Application:
For the example tv-casting-app which is run on a phone to have the ability to read the phone's device name to then display it on the example tv-app which will allow users to recognize the specific phone making a request to commission. 

### Platform

all

bzbarsky-apple, 2023-07-26:
> Currently the device name is hard coded

Where?  That is, which "device name" are you looking at here?

bzbarsky-apple, 2023-07-26:
@ajoletta Please see above.