project-chip/connectedhomeip #28283: [BUG] light_switch can not control light_bulb after remove OTBR from Thread network

lancersky2017, 2023-07-26:
### Reproduction steps

Hi,
I set up a test environment according to the following topology diagram.
[image]
I used the following command to set the rules for the [light_switch](https://github.com/nrfconnect/sdk-nrf/tree/v2.3.0/samples/matter/light_switch) to control the [light_bulb](https://github.com/nrfconnect/sdk-nrf/tree/v2.3.0/samples/matter/light_bulb). It works just fine.
```
out/host/chip-tool accesscontrol write acl '[{"fabricIndex":1, "privilege":5, "authMode":2, "subjects":[112233, 1005], "targets":null}]' 1006 0

out/host/chip-tool binding write binding '[{"fabricIndex":1, "node":1006, "endpoint":1, "cluster":6}]' 1005 1
```
But when I remove the OTBR, they don't work anymore for a long time. And rebooting both devices didn't help either.
Is this result expected? 
```
I: Button has been pressed, keep in this state for at least 500 ms to change light sensitivity of binded lighting devices.
I: Notify Bounded Cluster | endpoint: 1 cluster: 6
D: 14158 [CTL]OperationalSessionSetup[1:00000000000003EE]: State change 1 --> 2
D: 14165 [CTL]OperationalSessionSetup[1:00000000000003EE]: State change 2 --> 1
E: 14172 [SVR]Failed to establish connection to node 0x00000000000003EE
```

### Bug prevalence

failed all the time

### GitHub hash of the SDK that was being used

nrf-v2.3.0

### Platform

nrf

bzbarsky-apple, 2023-07-26:
And in particular, this bit:
```
D: 14158 [CTL]OperationalSessionSetup[1:00000000000003EE]: State change 1 --> 2
D: 14165 [CTL]OperationalSessionSetup[1:00000000000003EE]: State change 2 --> 1
```

State 1 is NeedsAddress.  State 2 is ResolvingAddress.  So the DNS-SD lookup failed.

lancersky2017, 2023-07-27:
Hi @bzbarsky-apple,  I just power off the OTBR to verify if the switch can still control the light if the OTBR fails. So it seems that OTBR is essential for the proper functioning of smart home networks.

bzbarsky-apple, 2023-07-27:
@lancersky2017 An SRP server is essential for the proper functioning of Thread Matter devices.  The SRP server does not have to be a full border router, as I understand it, though commonly the SRP servers you have around will in fact be border routers.