project-chip/connectedhomeip #28287: The test plan has "SHALL NOT: bits that are not reflected in this YAML.  File a followup issue, please, add a comment in the YAML pointing to it.

abpoth, 2023-07-26:
Many .yaml test cases don't seem to account for the "SHALL NOT" include values when testing AttributeList.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/28023#discussion_r1274099041_

manjunath-grl, 2023-08-08:
@abpoth Since we have no support for checking the range of values of lists via yaml, we perform this verification step through a manual prompt step. We have opened a ticket while ago in SDK repo for the same https://github.com/project-chip/connectedhomeip/issues/26570
CC @raju-apple @cjandhyala

andy31415, 2023-08-29:
Considering this not a cert-blocker for now - we are working on better range verification (I believe we had a "manual step", working on automation). #28902 is the original PR regarding this.

closed 2023-09-01 by mergify[bot]