project-chip/connectedhomeip #28289: [CERT-TEST-FAILURE] - TC-RVCRUNM- 3.2- test failed

Saravana-kr22, 2023-07-26:
### Feature Area

Area:RobotVacuums

### Test Case

TC-RVCRUNM-3.2

### Reproduction steps

Command to run the TC_RVCRUNM_3_2.py 
`python3 TC_RVCRUNM_3_2.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --storage-path admin_storage.json --PICS ci-pics-values --int-arg PIXIT_ENDPOINT:1 PIXIT_MODEOK:0 PIXIT_MODEFAIL:4 --paa-trust-store-path /paa_roots`

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

raspi

### Anything else?

TC_RVCRUNM_3_2.py  is failed due to the null value of the start-up-mode attribute. In step 1 the start-up-mode value is null and in step 2 (before physically power cycle the dut) it current mode value is 0 . So the python script skipped the step 3 and 4 by assuming the null and 0 are different. But after the physically power cycle the dut the current mode value is still 0 so the python script is failed with the error `CurrentMode must match StartUpMode after a power cycle` 

[TC-RVCRUNM-3-2.txt](https://github.com/project-chip/connectedhomeip/files/12171363/TC-RVCRUNM-3-2.txt)

plauric, 2023-07-26:
We're fixing in the TP and in the test script:

https://github.com/CHIP-Specifications/chip-test-plans/pull/3166

https://github.com/project-chip/connectedhomeip/pull/28297

Saravana-kr22, 2023-08-10:
This issue has been resolved in https://github.com/project-chip/connectedhomeip/commit/013e3ce1e0f58c9a05addd60bef948835b10afb0.

closed 2023-08-10 by Saravana-kr22