project-chip/connectedhomeip #28317: [1.1] Android build fail use script and Android Studio

ngovancuong2735, 2023-07-27:
### Reproduction steps

I want to build chip-tool android app following instructions: https://github.com/ksperling-apple/connectedhomeip/blob/4d75d75d11b0ec2563031d8e583b8af3ee40d47f/docs/guides/android_building.md
I use NDK: 23.2.8568313
I setup the environment:
```
export ANDROID_HOME=~/Android/Sdk
export ANDROID_NDK_HOME=~/Android/Sdk/ndk/23.2.8568313
export JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64
```

- When I use script : `./scripts/build/build_examples.py --target android-arm64-chip-tool build`. I have the error:

```
2023-07-27 10:45:55 INFO    BUILD SUCCESSFUL in 6s
2023-07-27 10:45:55 INFO    4 actionable tasks: 1 executed, 3 up-to-date
2023-07-27 10:45:55 INFO    Command ['python3', 'third_party/android_deps/set_up_android_deps.py'] completed
2023-07-27 10:45:55 INFO    Setting up Java deps
2023-07-27 10:46:15 INFO    Command ['third_party/java_deps/set_up_java_deps.sh'] completed
2023-07-27 10:46:15 INFO    Building /home/cuong/MatterHub/connectedhomeip/out/android-arm64-chip-tool
2023-07-27 10:46:15 INFO    Building JNI android-arm64-chip-tool
2023-07-27 10:46:15 INFO    ninja: Entering directory `/home/cuong/MatterHub/connectedhomeip/out/android-arm64-chip-tool'
Traceback (most recent call last):
  File "./scripts/build/build_examples.py", line 230, in <module>
2023-07-27 10:46:15 WARNING ninja: error: loading 'build.ninja': No such file or directory
   main(auto_envvar_prefix='CHIP')
```

- When I use Android Studio. I have the error:
```
In file included from /home/cuong/MatterHub/connectedhomeip/src/crypto/CHIPCryptoPALmbedTLS.cpp:23:
  /home/cuong/MatterHub/connectedhomeip/src/crypto/CHIPCryptoPAL.h:26:10: fatal error: 'crypto/CryptoBuildConfig.h' file not found
  #include <crypto/CryptoBuildConfig.h>
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
  1 error generated.
```
**In :app:compileDebugKotlin and :app:externalNativeBuildDebug**

### Bug prevalence

every

### GitHub hash of the SDK that was being used

v1.1 branch

### Platform

android

### Platform Version(s)

1.1

### Type

Platform Issue

yunhanw-google, 2023-08-01:
Have you run source scripts/bootstrap.sh? this step would take a while, but would be helpful for build.ninja

yunhanw-google, 2023-08-01:
@ngocuong86

ngocuong86, 2023-08-01:
> Have you run source scripts/bootstrap.sh? this step would take a while, but would be helpful for build.ninja

Yes, I ran bootstrap

ngocuong86, 2023-08-01:
> Have you run source scripts/bootstrap.sh? this step would take a while, but would be helpful for build.ninja

I don't see build.ninja in android-arm64-chip-tool
[image]

yunhanw-google, 2023-08-09:
could you consider to use  docker to complete android chip-tool
docker run -it -v ~/workspace/connectedhomeip:/connectedhomeip ghcr.io/project-chip/chip-build-android:1
git config --global --add safe.directory /connectedhomeip 
git config --global --add safe.directory /connectedhomeip/third_party/pigweed/repo
source scripts/bootstrap.sh
./scripts/build/build_examples.py --target android-arm64-chip-tool build

yunhanw-google, 2023-08-09:
if further questions, you can find me in slack channel for connectedhomeip for the help. thanks

jorgezim, 2023-08-23:
Hello, I am having this issue as well while building the android example app via Android Studio. Any ETA of when this will be fixed?