project-chip/connectedhomeip #28325: [CERT-TEST-FAILURE] [TC-REFALM-2.2] - Test Failed

Rajashreekalmane, 2023-07-27:
### Feature Area

Area:Refrigerators

### Test Case

TC-REFALM-2.2

### Reproduction steps

 Chip-tool should implement the below mentioned command 

- [ ] Reset Command

PFA execution screenshot : 
[image]

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

raspi

jadhavrohit924, 2023-07-28:
Reset feature should be disabled in RefrigeratorAlarm cluster, so commands/attributes with RESET conference should be disabled. https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/RefrigeratorAlarm.adoc#4-features

jadhavrohit924, 2023-07-31:
@Rajashreekalmane Can we close this?

closed 2023-08-08 by Rajashreekalmane