project-chip/connectedhomeip #28329: Question about TLV::TLVReader class

OlivierGre, 2023-07-27:
Is there a way to make a full copy of a TLV::TLVReader object?

I have tried this:
```
TLV::TLVReader myTlvReader;
myTlvReader.Init(aReader);
```
Right affter this init, I dump the two readers (aReader and myTlvReader) and they are indeed the same. But it I dump myTlvReader later on (when "aReader" is no more valid), I can see that the container data are no more available. So it seems that both object were in fact using the same container data.

Would you have advices to do a full copy of a TLVReader object? 
Is there a way to copy the TLVReader data into an uint8_t array?

Thank you

bzbarsky-apple, 2023-07-27:
> But it I dump myTlvReader later on (when "aReader" is no more valid), I can see that the container data are no more available.

Generally, what you are doing should work, as long as the underlying buffer backing `aReader` sticks around.

It's hard to say what "later on when aReader is no more valid" means, but the real question is: what happened to its backing buffer?

>  Is there a way to copy the TLVReader data into an uint8_t array?

A TLVReader, in general, can be pointing to a data stream.  Including an infinite data stream.

OlivierGre, 2023-07-28:
> But it I dump myTlvReader later on (when "aReader" is no more valid), I can see that the container data are no more available.

I can see that there was a typo. I meant: "But if I dump myTlvReader  later".

> Generally, what you are doing should work, as long as the underlying buffer backing `aReader` sticks around.

> It's hard to say what "later on when aReader is no more valid" means, but the real question is: what happened to its backing buffer?

That's the point. I want to be able to keep a copy of the TlvReader even when aReader will no more be valid anymore.

> A TLVReader, in general, can be pointing to a data stream. Including an infinite data stream.

Would you have an example?

Thanks

bzbarsky-apple, 2023-07-28:
>  I want to be able to keep a copy of the TlvReader even when aReader will no more be valid anymore.

Then you have to actually copy the backing state.  I don't believe there is a good way to do that, in general, because the backing state might not allow rewinding.

If you know your reader is backed by a single buffer and does not have the "backing state cannot be rewound" problem, you can init a reader as above, set up a new buffer with a TLVWriter pointed to it, keep doing CopyElement on the writer until you run out of reader bits, finalize the writer, and then create a new TLVReader pointing to your buffer.

> Would you have an example?

A reader initialized via:
```
    CHIP_ERROR Init(TLVBackingStore & backingStore, uint32_t maxLen = UINT32_MAX);
```

The backing store can keep returning things from `GetNextBuffer` forever.  Why one would do that in practice is a separate issue, of course.

closed 2025-05-20 by OlivierGre