project-chip/connectedhomeip #28330: need to know how to send data to attributes

aravindone, 2023-07-27:
### Reproduction steps

Dear devs  i need to know how to push the data got by reading from gpio or virtually the file to their respective attributes on an ESP 32. some links provided will be useful. and  if have docs related to temperature measurement example using the esp 32 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

bfa05804ac3c224c9e70921a857ffcac32ccc7f4

### Platform

esp32, other, core

bzbarsky-apple, 2023-07-27:
> to their respective attributes 

It depends on the cluster.  Generally, clusters should define APIs for manipulating their state, but not all of them do.

So what is the specific situation (which cluster, which attribute?) here @aravindone ?

aravindone, 2023-07-28:
@bzbarsky-apple  i need to update the values in Electrical measurement cluster (Instanteous voltage , Active power max etc ) for my project

bzbarsky-apple, 2023-07-28:
For that specific implementation, there is no particular API defined, so the best you can do are the Get/Set functions in the relevant attribute namespaces in `zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.h`.

So for example, from that file:
```
namespace ElectricalMeasurement {
namespace Attributes {

namespace MeasurementType {
EmberAfStatus Get(chip::EndpointId endpoint, uint32_t * value); // bitmap32
EmberAfStatus Set(chip::EndpointId endpoint, uint32_t value);
} // namespace MeasurementType

// etc
```

aravindone, 2023-08-08:
Hi @bzbarsky-apple and @shubhamdp  to call this above header ,i can do the calling in main.cpp file of esp32 ? any other checkpoints or dependencies i need to consider ?

bzbarsky-apple, 2023-08-08:
@aravindone You can call it wherever, but you must make sure to either hold the Matter lock or do it from code that was dispatched to run on the Matter event queue.

aravindone, 2023-08-11:
@shubhamdp @dhrishi 

while compiling the modified main.cpp of all-cluster-app
the script modification .
`
#include <app-common/zap-generated/attributes/Accessors.h>
extern "C" void app_main()
{
    ESP_LOGI(TAG, "==========printcpprunning====");
    ElectricalMeasurement::Attributes::MeasurementType myObj; `

output wihle building :

[modification_in _app_main.TXT](https://github.com/project-chip/connectedhomeip/files/12322276/modification_in._app_main.TXT)

bzbarsky-apple, 2023-08-11:
> `ElectricalMeasurement::Attributes::MeasurementType myObj;`

That's not the right type.  First of all, unless you have the right namespaces pulled in it should be `chip::app::Clusters::ElectricalMeasurement::Attributes::MeasurementType`.  But then that's a namespace, not a type.  If you want a type, `chip::app::Clusters::ElectricalMeasurement::Attributes::MeasurementType::TypeInfo::Type`, or just use `uint32_t`.

aravindone, 2023-08-12:
Hi @bzbarsky-apple  thanks for the response.

https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32/main/main.cpp file .

`#include <app-common/zap-generated/attributes/Accessors.h>
extern "C" void app_main()
{
 chip::app::Clusters::ElectricalMeasurement::Attributes::TotalApparentPower myObj;`

facing error: expected primary-expression before 'myObj' 

what could be the reason for this?

bzbarsky-apple, 2023-08-12:
> what could be the reason for this?

Again: `chip::app::Clusters::ElectricalMeasurement::Attributes::TotalApparentPower` is not a type.  It's a namespace.

aravindone, 2023-08-12:
Hi @bzbarsky-apple thanks for the response.

please refer the below error logs what could be the reason.
**"chip[ZCL]: 0x3f409515ep 1 clus 0x0000_0B04 attr 0x0000_0306 not supported"**

https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32/main/main.cpp file .

#include <app-common/zap-generated/attributes/Accessors.h> 
extern "C" void app_main() {
    ESP_LOGI(TAG, "==========printcpprunning====");
    uint32_t value2 = 10;
    chip::EndpointId DeviceID =1;
    
   EmberAfStatus setStatus = chip::app::Clusters::ElectricalMeasurement::Attributes::TotalApparentPower::Set(DeviceID, value2);
    if (setStatus == EMBER_ZCL_STATUS_SUCCESS) {
    
    ESP_LOGI(TAG, "=======Attributesetsuccessfully====");
    } else {
    
    ESP_LOGE(TAG, "=======Attributeset_error====");
    } `

error logs:
`I (855) all-clusters-app: ==========printcpprunning====
I (855) chip[ZCL]: 0x3f409515ep 1 clus 0x0000_0B04 attr 0x0000_0306 not supported
E (865) all-clusters-app: =======Attributeset_error====`

aravindone, 2023-08-12:
sample linky electrical meter reading 

ADCO 841875399223 T
OPTARIF BASE 0
ISOUSC 30 9
BASE 007491943 0
PTEC TH.. $
IINST 001 X
IMAX 090 H
PAPP 00350 )
HHPHC A ,
MOTDETAT 000000 B

to set "PAPP" im tapping to TotalApparentPower Attribute as above in sample code,but facing error.
If it works, as a result i will ckech from Chip Repl tool whether value 10 is loaded to TotalApparentPower Attribute

aravindone, 2023-08-16:
blob:https://teams.microsoft.com/d847d7ba-eb55-47d4-b4a3-131b202e6921

aravindone, 2023-08-16:
> "chip[ZCL]: 0x3f409515ep 1 clus 0x0000_0B04 attr 0x0000_0306 not supported"

hi @bzbarsky-apple  and @shubhamdp .In addition to the above comment . i have checked with the python repl after commissioning the change of the attribute values doesn't happen , need to know why the attribute values are not updating and please guide us for the solution please.

Best regards 
Aravind

bzbarsky-apple, 2023-08-28:
> what could be the reason

For the "not supported" bit?  Presumably the attribute in question for the cluster instance in question is not enabled in the endpoint config for the device.  If you are using ZAP, check your ZAP config.  If you are using some sort of dynamic setup, check your dynamic definition of the endpoint config.

aravindone, 2023-08-31:
@bzbarsky-apple thank you for the reply , im using zap , i changed the Access in the .matter file and it works ,im closing this issue now

closed 2023-08-31 by aravindone

bzbarsky-apple, 2023-08-31:
>  i changed the Access in the .matter file 

Should be changed in the .zap file, really...

aravindone, 2023-09-01:
Hi @bzbarsky-apple .
> Should be changed in the .zap file, really...
from the below screenshot i tried with Instantaneous line current attribute ,i could say that .zap file is used for data model structure and .matter is used for the callbacks files while i use idf.py Build command  and previously it was in readonly condition i changed it to write access and now the values are updating.
 
[image]
[image]
[image]
[image]
[image]

bzbarsky-apple, 2023-09-02:
Um....  Per spec that attribute is readonly.  That only applies to writes over the wire, though, not from inside the application....  You should not have to change it to be writable (a spec violation) to be able to modify its value.

aravindone, 2023-09-05:
hi @bzbarsky-apple , yes i understand the vulnerability of give write access to attribute ,the thing is 

> EmberAfStatus Set(chip::EndpointId endpoint, uint32_t value);

https://developers.home.google.com/matter/tools/zap : link i followed for program on device
for this set function to work it should have the write permission according to that my thinking

> That only applies to writes over the wire, though, not from inside the application....

 if write can be done over wire what would be the steps to follow?

bzbarsky-apple, 2023-09-07:
> for this set function to work it should have the write permissio

No.  It will work for readonly attributes.  The "is writable" check is only done for writes coming from the wire (i.e. from Write Request messages).

> if write can be done over wire what would be the steps to follow?

I'm not sure I follow the question.