project-chip/connectedhomeip #28360: [CERT-TEST-FAILURE] TC-IDM-4.2 - Test Failed

Rajashreekalmane, 2023-07-28:
### Feature Area

Other

### Test Case

 TC-IDM-4.2 

### Reproduction steps

TC-IDM-4.2 : As per test-plan Step 6 is expecting to configure TH2 without access to a specific node ID, 

Is there a way to give access to a specific node ID in chip-tool ?

PFA test-plan screenshot :
[image]

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

c1ec2d777456924dcaa59b53351b00d73caf378f

### Platform

raspi

andy31415, 2023-08-29:
To deny access to a node, relevant ACLs need to be set (or removed).

andy31415, 2023-08-29:
Needs an owner to describe how ACLs can be set/updated to meet the test requirement.

Rajashreekalmane, 2023-10-09:
Validated the test step by following the method as mentioned below 

sending a command along with unregistered commissioner node id (5)

 any subscribe-by-id 0xFFFFFFFF 0xFFFFFFFF 10 100 1 0XFFFF --commissioner-nodeid 5
On CR2, Verify that the DUT returns a ""INVALID_ACTION"" status response.

[1695815340.162063][2522:2524] CHIP:EM: Rxd Ack; Removing MessageCounter:187875065 from Retrans Table on exchange 10455i
[1695815340.162181][2522:2524] CHIP:DMG: StatusResponseMessage =
[1695815340.162244][2522:2524] CHIP:DMG: {
[1695815340.162298][2522:2524] CHIP:DMG:         Status = 0x80 (INVALID_ACTION),
[1695815340.162357][2522:2524] CHIP:DMG:         InteractionModelRevision = 11
[1695815340.162412][2522:2524] CHIP:DMG: }
[1695815340.162501][2522:2524] CHIP:IM: Received status response, status is 0x80 (INVALID_ACTION)

closed 2023-10-09 by Rajashreekalmane