project-chip/connectedhomeip #28362: [CHIPTool] getDiscoveredDevice returns null

mukesh-simform, 2023-07-28:
### Reproduction steps

I've to get all available matter devices so that I can commission devices using IP Addresses, But unfortunately, I'm not getting device data, all the time getDiscoveredDevice returns a null value.

I've implemented the functioning in a separate custom app and for the chip controller I've used the following .jar and .so files.

https://github.com/google-home/sample-apps-for-matter-android/tree/main/3p-ecosystem/third_party/connectedhomeip/libs

Because I'm not able to generate .jar and .so files here, and I think that may be an issue because the same functioning is working file in the chip tool sample app and even I'm getting node data using this command `./chip-tool discover commissionables`

**So maybe something is wrong with the .jar and .so files and I'm not able to generate the same files using Chiptool**

**How can I generate and locate .jar and .so files ?**

**Custom App Logs**

[image]

**Chiptool App Logs**

[image]

**Implementation**

[image]

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

8f66f42

### Platform

android

### Platform Version(s)

v1.1.0.1

### Anything else?

**How can I generate and locate .jar and .so files ?**

topraveenpaul, 2023-08-07:
@mukesh-simform .  I see that you have mentioned that you get output from the discover commissionables. However, im unable to.

Can you please check this defect please? I'm really struggling to get this work. 

https://github.com/project-chip/connectedhomeip/issues/27964

mukesh-simform, 2023-08-11:
@topraveenpaul I'm not sure about your issue but this is how I resolved to get discovered devices

I've generated .so and .jar files from the matter SDK repository and replaced legacy files with the newly generated ones and now I'm able to discover the devices and get IP Addresses.

But Before I try to discover commissionable I have to enable BLE advertisement from ESP32 and then only it finds the service

**Prepare environment**

source scripts/bootstrap.sh

**Build directories**

gn gen out/host

ninja -C out/host

**Prepare some of the dependencies**

TARGET_CPU=arm64 ./scripts/examples/android_app_ide.sh

**Generate the jar and so files**

./scripts/build/build_examples.py --target android-arm64-chip-tool build

--target android-arm64-chip-tool 

--target android-arm-chip-tool 

--target android-x86-chip-tool 

--target android-x64-chip-tool

topraveenpaul, 2023-08-11:
Thank you very much for your input. I'll strictly follow the suggested step to see if it works.

topraveenpaul, 2023-11-07:
> I've generated .so and .jar files from the matter SDK repository and replaced legacy files with the newly generated ones and now I'm able to discover the devices and get IP Addresses.

Sorry, i'm getting back on this after a while. 
I followed your directions and generated shared objects and jar files however, it still didnt make any chance. Did not discover the devices.  Which specific version of SDK that you tried this on? 

Every time I tried, i had ensured that the device was on BLE advertising mode and was detectable by hcitool. 
Any other suggestion?