project-chip/connectedhomeip #28387: [Feature] Review Common Profile Tags Support Within Json-Tlv-Json Convertor

emargolis, 2023-07-29:
### Feature description

Current Json-Tlv-Json converter supports Context and Common Profile tags. There are some issue associated with that, reflected in the following comments:
https://github.com/project-chip/connectedhomeip/pull/27635/files/f25def68e4127a807121069f11dfd5a411969b38#r1275438623
https://github.com/project-chip/connectedhomeip/pull/27635/files/f25def68e4127a807121069f11dfd5a411969b38#r1275439222

The current implementation requires Common Profile Tag Number to be greater than 255. If the Common Profile tag number is less or equal to 255 it cannot be distinguished from the Context Tag in Json. 

@turon @mrjerryjohns @yunhanw-google 

### Platform

core

yunhanw-google, 2023-08-04:
We have decided to replace common profile id with implicit profile id, and would update across kotlin side as well using this ticket.

yunhanw-google, 2023-09-05:
> From @bzbarsky-apple , So if we had a Common tag that happened to fall into the "less than 256" range, we would treat it as a Context tag?
> [InternalConvertTlvTag](https://github.com/project-chip/connectedhomeip/blob/54038c0db7349cb54461dbb59418674d747123cf/src/lib/support/jsontlv/JsonToTlv.cpp#L182)
> 
>     else if (tagNumber <= UINT32_MAX)
>     {
>         tag = TLV::ProfileTag(profileId, static_cast<uint32_t>(tagNumber));
>     }
I think we should immediately raise the error when detecting the number within UINT32_MAX range with implicit ProfileId.

bzbarsky-apple, 2023-09-08:
More useful links to the discussion, with my notes:

* https://github.com/project-chip/connectedhomeip/pull/27635#discussion_r1275438623 -- this is because we are trying to support things that are not actual data model payloads.  The format of these things is not well-defined in the jsontlv API, but needs to be.
* https://github.com/project-chip/connectedhomeip/pull/27635#discussion_r1275439222 -- this is fixed per the code cited in https://github.com/project-chip/connectedhomeip/issues/28387#issuecomment-1705831897

yunhanw-google, 2024-03-27:
https://github.com/project-chip/connectedhomeip/pull/32276

closed 2024-03-27 by yunhanw-google