project-chip/connectedhomeip #28392: [1.0] Failed in verifying 'Attestation Information' command received from the device: err 601

vineelasmile, 2023-07-29:
### Reproduction steps

I am working on Matter SDK built from 1.0 and hit the same issue as reported in #23164
I am trying to pair with ESP devkit light app. 

When pairing from chip-tool, I see that the pairing progresses through commissioning step however when programmatically doing from our logic , it fails with the error :
Failed in verifying 'Attestation Information' command received from the device: err 601

It is clear from the debug logs that the key is not in the testCDs that Matter SDK is looking for here: 
```
[connectedhomeip/src/credentials/attestation_verifier/DefaultDeviceAttestationVerifier.cpp](https://github.com/project-chip/connectedhomeip/blob/7b284c027bd5499ca3c5be00d51f83715e8969d6/src/credentials/attestation_verifier/DefaultDeviceAttestationVerifier.cpp#L472-L479)

Lines 472 to 479 in [7b284c0](https://github.com/project-chip/connectedhomeip/commit/7b284c027bd5499ca3c5be00d51f83715e8969d6)

 CHIP_ERROR err = mCdKeysTrustStore.LookupVerifyingKey(kid, verifyingKey); 
 VerifyOrReturnError(err == CHIP_NO_ERROR, AttestationVerificationResult::kCertificationDeclarationNoCertificateFound); 
  
 // Disallow test key if support not enabled 
 if (mCdKeysTrustStore.IsCdTestKey(kid) && !IsCdTestKeySupported()) 
 { 
     return AttestationVerificationResult::kCertificationDeclarationNoCertificateFound; 
 }
```

My logs:
```
07-29 04:04:07.574 12576 12591 D CTL     : 0xfe, 0x34, 0x3f, 0x95, 0x99, 0x47, 0x76, 0x3b,
07-29 04:04:07.574 12576 12591 D CTL     : 0x61, 0xee, 0x45, 0x39, 0x13, 0x13, 0x38, 0x49,
07-29 04:04:07.574 12576 12591 D CTL     : 0x4f, 0xe6, 0x7d, 0x8e, ---> kid

07-29 04:04:07.574 12576 12591 D CTL     : 0x62, 0xfa, 0x82, 0x33, 0x59, 0xac, 0xfa, 0xa9,
07-29 04:04:07.574 12576 12591 D CTL     : 0x96, 0x3e, 0x1c, 0xfa, 0x14, 0x0a, 0xdd, 0xf5,
07-29 04:04:07.574 12576 12591 D CTL     : 0x04, 0xf3, 0x71, 0x60, --> gCdKids[0]

07-29 04:04:07.574 12576 12591 D CTL     : 0x83, 0xfa, 0xd7, 0x81, 0xe8, 0x2d, 0x01, 0x8b,
07-29 04:04:07.574 12576 12591 D CTL     : 0x4f, 0x14, 0xf0, 0xa1, 0xc7, 0xf6, 0x84, 0x01,
07-29 04:04:07.574 12576 12591 D CTL     : 0x8c, 0xc5, 0xd1, 0x9f, --> gCdKids[1]

.....
07-29 04:04:07.574 12576 12591 E CTL     : Failed in verifying 'Attestation Information' command received from the device: err 601. Look at AttestationVerificationResult enum to understand the errors
```

* What is the significance of SHA commit hash where matter SDK was built from and what is flashed on ESP32 devkit (as mentioned in the closed issue)?
* Any hints and pointers on how to fix the problem ?

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

1.0 head

### Platform

efr32

### Type

Platform Issue

gokul-bhalodiya-7, 2023-07-30:
> What is the significance of SHA commit hash where matter SDK was built from and what is flashed on ESP32 devkit (as mentioned in the closed issue)?

If you have flashed binary in end-device(ESP devkit) having CD certs other than test CD certs then you need to pass the path of CD certs while pairing from chip-tool.

> Any hints and pointers on how to fix the problem ?

You can try as @bzbarsky-apple mentioned in reply of an issue [https://github.com/project-chip/connectedhomeip/issues/26142#issuecomment-1517939129](https://github.com/project-chip/connectedhomeip/issues/26142#issuecomment-1517939129)

vineelasmile, 2023-07-31:
In my case, I am not flashing the ESP devkit from my code base but just using what they have provided here: https://espressif.github.io/esp-launchpad/?flashConfigURL=https%3A%2F%2Fespressif.github.io%2Fesp-matter%2Flaunchpad.toml

I think i know the root cause, based on https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/attestation_verifier/DefaultDeviceAttestationVerifier.cpp the test key is validated using `gCdSigningKey001Kid` however this key changed between v1.0 and master. 

For testing purposes, if I swap my local version of the code(ie v1.0) to use master's test key is that okay ? Or any recommended steps to follow ?

vineelasmile, 2023-07-31:
Thank you ! Swapping the keys unblocks me and I am able to progress beyond chip device attestation.

closed 2023-07-31 by vineelasmile