project-chip/connectedhomeip #28409: Matter client developer profile

wilfriedgarry45, 2023-07-31:
### Test issue(s)

To be able to paire a matter device on ios I need to installed on my iphone the **matter client developper profile** as indicate **on the connectedhomeip documentation**
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md

and in **the apple documentation**
[Adding Matter support to your ecosystem | Apple Developer Documentation](https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem?language=objc)
(To test your app on the iOS or macOS device that initiates the pairing, [download](https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableBluetoothCentralMatterClientDeveloperMode.mobileconfig) the developer profile now, then install it.)

When I do that it works perfectly. 
Otherwise the documentation says that the profile is only needed for development but when I want to use my app from the apple store (validate by Apple) and when I remove the profile it doesn't work anymore.

What do I have to do to paire Matter device on iphone without the Matter client developer profile.

### Platform

darwin

melo213, 2023-08-23:
same for me

wilfriedgarry45, 2023-08-23:
> same for me
Hi melo213,
Did you implement the iOS Matter library on your own app or do you use the connected home ip app for testing ?

bzbarsky-apple, 2023-09-01:
> What do I have to do to paire Matter device on iphone without the Matter client developer profile.

@wilfriedgarry45 did you implement the bits MatterSupport needs (the app extension, etc)?  If you do that and use MatterSupport to bring the device onto the network, then use the setup payload MatterSupport hands you to commission, things should work.

closed 2023-09-01 by bzbarsky-apple

melo213, 2023-09-04:
@bzbarsky-apple
I did this as you say. I use the setup payload MatterSupport hands me, use the Matter.framework method "setupCommissioningSessionWithPayload:newNodeID:error:", then got an error "[Discovery] Mdns: Browse failure (kDNSServiceErr_Timeout)"

bzbarsky-apple, 2023-09-08:
@melo213 Please file a feedback assistant ticket with logs and put the ticket id here so I can follow up.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/

wilfriedgarry45, 2024-03-14:
Hi Robin,

Is it possible to have more information about this video. Is it with Mattersupport framework or only matter with ble profile installed on the iphone.

Otherwise the only trouble I have is for storing credential on the iphone. If I use the default threadnetwork on the iphone the matter pairing is working well.

Best regards,

[Logo] <https://www.otodo.com/>
104 boulevard du Montparnasse
75014 Paris
www.otodo.com<http://www.otodo.com>

Wilfried Garry
Software Developer
***@***.******@***.***>
[Facebook icon] <https://www.facebook.com/OtodoOfficiel/> [Twitter icon]  <https://twitter.com/OtodoOfficiel> [Linkedin icon]  <https://www.linkedin.com/company/otodo/?lipi=urn%3Ali%3Apage%3Ad_flagship3_company_admin%3BJ1FpXoKeRwWGYatbYhzdaQ%3D%3D> [Youtube icon]  <https://www.youtube.com/channel/UCHC7TXW-IldDkKhoZ8MAfMA?view_as=subscriber>