project-chip/connectedhomeip #28429: [BUG] libCHIPController.so crash when invoke readOnOffAttribute  method of OnOffCluster class

netscell, 2023-08-01:
### Reproduction steps

after pair device , read the matter plug onoff state, occurs crash sometimes:

log1:

Process: 
PID: 20264
UID: 10178
Frozen: false
Flags: 0x20c83e46
Package: 
Foreground: No
Process-Runtime: 4879686
Lifetime: 4879s
Build: 
Loading-Progress: 1.0
Wakefulness: Dozing 2023-07-31 18:21:08.857
KeyguardShowing: true 2023-07-31 18:21:09.220
Dropped-Count: 0

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 
Revision: '0'
ABI: 'arm64'
Timestamp: 2023-07-31 19:40:15.214378543+0800
Process uptime: 4207s
Cmdline: 
pid: 20264, tid: 21294, name: 
uid: 10178
tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
pac_enabled_keys: 000000000000000f (PR_PAC_APIAKEY, PR_PAC_APIBKEY, PR_PAC_APDAKEY, PR_PAC_APDBKEY)
signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
    x0  0000000000000000  x1  000000000000532e  x2  0000000000000006  x3  0000007896a186c0
    x4  0000000000000001  x5  0000000000000001  x6  0000000000000001  x7  0000000000000000
    x8  00000000000000f0  x9  0000007c8bc51398  x10 0000000000000001  x11 0000007c8bc915b8
    x12 0000000000004eb6  x13 12a523d4061d6ec5  x14 0000000000000006  x15 ffffffffffffffff
    x16 0000007c8bcf5f88  x17 0000007c8bcd3450  x18 00000078968da000  x19 0000000000004f28
    x20 000000000000532e  x21 00000000ffffffff  x22 000000789f7f2528  x23 000000789fc044cd
    x24 00000079cd400880  x25 0000007896a18c38  x26 0000007896a18c48  x27 0000007896a18c38
    x28 0000007896a18b30  x29 0000007896a18740
    lr  0000007c8bc83044  sp  0000007896a186a0  pc  0000007c8bc83070  pst 0000000000001000
backtrace:
  NOTE: Function names and BuildId information is missing for some frames due
  NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
  NOTE: found under the lib/ directory are readable.
  NOTE: On this device, run setenforce 0 to make the libraries readable.
  NOTE: Unreadable libraries:
  NOTE:   lib/arm64-v8a/libCHIPController.so
      #00 pc 0000000000054070  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: 25eadd32a5b6753e7e58e01ae8014530)
      #01 pc 000000000087a934  /lib/arm64-v8a/libCHIPController.so
      #02 pc 000000000087a88c  /lib/arm64-v8a/libCHIPController.so
      #03 pc 0000000000c6a764  /lib/arm64-v8a/libCHIPController.so
      #04 pc 0000000000ca0da4  lib/arm64-v8a/libCHIPController.so
      #05 pc 0000000000ca0c6c  lib/arm64-v8a/libCHIPController.so
      #06 pc 000000000021a354  /apex/com.android.art/lib64/libart.so (art_quick_generic_jni_trampoline+148) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #07 pc 000000000020a910  /apex/com.android.art/lib64/libart.so (nterp_helper+5648) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #08 pc 000000000013c528  [anon:dalvik-classes8.dex extracted in memory from debug.apk!classes8.dex] (chip.devicecontroller.ChipClusters$OnOffCluster.readOnOffAttribute+4)
      #09 pc 000000000020a254  /apex/com.android.art/lib64/libart.so (nterp_helper+3924) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #10 pc 0000000000180b00  [anon:dalvik-classes8.dex extracted in memory from debug.apk!classes8.dex] (chip.devicecontroller.ClusterReadMapping.lambda$getReadAttributeMap$52+8)
      #11 pc 0000000000209334  /apex/com.android.art/lib64/libart.so (nterp_helper+52) (BuildId: d53f2ad8deb01876e35b599a414617a1)

### Bug prevalence

10 + times one day 

### GitHub hash of the SDK that was being used

NA

### Platform

android

netscell, 2023-08-01:
android crash log2:
PID: 19838
UID: 10524
Frozen: false
Flags: 0x20c83e46
Package: 
Foreground: No
Process-Runtime: 689038010
Lifetime: 689038s
Build: 
Loading-Progress: 1.0
Wakefulness: Dozing 2023-07-31 15:40:55.241
KeyguardShowing: true 2023-07-31 15:40:55.657
Dropped-Count: 0

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 
Revision: '0'
ABI: 'arm64'
Timestamp: 2023-07-31 15:44:01.733185633+0800
Process uptime: 19s
Cmdline: 
pid: 19838, tid: 20152, name: 
uid: 10524
tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
pac_enabled_keys: 000000000000000f (PR_PAC_APIAKEY, PR_PAC_APIBKEY, PR_PAC_APDAKEY, PR_PAC_APDBKEY)
signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
    x0  0000000000000000  x1  0000000000004eb8  x2  0000000000000006  x3  00000071d47a16c0
    x4  0000000000000001  x5  0000000000000001  x6  0000000000000001  x7  0000000000000038
    x8  00000000000000f0  x9  00000074ae25c398  x10 0000000000000001  x11 00000074ae29c5b8
    x12 0000000000001f93  x13 0000000000000006  x14 0000000000000000  x15 00000074ae25fffa
    x16 00000074ae300f88  x17 00000074ae2de450  x18 00000070ec666000  x19 0000000000004d7e
    x20 0000000000004eb8  x21 00000000ffffffff  x22 00000071308a2b0c  x23 0000007130cbf4cd
    x24 00000071e9c00880  x25 00000071d47a1c38  x26 00000071d47a1c48  x27 00000071d47a1c38
    x28 00000071d47a1b30  x29 00000071d47a1740
    lr  00000074ae28e044  sp  00000071d47a16a0  pc  00000074ae28e070  pst 0000000000001000
backtrace:
  NOTE: Function names and BuildId information is missing for some frames due
  NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
  NOTE: found under the lib/ directory are readable.
  NOTE: On this device, run setenforce 0 to make the libraries readable.
  NOTE: Unreadable libraries:
  NOTE:   /lib/arm64-v8a/libCHIPController.so
      #00 pc 0000000000054070  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: 25eadd32a5b6753e7e58e01ae8014530)
      #01 pc 000000000087a934  /lib/arm64-v8a/libCHIPController.so
      #02 pc 000000000087a88c  /lib/arm64-v8a/libCHIPController.so
      #03 pc 0000000000d88284  /lib/arm64-v8a/libCHIPController.so
      #04 pc 0000000000ddd2a4  /lib/arm64-v8a/libCHIPController.so
      #05 pc 0000000000ddd16c  /lib/arm64-v8a/libCHIPController.so
      #06 pc 000000000021a354  /apex/com.android.art/lib64/libart.so (art_quick_generic_jni_trampoline+148) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #07 pc 000000000020a910  /apex/com.android.art/lib64/libart.so (nterp_helper+5648) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #08 pc 0000000000131b0c  [anon:dalvik-classes8.dex extracted in memory from debug.apk!classes8.dex] (chip.devicecontroller.ChipClusters$BasicInformationCluster.readSoftwareVersionStringAttribute+4)
      #09 pc 000000000020a254  /apex/com.android.art/lib64/libart.so (nterp_helper+3924) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #10 pc 000000000017f7e0  [anon:dalvik-classes8.dex extracted in memory from debug.apk!classes8.dex] (chip.devicecontroller.ClusterReadMapping.lambda$getReadAttributeMap$242+8)
      #11 pc 0000000000209334  /apex/com.android.art/lib64/libart.so (nterp_helper+52) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #12 pc 0000000000118d98  [anon:dalvik-classes8.dex extracted in memory from debug.apk!classes8.dex] (chip.devicecontroller.-$$Lambda$ClusterReadMapping$KyXGKk8eDmcdLgBkehGDxIfYGr0.invokeCommand+0)

netscell, 2023-08-01:
crash log:
Process: 
PID: 22183
UID: 10523
Frozen: false
Flags: 0x20c83e46
Package: 
Foreground: No
Process-Runtime: 689095799
Lifetime: 689095s
Build: 
Loading-Progress: 1.0
Wakefulness: Dozing 2023-07-31 14:37:01.097
KeyguardShowing: true 2023-07-31 14:37:01.529
Dropped-Count: 0

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: ''
Revision: '0'
ABI: 'arm64'
Timestamp: 2023-07-31 14:43:53.832606437+0800
Process uptime: 76s
Cmdline: 
pid: 22183, tid: 22407, name: CHIP Device Con  
uid: 10523
tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
pac_enabled_keys: 000000000000000f (PR_PAC_APIAKEY, PR_PAC_APIBKEY, PR_PAC_APDAKEY, PR_PAC_APDBKEY)
signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
    x0  0000000000000000  x1  0000000000005787  x2  0000000000000006  x3  00000070ea548060
    x4  00000000ffffffff  x5  00000000ffffffff  x6  00000000ffffffff  x7  0000000000000010
    x8  00000000000000f0  x9  00000074ae25c398  x10 0000000000000001  x11 00000074ae29c5b8
    x12 000000000000b656  x13 ffffffffffffffff  x14 ff00000000000000  x15 ffffffffffffffff
    x16 00000074ae300f88  x17 00000074ae2de450  x18 00000070e9720000  x19 00000000000056a7
    x20 0000000000005787  x21 00000000ffffffff  x22 00000000000056dc  x23 00000000000056a7
    x24 00000070ea549cb0  x25 00000070ea549cb0  x26 00000070ea549ff8  x27 00000000000fc000
    x28 00000000000fe000  x29 00000070ea5480e0
    lr  00000074ae28e044  sp  00000070ea548040  pc  00000074ae28e070  pst 0000000000001000
backtrace:
  NOTE: Function names and BuildId information is missing for some frames due
  NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
  NOTE: found under the lib/ directory are readable.
  NOTE: On this device, run setenforce 0 to make the libraries readable.
  NOTE: Unreadable libraries:
  NOTE:  /lib/arm64-v8a/libCHIPController.so
      #00 pc 0000000000054070  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: 25eadd32a5b6753e7e58e01ae8014530)
      #01 pc 000000000087a934  lib/arm64-v8a/libCHIPController.so
      #02 pc 000000000174547c  lib/arm64-v8a/libCHIPController.so
      #03 pc 00000000017757c8  /lib/arm64-v8a/libCHIPController.so
      #04 pc 0000000001776390  /lib/arm64-v8a/libCHIPController.so
      #05 pc 00000000018fdbe0  /lib/arm64-v8a/libCHIPController.so
      #06 pc 00000000018eef4c  /lib/arm64-v8a/libCHIPController.so
      #07 pc 00000000018f48b8  /lib/arm64-v8a/libCHIPController.so
      #08 pc 00000000017de938  /lib/arm64-v8a/libCHIPController.so
      #09 pc 00000000017e1eb8  /lib/arm64-v8a/libCHIPController.so
      #10 pc 00000000017e1d94  /lib/arm64-v8a/libCHIPController.so
      #11 pc 00000000017d97e4  /lib/arm64-v8a/libCHIPController.so
      #12 pc 00000000017e16a8  /lib/arm64-v8a/libCHIPController.so
      #13 pc 00000000017e0ebc /lib/arm64-v8a/libCHIPController.so
      #14 pc 00000000017fda50  /lib/arm64-v8a/libCHIPController.so
      #15 pc 00000000017fc628  /lib/arm64-v8a/libCHIPController.so
      #16 pc 00000000018052d4  /lib/arm64-v8a/libCHIPController.so
      #17 pc 00000000018e6934 /lib/arm64-v8a/libCHIPController.so
      #18 pc 00000000018e65bc  /lib/arm64-v8a/libCHIPController.so
      #19 pc 000000000189dffc  /lib/arm64-v8a/libCHIPController.so
      #20 pc 000000000189d3b4  /lib/arm64-v8a/libCHIPController.so
      #21 pc 00000000017f463c  /lib/arm64-v8a/libCHIPController.so
      #22 pc 00000000018a8cf0  /lib/arm64-v8a/libCHIPController.so
      #23 pc 00000000008a0078  /lib/arm64-v8a/libCHIPController.so
      #24 pc 0000000000894abc  /lib/arm64-v8a/libCHIPController.so
      #25 pc 00000000000b8b98  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+208) (BuildId: 25eadd32a5b6753e7e58e01ae8014530)
      #26 pc 0000000000055794  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 25eadd32a5b6753e7e58e01ae8014530)

netscell, 2023-08-01:
another crash log:
Process: 
PID: 28196
UID: 10523
Frozen: false
Flags: 0x20c83e46
Package: 
Foreground: No
Process-Runtime: 689106576
Lifetime: 689106s
Build: 
Loading-Progress: 1.0
Wakefulness: Dozing 2023-07-31 12:03:21.863
KeyguardShowing: true 2023-07-31 12:03:22.373
Dropped-Count: 0

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 
Revision: '0'
ABI: 'arm64'
Timestamp: 2023-07-31 12:05:25.058147199+0800
Process uptime: 88s
Cmdline: 
pid: 28196, tid: 28402, name: 
uid: 10523
tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
pac_enabled_keys: 000000000000000f (PR_PAC_APIAKEY, PR_PAC_APIBKEY, PR_PAC_APDAKEY, PR_PAC_APDBKEY)
signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
    x0  0000000000000000  x1  0000000000006ef2  x2  0000000000000006  x3  00000071d47c07d0
    x4  0000000000000001  x5  0000000000000001  x6  0000000000000001  x7  00000071d47c0140
    x8  00000000000000f0  x9  00000074ae25c398  x10 0000000000000001  x11 00000074ae29c5b8
    x12 0000000000006867  x13 00000071d47bf710  x14 000000000000000a  x15 00000000ebad6a89
    x16 00000074ae300f88  x17 00000074ae2de450  x18 00000070ed330000  x19 0000000000006e24
    x20 0000000000006ef2  x21 00000000ffffffff  x22 0000007130895ad4  x23 0000007130cb24cd
    x24 00000071e9c00880  x25 00000071d47c0d48  x26 00000071d47c0d58  x27 00000071d47c0d48
    x28 00000071d47c0c40  x29 00000071d47c0850
    lr  00000074ae28e044  sp  00000071d47c07b0  pc  00000074ae28e070  pst 0000000000001000
backtrace:
  NOTE: Function names and BuildId information is missing for some frames due
  NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
  NOTE: found under the lib/ directory are readable.
  NOTE: On this device, run setenforce 0 to make the libraries readable.
  NOTE: Unreadable libraries:
  NOTE:   /lib/arm64-v8a/libCHIPController.so
      #00 pc 0000000000054070  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: 25eadd32a5b6753e7e58e01ae8014530)
      #01 pc 000000000087a934  /lib/arm64-v8a/libCHIPController.so
      #02 pc 000000000087a88c  /lib/arm64-v8a/libCHIPController.so
      #03 pc 0000000000d88284  /lib/arm64-v8a/libCHIPController.so
      #04 pc 0000000000dde528  /lib/arm64-v8a/libCHIPController.so
      #05 pc 0000000000dde3f0  /lib/arm64-v8a/libCHIPController.so
      #06 pc 000000000021a354  /apex/com.android.art/lib64/libart.so (art_quick_generic_jni_trampoline+148) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #07 pc 000000000020a910  /apex/com.android.art/lib64/libart.so (nterp_helper+5648) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #08 pc 0000000000131ad4  [anon:dalvik-classes8.dex extracted in memory from debug.apk!classes8.dex] (chip.devicecontroller.ChipClusters$BasicInformationCluster.readSerialNumberAttribute+4)
      #09 pc 000000000020a254  /apex/com.android.art/lib64/libart.so (nterp_helper+3924) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #10 pc 000000000017f880  [anon:dalvik-classes8.dex extracted in memory from debug.apk!classes8.dex] (chip.devicecontroller.ClusterReadMapping.lambda$getReadAttributeMap$252+8)
      #11 pc 0000000000209334  /apex/com.android.art/lib64/libart.so (nterp_helper+52) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #12 pc 00000000001285f8  [anon:dalvik-classes8.dex extracted in memory from debug.apk!classes8.dex] (chip.devicecontroller.-$$Lambda$ClusterReadMapping$s1qB_uprJrMdRR0v2tXyJMEx9yw.invokeCommand+0)
      #13 pc 000000000020b074  /apex/com.android.art/lib64/libart.so (nterp_helper+7540) (BuildId: d53f2ad8deb01876e35b599a414617a1)

netscell, 2023-08-10:
the above log may related to this reason:
when application invoke cluster's readxxxAttribute ,sometimes timeout occurs, and the session could be closed by matter sdk;
the application invoke another readxxxAttribute method using the same session,the sdk will encouter crash .

so is there a api can get the session status ? and  the sdk  using the closed session don't crashed;

pengxuetian, 2023-08-18:
I have encountered a similar bug. Is there a solution?

yunhanw-google, 2024-01-14:
@netscell could you retry this issue with latest code? session lifetime is managed in core sdk. Recently we have largely improve the stability, maybe it has been resolved? thanks

AriesHaw, 2024-01-17:
same issue，there is crash logs:
01-16 20:49:17.688 11782  9989 F google-breakpad: Microdump skipped (uninteresting)
01-16 20:49:17.724  9926  9989 F libc    : Fatal signal 6 (SIGABRT), code -6 in tid 9989 (mqt_native_modu), pid 9926 (.ewelinkcastapp)
01-16 20:49:17.863 11785 11785 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-16 20:49:17.863 11785 11785 F DEBUG   : Build fingerprint: 'Xiaomi/clover/clover:8.1.0/OPM1.171019.019/V10.3.2.0.ODJCNXM:user/release-keys'
01-16 20:49:17.863 11785 11785 F DEBUG   : Revision: '0'
01-16 20:49:17.864 11785 11785 F DEBUG   : ABI: 'arm64'
01-16 20:49:17.864 11785 11785 F DEBUG   : pid: 9926, tid: 9989, name: mqt_native_modu  >>> com.ewelinkcastapp <<<
01-16 20:49:17.864 11785 11785 F DEBUG   : signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
01-16 20:49:17.864 11785 11785 F DEBUG   :     x0   0000000000000000  x1   0000000000002705  x2   0000000000000006  x3   0000000000000008
01-16 20:49:17.864 11785 11785 F DEBUG   :     x4   0000000013bbd5b0  x5   0000000013bbd5b0  x6   0000000013bbd5b0  x7   0000000000000000
01-16 20:49:17.864 11785 11785 F DEBUG   :     x8   0000000000000083  x9   5848d8670f37abdb  x10  0000000000000000  x11  0000000000000001
01-16 20:49:17.864 11785 11785 F DEBUG   :     x12  ffffffffffffffff  x13  0000000000000000  x14  0000007f2f9d3720  x15  0000000000000000
01-16 20:49:17.864 11785 11785 F DEBUG   :     x16  0000000f8bacbfa8  x17  0000007fb0fb264c  x18  0000000000000008  x19  00000000000026c6
01-16 20:49:17.864 11785 11785 F DEBUG   :     x20  0000000000002705  x21  0000007f13cc7c00  x22  0000007f125e7ddc  x23  0000007f16c7bab2
01-16 20:49:17.864 11785 11785 F DEBUG   :     x24  0000000000000014  x25  0000007f125ea588  x26  0000007f13cc7ca0  x27  0000000000000005
01-16 20:49:17.864 11785 11785 F DEBUG   :     x28  0000000000000003  x29  0000007f125e7680  x30  0000007fb0f67eac
01-16 20:49:17.864 11785 11785 F DEBUG   :     sp   0000007f125e7640  pc   0000007fb0f67ec8  pstate 0000000060000000
01-16 20:49:17.867 11785 11785 F DEBUG   : 
01-16 20:49:17.867 11785 11785 F DEBUG   : backtrace:
01-16 20:49:17.867 11785 11785 F DEBUG   :     #00 pc 000000000001dec8  /system/lib64/libc.so (abort+104)
01-16 20:49:17.867 11785 11785 F DEBUG   :     #01 pc 0000000000007760  /data/app/com.ewelinkcastapp-WcD4K1rglFnv49PgDx8fBw==/lib/arm64/libCHIPController.so (offset 0x8d7000)

yunhanw-google, 2024-01-18:
@AriesHaw  may i know which commit id are you using when crash happens, thanks

yunhanw-google, 2024-01-18:
> the above log may related to this reason: when application invoke cluster's readxxxAttribute ,sometimes timeout occurs, and the session could be closed by matter sdk; the application invoke another [...]

@netscell  we cannot detect whether session is alive or not, 
1. when you continue to use the valid session(the other end is not alive), you fail to resend 4 message in exchange layer, then timeout  happens.
2. when you continue to use invalid session, you immediately get the error exception.

yunhanw-google, 2024-01-18:
sounds like it is duplicate with https://github.com/project-chip/connectedhomeip/issues/30624, close this one, working on 30624, thanks

closed 2024-01-18 by yunhanw-google