project-chip/connectedhomeip #28430: [BUG] crash occured when connect matter device using getConnectedDevicePointer

netscell, 2023-08-01:
### Reproduction steps

after pair device successfully,call getConnectedDevicePointer crash :

android crash log:
PID: 29428
UID: 10523
Frozen: false
Flags: 0x20c83e46
Package: 
Foreground: No
Process-Runtime: 689489520
Lifetime: 689489s
Build: 
Loading-Progress: 1.0
Wakefulness: Dozing 2023-07-31 15:10:45.643
KeyguardShowing: true 2023-07-31 15:10:46.024
Dropped-Count: 0

*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
Build fingerprint: 
Revision: '0'
ABI: 'arm64'
Timestamp: 2023-07-31 15:15:08.118542909+0800
Process uptime: 470s
Cmdline: 
pid: 29428, tid: 29624, name: 
uid: 10523
tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
pac_enabled_keys: 000000000000000f (PR_PAC_APIAKEY, PR_PAC_APIBKEY, PR_PAC_APDAKEY, PR_PAC_APDBKEY)
signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
    x0  0000000000000000  x1  00000000000073b8  x2  0000000000000006  x3  00000070ef739da0
    x4  00000000ffffffff  x5  00000000ffffffff  x6  00000000ffffffff  x7  0000000000000010
    x8  00000000000000f0  x9  00000074ae25c398  x10 0000000000000001  x11 00000074ae29c5b8
    x12 000000000000bd6f  x13 ffffffffffffffff  x14 ff00000000000000  x15 ffffffffffffffff
    x16 00000074ae300f88  x17 00000074ae2de450  x18 00000070edd68000  x19 00000000000072f4
    x20 00000000000073b8  x21 00000000ffffffff  x22 0000000014781730  x23 0000000000000038
    x24 0000000012cf5e68  x25 0000000012cf5e88  x26 0000007330acdad0  x27 000000725eb02770
    x28 00000070ef73a3b0  x29 00000070ef739e20
    lr  00000074ae28e044  sp  00000070ef739d80  pc  00000074ae28e070  pst 0000000000001000
backtrace:
  NOTE: Function names and BuildId information is missing for some frames due
  NOTE: to unreadable libraries. For unwinds of apps, only shared libraries
  NOTE: found under the lib/ directory are readable.
  NOTE: On this device, run setenforce 0 to make the libraries readable.
  NOTE: Unreadable libraries:
  NOTE:   /lib/arm64-v8a/libCHIPController.so
      #00 pc 0000000000054070  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: 25eadd32a5b6753e7e58e01ae8014530)
      #01 pc 000000000087a934  /lib/arm64-v8a/libCHIPController.so
      #02 pc 00000000017d95a0  /lib/arm64-v8a/libCHIPController.so
      #03 pc 00000000017b82d4  /lib/arm64-v8a/libCHIPController.so
      #04 pc 00000000017b7ae8  /lib/arm64-v8a/libCHIPController.so
      #05 pc 00000000017456f0  /lib/arm64-v8a/libCHIPController.so
      #06 pc 0000000001775824  /lib/arm64-v8a/libCHIPController.so
      #07 pc 0000000001775334  /lib/arm64-v8a/libCHIPController.so
      #08 pc 0000000001745204  /lib/arm64-v8a/libCHIPController.so
      #09 pc 0000000000897500  /lib/arm64-v8a/libCHIPController.so
      #10 pc 00000000008973a4  /lib/arm64-v8a/libCHIPController.so
      #11 pc 000000000021a354  /apex/com.android.art/lib64/libart.so (art_quick_generic_jni_trampoline+148) (BuildId: d53f2ad8deb01876e35b599a414617a1)
      #12 pc 0000000002173fbc  /memfd:jit-cache (deleted) (chip.devicecontroller.ChipDeviceController.getConnectedDevicePointer+236)

### Bug prevalence

1-2times a day

### GitHub hash of the SDK that was being used

NA

### Platform

android

xy371661665, 2023-08-30:
Same issue. And I found it must be crashed when App open with a long time.

jiangjun1989, 2023-09-12:
Same issue.

xy371661665, 2023-09-12:
谢谢，已经收到你的邮件~~

yunhanw-google, 2024-01-14:
Hi, @netscell @jiangjun1989 @xy371661665  could you try the latest master code, this issue may be fixed potentially in https://github.com/project-chip/connectedhomeip/pull/29710/files? thanks

xy371661665, 2024-01-14:
谢谢，已经收到你的邮件~~

yunhanw-google, 2024-01-31:
@netscell is this still issue?

xy371661665, 2024-01-31:
谢谢，已经收到你的邮件~~

xy371661665, 2024-02-01:
> Hi, @netscell @jiangjun1989 @xy371661665 could you try the latest master code, this issue may be fixed potentially in https://github.com/project-chip/connectedhomeip/pull/29710/files? thanks

This is not fixed. But I found a method to reproduction this crash.
It will crash when send any cluster after subscription is broken.

yunhanw-google, 2024-02-20:
@xy371661665 have you used android chip-tool to reproduce this issue. I just use light on/off level cluster fragment, create the subscription succesfully, toggle works fine, then power off my matter outlet, run toggle lots of times, and cannot reproduce this issue, thanks

xy371661665, 2024-02-20:
谢谢，已经收到你的邮件~~

yunhanw-google, 2024-02-20:
@xy371661665 have you used android chip-tool to reproduce this issue. I just use android chip-tool and light on/off level cluster fragment, and create the subscription with my matter outlet succesfully, toggle is fine, then unplug the outlet, press toggle in android chip-tool several times, then I see UI crash with java.lang.IllegalStateException: Unable to get connected device with nodeId 1, which seems to be matched with the code here, https://github.com/project-chip/connectedhomeip/blob/6b27278831f83c2bd151df8d948ab6c604a06b96/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/ChipClient.kt#L106

02-19 23:53:40.096  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.096  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.099  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.099  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.103  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.103  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.106  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.106  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.108  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.108  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.110  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.110  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.113  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.113  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.115  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.115  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.118  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.118  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.121  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.121  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.123  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.123  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.125  2087  2415 E ChipClient: Unable to get connected device with nodeId 1
02-19 23:53:40.125  2087  2415 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:559: CHIP Error 0x00000032: Timeout
02-19 23:53:40.148  1113  3095 I WifiHAL : Creating message to get link statistics; iface = 47
02-19 23:53:40.174  2087  2087 E AndroidRuntime: FATAL EXCEPTION: main
02-19 23:53:40.174  2087  2087 E AndroidRuntime: Process: com.google.chip.chiptool, PID: 2087
02-19 23:53:40.174  2087  2087 E AndroidRuntime: java.lang.IllegalStateException: Unable to get connected device with nodeId 1
02-19 23:53:40.174  2087  2087 E AndroidRuntime: 	at com.google.chip.chiptool.ChipClient$getConnectedDevicePointer$2$1.onConnectionFailure(ChipClient.kt:106)

closed 2024-02-21 by yunhanw-google

yunhanw-google, 2024-02-22:
@xy371661665 we just merge the fix, let me know whether it is still issue and feel free to reopen this bug. thanks

PR #32237 (2024-02-21): [Android] Add excetpion try-catch in getConnectedPointer
Fix #28430 

For avoid crash, I added try-catch in getConnectedPointer methods.