project-chip/connectedhomeip #28432: [BUG] pair device failed because of network unreachable

netscell, 2023-08-01:
### Reproduction steps

pair device use ble or ip failed;
the mdns resolve target device  ip  start with fe8;:

07-27 11:59:55.241 26466 26774 I NsdServiceFinderAndResolver: Resolved service '40C0C36C1E4E2A47-0000000000000020' to /fe80::32de:4bff:fedb:8aa7
07-27 11:59:55.242 26466 26774 D DIS     :  ----- entry [0] : T 1
07-27 11:59:55.242 26466 26774 D DIS     :  ----- entry [1] : SAI 300
07-27 11:59:55.242 26466 26774 D DIS     :  ----- entry [2] : SII 5000
07-27 11:59:55.243 26466 26774 D DIS     : Node ID resolved for 40C0C36C1E4E2A47:0000000000000020
07-27 11:59:55.243 26466 26774 D DIS     : 	Hostname: fe80::32de:4bff:
07-27 11:59:55.243 26466 26774 D DIS     : 	IP Address #1: fe80::32de:4bff:fedb:8aa7
07-27 11:59:55.243 26466 26774 D DIS     : 	Port: 5540
07-27 11:59:55.243 26466 26774 D DIS     : 	Mrp Interval idle: 5000 ms
07-27 11:59:55.243 26466 26774 D DIS     : 	Mrp Interval active: 300 ms
07-27 11:59:55.243 26466 26774 D DIS     : 	TCP Supported: 1
07-27 11:59:55.243 26466 26774 D DIS     : UDP:[fe80::32de:4bff:fedb:8aa7]:5540: new best score: 3
07-27 11:59:55.243 26466 26774 D DIS     : Checking node lookup status after 131 ms
07-27 11:59:55.243 26466 26774 D DIS     : Keeping DNSSD lookup active
07-27 11:59:55.312 26466 26896 D DIS     : Checking node lookup status after 11565 ms
07-27 11:59:55.312 26466 26896 D DIS     : Checking node lookup status after 200 ms
07-27 11:59:55.312 26466 26896 D DIS     : OperationalSessionSetup[1:0000000000000020]: Updating device address to UDP:[fe80::32de:4bff:fedb:8aa7]:5540 while in state 2
07-27 11:59:55.312 26466 26896 D DIS     : OperationalSessionSetup[1:0000000000000020]: State change 2 --> 3
07-27 11:59:55.313 26466 26896 D IN      : SecureSession[0x735aa99c08]: Allocated Type:2 LSID:15283
07-27 11:59:55.313 26466 26896 D SC      : Initiating session on local FabricIndex 1 from 0x000000000000000D -> 0x0000000000000020
07-27 11:59:55.345 26466 26896 D TOO     : KVS: Getting key f/1/g
07-27 11:59:55.345 26466 26896 D TOO     : KVS: Getting key f/1/k/0
07-27 11:59:55.346 26466 26896 D TOO     : KVS: Getting key f/1/s/0000000000000020
07-27 11:59:55.346 26466 26896 D KeyValueStoreManager: Key 'f/1/s/0000000000000020' not found in shared preferences
07-27 11:59:55.347 26466 26896 D EM      : <<< [E:25882i S:0 M:169514501] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
07-27 11:59:55.347 26466 26896 D IN      : (U) Sending msg 169514501 to IP address 'UDP:[fe80::32de:4bff:fedb:8aa7]:5540'
07-27 11:59:55.347 26466 26896 D IN      : SecureSession[0x735aa99c08]: Released - Type:2 LSID:15283
07-27 11:59:55.348 26466 26896 D CTL     : Device connection failed. Error ../../src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000065: Network is unreachable
07-27 11:59:55.348 26466 26896 D CTL     : Error on commissioning step 'FindOperational': '../../src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000065: Network is unreachable'
07-27 11:59:55.348 26466 26896 E CTL     : Failed to perform commissioning step 18
07-27 11:59:55.348 26466 26896 D CTL     : Going from commissioning step 'FindOperational' with lastErr = '../../src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000065: Network is unreachable' -> 'Cleanup'
07-27 11:59:55.348 26466 26896 D CTL     : Performing next commissioning step 'Cleanup' with completion status = '../../src/inet/UDPEndPointImplSockets.cpp:411: OS Error 0x02000065: Network is unreachable'
07-27 11:59:55.348 26466 26896 D CTL     : Successfully finished commissioning step 'Cleanup'

### Bug prevalence

3-7times a week

### GitHub hash of the SDK that was being used

NA

### Platform

android

bzbarsky-apple, 2023-08-01:
Sounds like the phone can't send a UDP message to `fe80::32de:4bff:fedb:8aa7`... This really depends on your networking setup, right?

netscell, 2023-08-13:
the network which used  phone ap and  wifi router  both have the unreachable problem;no extra setting for network;

bzbarsky-apple, 2023-08-28:
Well, can the relevant device (the one trying to send the packet) ping the relevant IP?  Because it looks like not....

netscell, 2023-08-29:
ping the matter device using LLA （fe80::32de:4bff:fedb:8aa7 or fe80::32de:4bff:fedb:8aa7%wlan0） is ok in the same wifi；

bzbarsky-apple, 2023-08-29:
@netscell Is that a ping _from the phone involved_?

netscell, 2023-11-21:
yes,after  failed,use ping tool android app  from the same phone