project-chip/connectedhomeip #28433: [BUG] PASE time out after unpair device

netscell, 2023-08-01:
### Reproduction steps

1.pair device complete
2.unpair device successfully
3.pair device with ip failed:
 08-01 12:21:28.978 28023 28045 D SPL     : JNI_OnLoad() called
 08-01 12:21:28.979 28023 28045 D CTL     : setUseJavaCallbackForNOCRequest() called
 08-01 12:21:28.979 28023 28045 D CTL     : setAttestationTrustStoreDelegate() called
 08-01 12:21:28.982 28023 28045 I  verifyWithIp: ip:192.168.3.12 port :5540 dis:333 pin:81683209
08-01 12:21:28.982 28023 28045 D CTL     : pairDeviceWithAddress() called
 08-01 12:21:28.982 28023 28045 D IN      : SecureSession[0x7b3d7b9118]: Allocated Type:1 LSID:61420
 08-01 12:21:28.982 28023 28045 D SC      : Assigned local session key ID 61420
08-01 12:21:28.983 28023 28045 D EM      : <<< [E:14236i S:0 M:186015562] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
08-01 12:21:28.983 28023 28045 D IN      : (U) Sending msg 186015562 to IP address 'UDP:192.168.3.12:5540'
08-01 12:21:28.983 28023 28045 D SC      : Sent PBKDF param request
 08-01 12:21:28.983 28023 28045 D CTL     : Setting attestation nonce from parameters
 08-01 12:21:28.983 28023 28045 D CTL     : Setting CSR nonce from parameters
 08-01 12:21:28.983 28023 28045 D CTL     : Commission called for node ID 0x0000000000000003
 08-01 12:21:28.993 28023 28399 D EM      : >>> [E:14236i S:0 M:148031423 (Ack:186015562)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
 08-01 12:21:28.993 28023 28399 D EM      : Found matching exchange: 14236i, Delegate: 0x7bbb7dd0e8
08-01 12:21:28.993 28023 28399 D EM      : Rxd Ack; Removing MessageCounter:186015562 from Retrans Table on exchange 14236i
08-01 12:21:35.210 28023 28399 E SC      : PASESession timed out while waiting for a response from the peer. Expected message type was 33
08-01 12:21:35.211 28023 28399 D IN      : SecureSession[0x7b3d7b9118]: Released - Type:1 LSID:61420

### Bug prevalence

high occurrence

### GitHub hash of the SDK that was being used

NA

### Platform

android

bzbarsky-apple, 2023-08-01:
@netscell Why would pairing over IP work after "unpair"?  Was a commmissioning window actually opened on the device?

netscell, 2023-08-02:
> @netscell Why would pairing over IP work after "unpair"? Was a commmissioning window actually opened on the device?

yes,  the matter device  can found througth mdns discovery

bzbarsky-apple, 2023-08-03:
> so i want to know what's the spec for the commissioning window status after unpair device

Generally speaking: unless that factory resets the device, removing the last fabric does not reopen a commissioning window on its own.  You typically need to at least power-cycle the device, or do whatever device interaction opens commissioning windows when it comes out of the box initially.

netscell, 2023-08-04:
> Generally speaking: unless that factory resets the device, removing the last fabric does not reopen a commissioning window on its own. You typically need to at least power-cycle the device, or do whatever device interaction opens commissioning windows when it comes out of the box initially.

after try to power-cycle the device ,the device commissioning window is also closed;

bzbarsky-apple, 2023-08-05:
@netscell Is this a device that would normally enter commissioning mode on power-up?  Not all device categories do that.

closed 2023-08-29 by netscell