project-chip/connectedhomeip #28435: [BUG] setup env Failed: run /scripts/activate.sh get fialed (requirements.esp32.txt)

resghst, 2023-08-01:
### Reproduction steps

I use docker image (debian:bullseye) to set up Matter env. Although success installed all [prerequisites](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md), run "scripts/activate.sh" get env setup failed. 
According to pip-requirements.log, installing "gevent" from scripts/setup/requirements.esp32.txt will get the error message.
If ignore install gdbgui==0.13.2.0 requirements.esp32.txt, activate can success building
# gdbgui==0.13.2.0 ; platform_machine != 'aarch64' and sys_platform == 'linux'

[pip-requirements.log](https://github.com/project-chip/connectedhomeip/files/12226629/pip-requirements.log)
[esp_requirement.txt](https://github.com/project-chip/connectedhomeip/files/12226631/esp_requirement.txt)

pip-requirements.log :
Given no hashes to check 1 links for project 'gevent': discarding no candidates
Collecting gevent<2.0,>=1.2.2 (from gdbgui==0.13.2.0->-r /var/chip/scripts/setup/requirements.esp32.txt (line 12))
Created temporary directory: /tmp/pip-unpack-uhfa2xsi
Downloading gevent-1.5.0.tar.gz (5.3 MB)
Added gevent<2.0,>=1.2.2 from https://files.pythonhosted.org/packages/5a/79/2c63d385d017b5dd7d70983a463dfd25befae70c824fedb857df6e72eff2/gevent-1.5.0.tar.gz (from gdbgui==0.13.2.0->-r /var/chip/scripts/setup/requirements.esp32.txt (line 12)) to build tracker '/tmp/pip-build-tracker-udvhdggl'

$ scripts/activate.sh

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (3m51.4s)
  Setting up Python environment.....Ignoring pyobjc-core: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyobjc-framework-cocoa: markers 'sys_platform == "darwin"' don't match your environment
Ignoring pyobjc-framework-corebluetooth: markers 'sys_platform == "darwin"' don't match your environment
Collecting pip-tools (from -r /var/chip/scripts/setup/requirements.txt (line 1))
  Downloading pip_tools-6.12.3-py3-none-any.whl (52 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 52.7/52.7 kB 487.5 kB/s eta 0:00:00
Collecting virtualenv (from -r /var/chip/scripts/setup/requirements.txt (line 2))
  Downloading virtualenv-20.20.0-py3-none-any.whl (8.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.7/8.7 MB 3.3 MB/s eta 0:00:00
Collecting click>=7.0 (from -r /var/chip/scripts/setup/requirements.esp32.txt (line 1))
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 96.6/96.6 kB 2.5 MB/s eta 0:00:00
Collecting future>=0.15.2 (from -r /var/chip/scripts/setup/requirements.esp32.txt (line 2))
  Downloading future-0.18.3.tar.gz (840 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 840.9/840.9 kB 4.6 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting pyparsing<2.4.0,>=2.0.3 (from -r /var/chip/scripts/setup/requirements.esp32.txt (line 3))
  Downloading pyparsing-2.3.1-py2.py3-none-any.whl (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.8/61.8 kB 1.6 MB/s eta 0:00:00
Collecting idf-component-manager (from -r /var/chip/scripts/setup/requirements.esp32.txt (line 4))
  Downloading idf_component_manager-1.2.2-py3-none-any.whl (116 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 116.7/116.7 kB 2.7 MB/s eta 0:00:00
Collecting pygdbmi<=0.9.0.2 (from -r /var/chip/scripts/setup/requirements.esp32.txt (line 5))
  Downloading pygdbmi-0.9.0.2-py3-none-any.whl (16 kB)
Collecting reedsolo<=1.5.4,>=1.5.3 (from -r /var/chip/scripts/setup/requirements.esp32.txt (line 6))
  Downloading reedsolo-1.5.4.tar.gz (271 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 271.3/271.3 kB 4.0 MB/s eta 0:00:00
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting bitstring>=3.1.6 (from -r /var/chip/scripts/setup/requirements.esp32.txt (line 7))
  Downloading bitstring-4.0.1-py3-none-any.whl (45 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 45.6/45.6 kB 835.3 kB/s eta 0:00:00
Collecting ecdsa>=0.16.0 (from -r /var/chip/scripts/setup/requirements.esp32.txt (line 8))
  Downloading ecdsa-0.18.0-py2.py3-none-any.whl (142 kB)
[... 194 lines cut ...]
  File "/var/chip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 296, in install
    pip_install(
  File "/var/chip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 262, in pip_install
    return _check_call(cmd)
  File "/var/chip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 88, in _check_call
    subprocess.check_call(args, **kwargs)
  File "/usr/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/var/chip/.environment/pigweed-venv/bin/python', '-m', 'pip', 'install', '--log', '/var/chip/.environment/pigweed-venv/pip-requirements.log', '--requirement=/var/chip/scripts/setup/requirements.txt', '--constraint=/var/chip/scripts/setup/constraints.txt']' returned non-zero exit status 1.

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

e059202613e91da78884aec4e1041d76da0af9bf

### Platform

python

### Platform Version(s)

docker, img: debian:bullseye

mmt-10, 2023-08-02:
I got same issue.

aDaisukeSakyo, 2023-08-04:
I ran into the same problem. To get around this problem, I have set python to 3.11 for now.

deveritec-rosc, 2023-08-04:
Same problem. I try to build a container using `connectedhomeip/chip-build-nrf-platform:0.7.25`. Also the stuff here in the gitlab container registry has the same problem.

Yinxq, 2023-09-04:
I also ran into this problem. I looked at the log file and it seems to be related to Cython. Cython was updated to 3.0 in July 23. Does this version conflict with gevent1.5?
`2023-09-01T21:45:22,983   Successfully installed Cython-3.0.2 cffi-1.15.1 greenlet-2.0.2 pycparser-2.21 setuptools-68.1.2 wheel-0.41.2
2023-09-01T21:45:23,175   Installing build dependencies: finished with status 'done'
2023-09-01T21:45:23,176   Getting requirements to build wheel: started
2023-09-01T21:45:23,176   Running command Getting requirements to build wheel
2023-09-01T21:45:23,760   warning: src/gevent/resolver/cares.pyx:32:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:23,760   warning: src/gevent/resolver/cares.pyx:33:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,092   Compiling src/gevent/resolver/cares.pyx because it changed.
2023-09-01T21:45:24,092   [1/1] Cythonizing src/gevent/resolver/cares.pyx
2023-09-01T21:45:24,230   warning: src/gevent/libev/corecext.pyx:325:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,242   warning: src/gevent/libev/corecext.pyx:778:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,242   warning: src/gevent/libev/corecext.pyx:780:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,242   warning: src/gevent/libev/corecext.pyx:782:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,242   warning: src/gevent/libev/corecext.pyx:786:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,691   warning: src/gevent/libev/corecext.pyx:1288:0: Global name __SYSERR_CALLBACK matched from within class scope in contradiction to to Python 'class private name' rules. This may change in a future release.
2023-09-01T21:45:24,692   warning: src/gevent/libev/corecext.pyx:1288:0: Global name __SYSERR_CALLBACK matched from within class scope in contradiction to to Python 'class private name' rules. This may change in a future release.
2023-09-01T21:45:24,743   warning: src/gevent/libev/corecext.pyx:1288:0: Global name __SYSERR_CALLBACK matched from within class scope in contradiction to to Python 'class private name' rules. This may change in a future release.
2023-09-01T21:45:24,747   warning: src/gevent/libev/corecext.pyx:1288:0: Global name __SYSERR_CALLBACK matched from within class scope in contradiction to to Python 'class private name' rules. This may change in a future release.

2023-09-01T21:45:24,772   Error compiling Cython file:
2023-09-01T21:45:24,772   ------------------------------------------------------------
2023-09-01T21:45:24,772   ...
2023-09-01T21:45:24,773           if events & ~(libev.EV__IOFDSET | libev.EV_READ | libev.EV_WRITE):
2023-09-01T21:45:24,773               raise ValueError('illegal event mask: %r' % events)
2023-09-01T21:45:24,773           # All the vfd_functions are no-ops on POSIX
2023-09-01T21:45:24,773           cdef int vfd = libev.vfd_open(fd)
2023-09-01T21:45:24,773           libev.ev_io_init(&self._watcher, <void *>gevent_callback_io, vfd, events)
2023-09-01T21:45:24,773           self.__watcher = <libev.ev_watcher*>&self._watcher
2023-09-01T21:45:24,773                            ^
2023-09-01T21:45:24,773   ------------------------------------------------------------

2023-09-01T21:45:24,773   src/gevent/libev/corecext.pyx:1027:25: Cannot convert 'ev_watcher *' to Python object

2023-09-01T21:45:24,773   Error compiling Cython file:
2023-09-01T21:45:24,773   ------------------------------------------------------------
2023-09-01T21:45:24,773   ...
2023-09-01T21:45:24,773               raise ValueError('illegal event mask: %r' % events)
2023-09-01T21:45:24,773           # All the vfd_functions are no-ops on POSIX
2023-09-01T21:45:24,773           cdef int vfd = libev.vfd_open(fd)
2023-09-01T21:45:24,773           libev.ev_io_init(&self._watcher, <void *>gevent_callback_io, vfd, events)
2023-09-01T21:45:24,773           self.__watcher = <libev.ev_watcher*>&self._watcher
2023-09-01T21:45:24,773           self.__ss = &io_ss
2023-09-01T21:45:24,773                       ^
2023-09-01T21:45:24,773   ------------------------------------------------------------

2023-09-01T21:45:24,774   src/gevent/libev/corecext.pyx:1028:20: Cannot convert 'start_and_stop *' to Python object

2023-09-01T21:45:24,776   Error compiling Cython file:
[... cut ...]
2023-09-01T21:45:25,030   [bold magenta]full command[/]: [blue]/home/bt4/HoneyComb/connectedhomeip/.environment/pigweed-venv/bin/python /home/bt4/HoneyComb/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py get_requires_for_build_wheel /tmp/tmp52z85b53[/]
2023-09-01T21:45:25,030   [bold magenta]cwd[/]: /tmp/pip-install-e5m9j1aa/gevent_e40587e89b5843858415c08f6abd4171
2023-09-01T21:45:25,030   Getting requirements to build wheel: finished with status 'error'
2023-09-01T21:45:25,031 ERROR: [present-rich] Getting requirements to build wheel exited with 1`

Yinxq, 2023-09-04:
I also ran into this problem. I looked at the log file and it seems to be related to Cython. Cython was updated to 3.0 in July 23, maybe this version conflicts with gevent1.5.
```
2023-09-01T21:45:22,983   Successfully installed Cython-3.0.2 cffi-1.15.1 greenlet-2.0.2 pycparser-2.21 setuptools-68.1.2 wheel-0.41.2
2023-09-01T21:45:23,175   Installing build dependencies: finished with status 'done'
2023-09-01T21:45:23,176   Getting requirements to build wheel: started
2023-09-01T21:45:23,176   Running command Getting requirements to build wheel
2023-09-01T21:45:23,760   warning: src/gevent/resolver/cares.pyx:32:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:23,760   warning: src/gevent/resolver/cares.pyx:33:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,092   Compiling src/gevent/resolver/cares.pyx because it changed.
2023-09-01T21:45:24,092   [1/1] Cythonizing src/gevent/resolver/cares.pyx
2023-09-01T21:45:24,230   warning: src/gevent/libev/corecext.pyx:325:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,242   warning: src/gevent/libev/corecext.pyx:778:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,242   warning: src/gevent/libev/corecext.pyx:780:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,242   warning: src/gevent/libev/corecext.pyx:782:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,242   warning: src/gevent/libev/corecext.pyx:786:0: The 'DEF' statement is deprecated and will be removed in a future Cython version. Consider using global variables, constants, and in-place literals instead. See https://github.com/cython/cython/issues/4310
2023-09-01T21:45:24,691   warning: src/gevent/libev/corecext.pyx:1288:0: Global name __SYSERR_CALLBACK matched from within class scope in contradiction to to Python 'class private name' rules. This may change in a future release.
2023-09-01T21:45:24,692   warning: src/gevent/libev/corecext.pyx:1288:0: Global name __SYSERR_CALLBACK matched from within class scope in contradiction to to Python 'class private name' rules. This may change in a future release.
2023-09-01T21:45:24,743   warning: src/gevent/libev/corecext.pyx:1288:0: Global name __SYSERR_CALLBACK matched from within class scope in contradiction to to Python 'class private name' rules. This may change in a future release.
2023-09-01T21:45:24,747   warning: src/gevent/libev/corecext.pyx:1288:0: Global name __SYSERR_CALLBACK matched from within class scope in contradiction to to Python 'class private name' rules. This may change in a future release.

2023-09-01T21:45:24,772   Error compiling Cython file:
2023-09-01T21:45:24,772   ------------------------------------------------------------
2023-09-01T21:45:24,772   ...
2023-09-01T21:45:24,773           if events & ~(libev.EV__IOFDSET | libev.EV_READ | libev.EV_WRITE):
2023-09-01T21:45:24,773               raise ValueError('illegal event mask: %r' % events)
2023-09-01T21:45:24,773           # All the vfd_functions are no-ops on POSIX
2023-09-01T21:45:24,773           cdef int vfd = libev.vfd_open(fd)
2023-09-01T21:45:24,773           libev.ev_io_init(&self._watcher, <void *>gevent_callback_io, vfd, events)
2023-09-01T21:45:24,773           self.__watcher = <libev.ev_watcher*>&self._watcher
2023-09-01T21:45:24,773                            ^
2023-09-01T21:45:24,773   ------------------------------------------------------------

2023-09-01T21:45:24,773   src/gevent/libev/corecext.pyx:1027:25: Cannot convert 'ev_watcher *' to Python object

2023-09-01T21:45:24,773   Error compiling Cython file:
2023-09-01T21:45:24,773   ------------------------------------------------------------
2023-09-01T21:45:24,773   ...
2023-09-01T21:45:24,773               raise ValueError('illegal event mask: %r' % events)
2023-09-01T21:45:24,773           # All the vfd_functions are no-ops on POSIX
2023-09-01T21:45:24,773           cdef int vfd = libev.vfd_open(fd)
2023-09-01T21:45:24,773           libev.ev_io_init(&self._watcher, <void *>gevent_callback_io, vfd, events)
2023-09-01T21:45:24,773           self.__watcher = <libev.ev_watcher*>&self._watcher
2023-09-01T21:45:24,773           self.__ss = &io_ss
2023-09-01T21:45:24,773                       ^
2023-09-01T21:45:24,773   ------------------------------------------------------------

2023-09-01T21:45:24,774   src/gevent/libev/corecext.pyx:1028:20: Cannot convert 'start_and_stop *' to Python object

2023-09-01T21:45:24,776   Error compiling Cython file:
[... cut ...]
2023-09-01T21:45:25,030   [bold magenta]full command[/]: [blue]/home/bt4/HoneyComb/connectedhomeip/.environment/pigweed-venv/bin/python /home/bt4/HoneyComb/connectedhomeip/.environment/pigweed-venv/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py get_requires_for_build_wheel /tmp/tmp52z85b53[/]
2023-09-01T21:45:25,030   [bold magenta]cwd[/]: /tmp/pip-install-e5m9j1aa/gevent_e40587e89b5843858415c08f6abd4171
2023-09-01T21:45:25,030   Getting requirements to build wheel: finished with status 'error'
2023-09-01T21:45:25,031 ERROR: [present-rich] Getting requirements to build wheel exited with 1
```

Yinxq, 2023-09-04:
> I also ran into this problem. I looked at the log file and it seems to be related to Cython. Cython was updated to 3.0 in July 23, maybe this version conflicts with gevent1.5.
>
> ``` [...]

I've tried a few ways to lower the Cython version to 0.29.36, but with no success.

Yinxq, 2023-09-04:
I've tried a few ways to lower the Cython version to 0.29.36, but with no success.

TPVChethan, 2024-05-29:
> I also ran into this problem. I looked at the log file and it seems to be related to Cython. Cython was updated to 3.0 in July 23, maybe this version conflicts with gevent1.5.
>
> ``` [...]

We also ran into same issue, even after installing all the relevant Python packages.
Could somebody share any workaround or any fixes to build v1.1.0.1 connected homeIp source code on Ubuntu 22.04?

closed 2024-07-11 by resghst

kagami555, 2025-04-18:
I have encountered the same problem and I would like to ask how you solved it. Thank you very much