project-chip/connectedhomeip #28436: [CERT-TEST-FAILURE]  KeySetWrite command does not return RESOURCE_EXHAUSTED instead returns FAILURE

sumaky, 2023-08-01:
### Feature Area

Other

### Test Case

TC-GRPKEY-2.1

### Reproduction steps

**The below conditions were tried to get   RESOURCE_EXHAUSTED  condition as per section 11.2.7.1. KeySetWrite Command**

1. After paring KeySetReadAllIndices command returns one entry groupKeySetID: 0  by default
KeySetReadAllIndicesResponse: {
[1690878193.679729][43677:43680] CHIP:TOO:     groupKeySetIDs: 1 entries
[1690878193.679741][43677:43680] CHIP:TOO:       [1]: 0
[1690878193.679749][43677:43680] CHIP:TOO:    }
**Is this expected?**

2. max-group-keys-per-fabric returns a value of 3 so attempted to add 3 entries with groupKeySetID's as 419, 420 and 421
but when KeySetWrite Command was given with groupKeySetID as 421 we see a FAILURE response. 

4. KeySetReadAllIndices returns  3 entries as 420,491, 0

6. Tried to remove groupKeySetID : 0  we see Status SUCCESS but following command like KeySetReadAllIndices does not get a valid response.
[GRPKEY2.2Error.txt](https://github.com/project-chip/connectedhomeip/files/12226630/GRPKEY2.2Error.txt)

### Bug prevalence

during manual execution of TC-GRPKEY-2.2

### GitHub hash of the SDK that was being used

NA

### Platform

other

tcarmelveilleux, 2023-08-01:
1) is an issue of lack of compliance.

4) is normal: if you remove GroupKeySet0, you are breaking the IPK and won't be able to communicate anymore. DO NOT remove KeySet index 0.

sumaky, 2023-08-02:
@tcarmelveilleux If GroupKeySet0 is going to exist by default should we change test plan -TC- GRPKEY-2.2 Step 16 to add keys upto  max-group-keys-per-fabric  -1 ?

tcarmelveilleux, 2023-08-02:
Yes, that step is going to lead to issues. 

>  {PICS_S}.C00.Rsp(KeySetWrite) | TH sends KeySetWrite command to DUT Max_GrpKey times, starting with GroupKeySetID 1 and incrementing by 1 each time.
-- | --

Suggest:

> {PICS_S}.C00.Rsp(KeySetWrite) | TH sends KeySetWrite command to DUT until size of list for TH's fabric in GroupKeyMap attribute contains Max_GrpKey entries, starting with GroupKeySetID 1 and incrementing by 1 each time.
-- | --

PR #28456 (2023-08-03): Fix RESOURCE_EXHAUSTED in KeySetWrite
Problem:

- FAILURE was returned instead of RESOURCE_EXHAUSTED status when trying to add one keyset too many.
- Fixes #28436

This PR:

- Correctly returns the error on failure to write the key set due to exhaustion of memory resources.

Testing done:

- Added integration test, regenerated YAML tests.
closed 2023-08-03 by mergify[bot]