project-chip/connectedhomeip #28437: [BUG] Extended Advertisement records leaked with Thread devices

bluebin14, 2023-08-01:
### Reproduction steps

1. Factory reset a BLE+Thread device that A) uses extended advertising and B) clears its Thread credentials when there are no more fabrics
2. Pair the device e.g. with chip-tool
3. Remove all fabrics
4. Examine the mdns advertisements for this device, there should be none but there is one

### Bug prevalence

always

### GitHub hash of the SDK that was being used

e059202613e91da78884aec4e1041d76da0af9bf

### Platform

nrf

### Platform Version(s)

All

### Anything else?

With Extended Discovery enabled, when removing the last fabric the SDK immediately begins extended advertising. In case of a spec compliant BLE+Thread device that clears its Thread credentials when there are no more fabrics this results in a leaked/dangling advertisement that does not correspond to any device. This advertisement sticks around for 2 hours.

Should be fixed at SDK level to avoid manufacturer workarounds.

bzbarsky-apple, 2023-08-01:
It sounds like we need to order the "begins extended advertising" to after the Thread credentials have been cleared....  Where does that "begins extended advertising" code live if you happen to know @bluebin14 ?

bzbarsky-apple, 2023-08-01:
> is handled entirely by the SDK,

Yes, but I was asking where.

> if there was some doNotStartExtendedAdvertisingAfterLastFabricIsRemoved setting

What's the point of extended advertising, then?  Put another way: why is this device enabling the extended advertising at all?

bluebin14, 2023-08-02:
The code is in Dnssd.cpp. The extended advertising is an Amazon Works with Alexa requirement/recommendation.  I agree it conflicts with this scenario and the requirement should be removed. Nevertheless it would still be applicable while there is at least one fabric present.