project-chip/connectedhomeip #28444: [BUG] - Unable to Pair Light-App using ChipTool

smishra-volansys, 2023-08-01:
### Reproduction steps

We are trying to test the Multi-Admin scenario for which we are using Matter Controller as Chip_tool and a PC as the Light App.
Please find the Steps below:
1) Run the chip-lighting-app on a PC having ble and wifi support.
2) Using the Chip-tool command try to commission the device using the following command:
chip-tool pairing ble-wifi <Node> <Wifi creds ssid+psk> 20202021 3840 <paa and cd certificates >
3) Once the Command has been fired the pairing gets stuck at a point before the ConnectDevice Successful instance. 

Attaching the Logs for a scenario where i used 2 controllers running light app in one and chip-tool in another. 
[chip-tool-pairing-with-PC.log](https://github.com/project-chip/connectedhomeip/files/12229221/chip-tool-pairing-with-PC.log)
 
[Uploading chip-tool.log…]()
d is fired from the controller, the ConnectDevice is successful but the device fails in WifiNetworkEnable. 

[chip-light-app.log](https://github.com/project-chip/connectedhomeip/files/12229161/chip-light-app.log)

### Bug prevalence

Frequent

### GitHub hash of the SDK that was being used

v1.1-branch

### Platform

other

### Platform Version(s)

v1.1-branch

bzbarsky-apple, 2023-08-01:
@smishra-volansys chip-tool-pairing-with-PC.log seems to be very much truncated, so hard to say what's going on there, other than inability to even connect over BLE.

chip-light-app.log shows some commissioner talking to the app over BLE, it sends ConnectNetwork, then the device keeps trying to do stuff with `wpa_supplicant` and failing.  I don't know which network you are trying to join and whether you have the right crendentials for it, but that's clearly the part that's failing.  You probably want to debug that part.

smishra-volansys, 2023-08-02:
Hi @bzbarsky-apple, 
The Chip-tool-with-PC logs are right from the command fire till the last, as it shows the device being discovered, 05:BC is the mac addr of the PC. 
at this time, the light app was running in the PC and chip-tool was running on our controller. 

Also, talking about the Network credentials, I have verified it thoroughly that we are using the right credentials for the Wifi Networks that we have available.

smishra-volansys, 2023-08-29:
Hi @bzbarsky-apple, We are still facing issues with pairing the Lighting app in the v1.1 release. Upon multiple tries, we observed that the example apps are giving us an error due to the certificates, but are able to pair the 3rd party devices out of the box.
Hence Closing this.

bzbarsky-apple, 2023-08-29:
> Upon multiple tries, we observed that the example apps are giving us an error due to the certificates

What error?  Things should be working here.  If they are not, please file an issue with complete logs from both sides.

smishra-volansys, 2023-09-08:
Errors 601 and 603 only, though after some tries we were able to pair the devices. 
We were facing an issue due to the version difference between the chip-tool and the lighting app. 
Our Chip-tool was built from branch v1.1 and the lighting app was from v1.0 which we think might have created the issue

bzbarsky-apple, 2023-09-08:
That's odd.  1.1 controllers should obviously be able to pair 1.0 devices...

If 601 and 603 refer to the list at https://github.com/project-chip/connectedhomeip/blob/5ae3c657a8f17582d1af0c8abc9a8642c7185ad2/src/credentials/attestation_verifier/DeviceAttestationVerifier.h#L69-L71 ... then I would not expect those errors to happen just because of the version difference.