project-chip/connectedhomeip #28463: [1.0] CASE session establishment fails after sending Sigma1

vineelasmile, 2023-08-01:
### Reproduction steps

We are developing on top of Matter SDK v1.0 attempting to work with ESP devkit. 
1. Using Android chip-tool logic pair with ESP lighting app. 

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

1.0 head

### Platform

android

### Type

Platform Issue

### Anything else?

From the controller(android) , here are the relavent logs when CASE session establishment starts:
```
08-01 23:31:50.056 19571 19616 D DIS     : Node ID resolved for A377CD260F42F699:0000000000000001
08-01 23:31:50.056 19571 19616 D DIS     : 	Hostname: espressif.lan
08-01 23:31:50.056 19571 19616 D DIS     : 	IP Address #1: 192.168.86.35
08-01 23:31:50.056 19571 19616 D DIS     : 	Port: 5540
08-01 23:31:50.056 19571 19616 D DIS     : 	Mrp Interval idle: not present
08-01 23:31:50.056 19571 19616 D DIS     : 	Mrp Interval active: not present
08-01 23:31:50.056 19571 19616 D DIS     : 	TCP Supported: 1
08-01 23:31:50.056 19571 19616 D DIS     : Lookup clearing interface for non LL address
08-01 23:31:50.056 19571 19616 D DIS     : UDP:192.168.86.35:5540: new best score: 2
08-01 23:31:50.056 19571 19616 D DIS     : Checking node lookup status after 1014 ms
08-01 23:31:50.056 19571 19616 D DIS     : OperationalSessionSetup[1:0000000000000001]: Updating device address to UDP:192.168.86.35:5540 while in state 2
08-01 23:31:50.057 19571 19616 D CTL     : OperationalSessionSetup[1:0000000000000001]: State change 2 --> 3
08-01 23:31:50.057 19571 19616 D IN      : SecureSession[0xb40000732fcb8368]: Allocated Type:2 LSID:64894
08-01 23:31:50.057 19571 19616 D SC      : Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000000001
08-01 23:31:50.078 19571 19616 D SC      : Including MRP parameters
08-01 23:31:50.079 19571 19616 D EM      : <<< [E:22343i M:24509042] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
08-01 23:31:50.079 19571 19616 D IN      : (U) Sending msg 24509042 to IP address 'UDP:192.168.86.35:5540'
08-01 23:31:50.079 19571 19616 D SC      : Sent Sigma1 msg
08-01 23:31:50.079 19571 19616 D CTL     : OperationalSessionSetup[1:0000000000000001]: State change 3 --> 4
```

On the devkit:
```
I (45034) chip[DIS]: Advertise operational node A377CD260F42F699-0000000000000001
I (45044) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: A377CD260F42F699-0000000000000001.
I (45054) chip[DIS]: mDNS service published: _matter._tcp
I (45234) wifi:<ba-add>idx:1 (ifx:0, 38:8b:59:d7:95:0d), tid:0, ssn:0, winSize:64
I (45234) chip[EM]: >>> [E:22343r S:0 M:24509042] (U) Msg RX from 0:29504484862350DE [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (45254) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fc9bc78
I (45264) chip[IN]: CASE Server disabling CASE session setups
I (45264) chip[EM]: <<< [E:22343r S:0 M:159881757 (Ack:24509042)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (45274) chip[IN]: (U) Sending msg 159881757 to IP address 'UDP:192.168.86.28%st1:5541'
I (45284) chip[SC]: Received Sigma1 msg
I (45294) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0x0000000000000001
I (45814) chip[EM]: <<< [E:22343r S:0 M:159881758 (Ack:24509042)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:31 (SecureChannel:CASE_Sigma2)
I (45824) chip[IN]: (U) Sending msg 159881758 to IP address 'UDP:192.168.86.28%st1:5541'
I (45824) chip[SC]: Sent Sigma2 msg
I (45824) chip[DL]: IP_EVENT_GOT_IP6
I (45834) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:5acf:79ff:fe1e:e098
I (45854) app_main: Interface IP Address changed
I (45854) chip[DIS]: Updating services using commissioning mode 0
I (45854) chip[DIS]: CHIP minimal mDNS started advertising.
I (45874) chip[DIS]: Advertise operational node A377CD260F42F699-0000000000000001
I (45884) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: A377CD260F42F699-0000000000000001.
I (45894) chip[DIS]: mDNS service published: _matter._tcp
E (45894) chip[SVR]: Server initialization complete
I (45904) chip[DIS]: Updating services using commissioning mode 0
I (45914) chip[DIS]: CHIP minimal mDNS started advertising.
I (45934) chip[DIS]: Advertise operational node A377CD260F42F699-0000000000000001
I (45934) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: A377CD260F42F699-0000000000000001.
I (45944) chip[DIS]: mDNS service published: _matter._tcp
I (45954) chip[IM]: No subscriptions to resume
I (46204) chip[IN]: (U) Sending msg 159881758 to IP address 'UDP:192.168.86.28%st1:5541'
I (46604) chip[IN]: (U) Sending msg 159881758 to IP address 'UDP:192.168.86.28%st1:5541'
I (47244) chip[IN]: (U) Sending msg 159881758 to IP address 'UDP:192.168.86.28%st1:5541'
I (48134) chip[IN]: (U) Sending msg 159881758 to IP address 'UDP:192.168.86.28%st1:5541'
E (49634) chip[EM]: Failed to Send CHIP MessageCounter:159881758 on exchange 22343r sendCount: 4 max retries: 4
E (59074) chip[BLE]: ack recv timeout, closing ep 0x3fca2160
I (59074) chip[BLE]: Releasing end point's BLE connection back to application.
I (63014) ROUTE_HOOK: Ignore invalid ICMP packet
E (80034) chip[SC]: CASESession timed out while waiting for a response from the peer. Current state was 2
E (80034) chip[IN]: CASE Session establishment failed: 32
I (80044) chip[IN]: CASE Server enabling CASE session setups
I (98164) chip[FS]: Fail-safe timer expired
E (98164) chip[SVR]: Failsafe timer expired
I (98164) chip[SC]: SecureSession[0x3fc9a3c0]: Moving from state 'kActive' --> 'kPendingEviction'
E (98174) chip[SVR]: Commissioning failed (attempt 1): 32
I (98184) chip[DIS]: Updating services using commissioning mode 1
I (98184) chip[DIS]: CHIP minimal mDNS started advertising.
I (98194) chip[DIS]: Advertise operational node A377CD260F42F699-0000000000000001
I (98194) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: A377CD260F42F699-0000000000000001.
I (98214) chip[DIS]: mDNS service published: _matter._tcp
I (98214) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (98224) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 365AD393A71F97FC.
I (98254) chip[DIS]: mDNS service published: _matterc._udp
E (98254) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
E (98254) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
E (98264) chip[FP]: Reverting pending fabric data for fabric 0x1
E (98284) chip[FP]: Warning: metadata not found during delete of fabric 0x1
I (98294) chip[FP]: Fabric (0x1) deleted.
I (98294) chip[ZCL]: OpCreds: Fabric index 0x1 was removed
I (98294) chip[DIS]: Updating services using commissioning mode 1
I (98304) chip[DIS]: CHIP minimal mDNS started advertising.
I (98334) chip[DIS]: Advertise commission parameter vendorID=65521 productID=32768 discriminator=3840/15 cm=1
I (98334) chip[DIS]: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 365AD393A71F97FC.
I (98354) chip[DIS]: mDNS service published: _matterc._udp
I (98374) chip[TS]: Pending Last Known Good Time: 2023-06-12T05:01:25
I (98374) chip[TS]: Previous Last Known Good Time: 2023-06-12T05:01:25
I (98384) chip[TS]: Reverted Last Known Good Time to previous value
E (98394) chip[FP]: Warning: metadata not found during delete of fabric 0x1
E (98394) chip[ZCL]: OpCreds: failed to delete fabric at index 1: d8
I (98404) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x00000030's Attribute 0x00000000 is 0 **********
I (98414) app_main: Commissioning failed, fail safe timer expired
I (98414) chip[FS]: Fail-safe cleanly disarmed
I (98434) app_main: Commissioning window opened
I (98434) app_main: Fabric will be removed
I (98444) app_main: Fabric removed successfully
I (98444) app_main: Fabric will be removed

```

As seen in the logs, Devkit claims to have sent the sigma2 message, however the controller has'nt received it. Is this a known issue in the branch ? 
* Any pointers to further root cause the problem ?
* Does the log message 'E (49634) chip[EM]: Failed to Send CHIP MessageCounter:159881758 on exchange 22343r sendCount: 4 max retries: 4
' mean that the device put the pkt on the radio ? Or it failed in some other way ?

jzwyssig, 2023-08-02:
Same problem here with connectedhomeip v1.1.0.1-1004-g5b4f800466:

```
I (340055) chip[DMG]: Registered a ReadHandler that will schedule a report between system Timestamp: 340057 and system Timestamp 340057.
I (340065) chip[IM]: Resuming subscriptionId 2607589860
I (340265) chip[DIS]: Checking node lookup status after 200 ms
I (340575) chip[DIS]: UDP:192.168.188.156%en1:5540: new best score: 2  <--- THIS IS GOOGLE NEST HUB
I (340575) chip[DIS]: UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121%en1]:5540: new best score: 4
I (340585) chip[DIS]: UDP:[2001:1708:2580:DD01:215F:5313:96E5:B383%en1]:5540: score has not improved: 4
I (340595) chip[DIS]: UDP:[2001:1708:2580:DD01:313E:CF5E:3D31:90D0%en1]:5540: score has not improved: 4
I (340605) chip[DIS]: UDP:[2001:1708:2580:DD01:7DA1:14F9:A889:64AE%en1]:5540: score has not improved: 4
I (340615) chip[DIS]: Checking node lookup status after 545 ms
I (340615) chip[SC]: Initiating session on local FabricIndex 1 from 0x000000001D16B0FB -> 0x00000000E34EC3E0
I (340685) chip[EM]: <<< [E:52478i S:0 M:73105355] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
I (340685) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
I (340695) chip[SC]: Sent Sigma1 msg
I (341045) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'  <--- THIS IS GOOGLE NEST HUB
I (341405) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
I (342005) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
E (342015) chip[EM]: Ignoring transient send error: 3000001 on exchange 52478i
I (342905) chip[IN]: (U) Sending msg 73105355 to IP address 'UDP:[2001:1708:2580:DD01:104E:1640:A26B:D121]:5540'
E (342905) chip[EM]: Ignoring transient send error: 3000001 on exchange 52478i
E (344525) chip[EM]: Failed to Send CHIP MessageCounter:73105355 on exchange 52478i sendCount: 4 max retries: 4
E (346905) chip[SC]: CASESession timed out while waiting for a response from the peer. Current state was 4
E (346905) chip[DMG]: Failed to establish CASE for subscription-resumption with error '32'
```

vineelasmile, 2023-08-14:
I have validated with tcpdump that there are packets coming from the devkit to the android device controller. however these dont seem to make it up the application on the controller. How can i debug this further ?

vineelasmile, 2023-09-08:
The problem has been root caused to not starting up IO thread that processes the events.

closed 2023-09-08 by vineelasmile

uascent, 2023-11-17:
@vineelasmile  hi guy, I run into the same problem, what do you mean don't start up IO Thread that process the events, I don't catch that , can you provide more detail for me?

MrLiuYunPing, 2023-11-28:
Resetting network of the android phone can resolve the problem. But why?

marcokhoo, 2024-12-16:
i am using Android with Matter SDK1.3 for commissioning and faced this similar problem some of the time. By rebooting the phone and /or wifi router the issue will be gone. Can anyone share what is the root cause and how to fix it? 

**APP log:** 
17:49:49.929  E  Failed to Send CHIP MessageCounter:25176658 on exchange 6916i sendCount: 4 max retries: 4
17:49:52.151  D  null.onCharacteristicChanged: 18ee2ef5-263d-4559-959f-4f9c429f9d12
17:49:54.655  D  Received SendWriteRequest
17:49:54.735  D  null.onCharacteristicWrite: 18ee2ef5-263d-4559-959f-4f9c429f9d11 -> 0
17:49:55.668  I  onMethodCall() -> method=discoverCommissionableNodes, arguments={fabricId=FAB909644871269F}
17:49:55.670  D  Starting service discovery for '_matterc._udp' with callbackHandle: 486471667968
17:49:55.674  I  Started service '_matterc._udp'
17:49:55.681  I  Found service '138DC0C25BA49FCE'
17:49:55.681  I  Found service '420729C9E124DCB9'
17:49:57.291  E  CASESession timed out while waiting for a response from peer <DEDEDEDE52808217, 8>. Current state was 1
17:49:57.292  D  SecureSession[0xb40000722d54fa70]: Released - Type:2 LSID:39992
17:49:57.292  D  OperationalSessionSetup[8:DEDEDEDE52808217]: State change 4 --> 2
17:49:57.292  D  OperationalSessionSetup[8:DEDEDEDE52808217]: State change 2 --> 1
17:49:57.292  D  OperationalSessionSetup:attempts done: 1, attempts left: 2, retry delay 1s, status src/system/SystemLayerImplSelect.cpp:214: Success
17:49:57.292  D  OperationalSessionSetup[8:DEDEDEDE52808217]: State change 1 --> 6
17:49:57.292  E  Session establishment failed for <DEDEDEDE52808217, 8>, error: src/protocols/secure_channel/CASESession.cpp:599: CHIP Error 0x00000032: Timeout.  Next retry expected to get a response to Sigma1 or fail within 11 seconds
17:49:57.292  D  Skipping arming failsafe: new time (71 seconds from now) before old time (861 seconds from now)

**Device log (ESP matter: ICD_APP):**
I (270814) chip[EM]: >>> [E:6917r S:0 M:25176659] (U) Msg RX from 0:1B87127A5F8F137E [0000] to 0000000000000000 --- Type 0000:30 (SecureChannel:CASE_Sigma1) (B:196)
I (270866) chip[EM]: <<< [E:6917r S:0 M:250376573 (Ack:25176659)] (U) Msg TX from 0000000000000000 to 0:1B87127A5F8F137E [0000] [UDP:[FD0E:9B1F:4691:66F9:ABC8:CA2E:3728:628B]:52781] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
I (349372) chip[EM]: >>> [E:6918r S:0 M:25176660] (U) Msg RX from 0:40270046C6F28D40 [0000] to 0000000000000000 --- Type 0000:30 (SecureChannel:CASE_Sigma1) (B:196)
I (349424) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x4081b530
I (349427) chip[EM]: <<< [E:6918r S:0 M:250376574 (Ack:25176660)] (U) Msg TX from 0000000000000000 to 0:40270046C6F28D40 [0000] [UDP:[FD0E:9B1F:4691:66F9:ABC8:CA2E:3728:628B]:52781] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
I (349448) chip[SC]: Received Sigma1 msg
I (349461) chip[SC]: CASE matched destination ID: fabricIndex 1, NodeID 0xDEDEDEDE52808217
I (349528) chip[EM]: <<< [E:6918r S:0 M:250376575 (Ack:25176660)] (U) Msg TX from 0000000000000000 to 0:40270046C6F28D40 [0000] [UDP:[FD0E:9B1F:4691:66F9:ABC8:CA2E:3728:628B]:52781] --- Type 0000:31 (SecureChannel:CASE_Sigma2) (B:855)
I (349544) chip[EM]: ??1 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000 in 1382ms [State:Active II:500 AI:300 AT:4000]
I (349553) chip[SC]: Sent Sigma2 msg
I (350932) chip[EM]: <<1 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000
I (350984) chip[EM]: ??2 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000 in 1380ms [State:Active II:500 AI:300 AT:4000]
I (352375) chip[EM]: <<2 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000
I (352427) chip[EM]: ??3 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000 in 1555ms [State:Active II:500 AI:300 AT:4000]
I (353991) chip[EM]: <<3 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000
I (354043) chip[EM]: ??4 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000 in 1876ms [State:Idle II:500 AI:300 AT:4000]
I (355930) chip[EM]: <<4 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000
I (355981) chip[EM]: ??5 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000 in 2471ms [State:Idle II:500 AI:300 AT:4000]
I (356192) chip[EM]: >>> [E:6918r S:0 M:25176660] (U) Msg RX from 0:40270046C6F28D40 [0000] to 0000000000000000 --- Type 0000:30 (SecureChannel:CASE_Sigma1) (B:196)
I (356202) chip[EM]: <<< [E:6918r S:0 M:250376576 (Ack:25176660)] (U) Msg TX from 0000000000000000 to 0:40270046C6F28D40 [0000] [UDP:[FD0E:9B1F:4691:66F9:ABC8:CA2E:3728:628B]:52781] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
E (358465) chip[EM]: <<5 [E:6918r S:0 M:250376575] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)

leafkuang, 2025-06-25:
It was a network problem. I set up the network and the problem was solved.