project-chip/connectedhomeip #28472: [Test Failed] Test_TC_DRLK_2_12 is racy

bzbarsky-apple, 2023-08-02:
### Test issue(s)

See the log at https://github.com/project-chip/connectedhomeip/actions/runs/5733155723/job/15537384607?pr=28423

It's running this part of the YAML (irrelevant things dropped):

```
    - label: "Step 16: TH reads the AutoRelockTime attribute from the DUT"
      PICS: DRLK.S.A0023 && PICS_SDK_CI_ONLY
      command: "readAttribute"
      attribute: "AutoRelockTime"
      response:
          value: 10
...
    - label:
          "Step 17: TH sends the Unbolt Door command to the DUT with valid
          PINCode"
      PICS: DRLK.S.C27.Rsp
      command: "UnboltDoor"
      timedInteractionTimeoutMs: 1000
      arguments:
          values:
              - name: "PINCode"
                value: "123456"
....
    - label: "Wait for AutoRelockTime Expires"
      cluster: "DelayCommands"
      command: "WaitForMs"
      PICS: DRLK.S.A0023 && PICS_SDK_CI_ONLY
      arguments:
          values:
              - name: "ms"
                value: 10000
...
    - label: "Step 18: TH reads LockState attribute"
      PICS: DRLK.S.A0000 && DRLK.S.C01.Rsp
      command: "readAttribute"
      attribute: "LockState"
      response:
          value: 1
```

The value it got was 2, not 1.  Looking at the app-side logs, we have:

```
ERROR   02:05:14.602 - APP  OUT  : [1690941914602] [14523:100853] [ZCL] Lock App: specified PIN code was found in the database, setting door lock state to "Unlocked" [endpointId=1,userIndex=0]
...
ERROR   02:05:14.603 - APP  OUT  : [1690941914603] [14523:100853] [EM] <<< [E:36638r S:59104 M:154600274 (Ack:196101412)] (S) Msg TX to 1:000000000001B669 [6AF1] --- Type 0001:09 (IM:InvokeCommandResponse)
...
ERROR   02:05:24.604 - APP  OUT  : [1690941924604] [14523:100853] [DMG] Reading attribute: Cluster=0x0000_0101 Endpoint=1 AttributeId=0x0000_0000 (expanded=0)
...
ERROR   02:05:24.619 - APP  OUT  : [1690941924618] [14523:100853] [ZCL] Door Auto relock timer expired. Locking...
```

So the read came in 14ms before the timer expired.  In terms of the ordering, the lock sends the response, then starts the timer.  If it lost the timeslice for a bit between those points in time, the timer might not have been started until _after_ the timer on the client was started.   And since they are set for the same amount of time, that fails the test.

@mmarc for CI we should probably just add some slop to the 10s timer on the client (e.g. set it for 11s).  For the non-CI case, does the test plan require exact timing here?  Seems a bit odd... especially for a real lock that might take time to re-lock after its timer fires.

### Platform

core (please add to version below)

mmarc, 2023-08-02:
@bzbarsky-apple the [test plan](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/door_lock_Cluster.adoc#tc-drlk-2-12-verification-for-unbolt-door-command-dut-server) says
> TH reads LockState attribute after AutoRelockTime Expires and the lock action has been completed (e.g. motor movement finished)

So with real locks automatic testing is tricky but agree, for CI we can just extend that.
I guess that also affects at least TC-DRLK-2.3 as TC-DRLK-2.12 is a copy of TC-DRLK-2.3 just Unlock replaced by Unbolt.

bzbarsky-apple, 2023-08-03:
@mmarc I think the test plan should add some slop for clocks running at slightly different rates and things like the issue above...  Not to mention the bolt movement for a real lock.

woody-apple, 2024-04-03:
Closing older issues, trying to focus on current build failures

closed 2024-04-03 by woody-apple