project-chip/connectedhomeip #28491: Missing TemperatureSetpointHold and TemperatureSetpointHoldDuration [Android]

MatteoSomensi, 2023-08-03:
### Reproduction steps

On chip.devicecontroller.ThermostatCluster attributes TemperatureSetpointHold and TemperatureSetpointHoldDuration and related methods to read/write are missing 

### Platform

android

### Type

Manually tested with SDK

MatteoSomensi, 2023-08-03:
Printing the list of attributes of the ThermostatCluster the 0x023 and 0x024 are not found. Instead the iOS SDK finds them (and of course has the relative methods on the ThermostatCluster to read/write)

joonhaengHeo, 2023-08-08:
I found this IDs.
https://github.com/project-chip/connectedhomeip/blob/367a0c672c18e0137e0162c15b78adbcb039a33a/src/controller/java/generated/java/chip/devicecontroller/ClusterIDMapping.java#L8948

And, I found below methods.
https://github.com/project-chip/connectedhomeip/blob/367a0c672c18e0137e0162c15b78adbcb039a33a/src/controller/java/zap-generated/chip/devicecontroller/ChipClusters.java#L21725

yunhanw-google, 2023-11-08:
it is supported, if there still issues, please reopen this ticket, thanks

closed 2023-11-08 by yunhanw-google