project-chip/connectedhomeip #28492: Is there a generic version of chip-tool?

chuckrui, 2023-08-03:
As far as I know, it seems that each version of each manufacturer uses the corresponding chip-tool to implement some corresponding test operations, such as Commissioning and binding. I would like to know whether it is possible to implement the universal chip-tool function?
    On the other hand, various ecosystems such as Apple, Amazon and Google have not implemented binding operations for the time being, such as binding a light bulb to a switch, which seems to be unable to be implemented in any ecosystem. The root cause is that the chip-tool integration is not made into a similar graphical interface to provide users with some corresponding binding operations.
Therefore, I have an idea, that is, to make a remote control, which can control all the matter certified bulbs by embedding some basic Commissioning and binding operations. 
    However, it is necessary to evaluate whether this is necessary. One is will each ecosystem continue to integrate chip-tool-like tools into its own ecosystem? The other is whether chip-tool Universal Edition can control all future matter-certified light bulbs? Because I use chip-tool to put a certified matter bulb Commissioning on the market, the following error is displayed 
 ([1690962486.549080][5259:5262] CHIP:CTL: Failed in verifying 'Attestation Information' command received from the device:  Err 101. Look at AttestationVerificationResult enum to understand the errors)

bzbarsky-apple, 2023-08-03:
>  The other is whether chip-tool Universal Edition can control all future matter-certified light bulbs?

That is the idea, yes.  That's why we have a standard....

> Because I use chip-tool to put a certified matter bulb Commissioning on the market, the following error is displayed

Because when you ran chip-tool, you did not point it to the set of PAA certificates you trust.See https://github.com/project-chip/connectedhomeip/blob/master/examples/chip-tool/README.md#trust-store

`credentials/production/paa-root-certs` and `credentials/development/paa-root-certs` in the Matter tree are periodically synced with the DCL, so pointing ` --paa-trust-store-path` will likely work for current devices.  Or you can get the set of PAAs from the DCL yourself and figure out which ones you trust.

If you want to work with all future devices, that does imply working with future PAAs, so every so often you will need to update your trust store.

chuckrui, 2023-08-04:
> That is the idea, yes. That's why we have a standard....
Therefore, can I understand that different software versions of different chip manufacturers correspond to different chip-tool versions? If I want to realize my idea, I have to find the corresponding chip-tool version according to different chip manufacturers.

> If you want to work with all future devices, that does imply working with future PAAs, so every so often you will need to update your trust store.
Should I be a CSA member to be eligible for DCL authentication PAA certificate authenticity? Either the product provides the PAA certificate to me, so that I can re-specify the PAA certificate path during Commissioning, so that the Commissioning can be successful, right?

bzbarsky-apple, 2023-08-04:
> can I understand that different software versions of different chip manufacturers correspond to different chip-tool versions? 

They are all supposed to be able to talk to each other, because again: we have a standard they are all following.

> Should I be a CSA member to be eligible for DCL authentication PAA certificate authenticity? 

No, reading from the DCL is open to anyone. 

>  Either the product provides the PAA certificate to me

That would totally defeat the purpose, no?  The point is, you decide which PAAs you trust.  The product says which one signed it.  If it's one you trust, then you treat the product as legitimate for your purposes.

chuckrui, 2023-08-07:
> `They are all supposed to be able to talk to each other, because again: we have a standard they are all following.`

chuckrui, 2023-08-07:
[https://webui.dcl.csa-iot.org/pki]
I can't seem to find PAA certificates for products with VID 5020 and PID 53250 in the link. It seems that the PAA certificates that can be downloaded are all named after the company.

bzbarsky-apple, 2023-08-07:
Again: you need to decide which companies to trust to be PAAs.

As a test, have you tried `--paa-trust-store-path credentials/production/paa-root-certs` to see if trusting all the in-tree ones works in your case?

chuckrui, 2023-08-08:
>  trusting all the in-tree ones works in your case?

bzbarsky-apple, 2023-08-08:
> Is it true that ZEMISMART's products are certified under other ATL names?

The issue is not their products, but who signed their device certificate.  You may want to look at the actual PAA certificate returned from the device and see what it claims...

>  (1) OTBR in DOCKER needs to download an image, is this image different from each version of each manufacturer?

I don't know what you are asking, sorry.  Not familiar with the docker otbr bits.

chuckrui, 2023-08-08:
> The issue is not their products, but who signed their device certificate. You may want to look at the actual PAA certificate returned from the device and see what it claims...

So can I understand that? 
If I can know which company signed the PAA certificate actually returned by the device, then I can download the corresponding PAA certificate in the DCL. 
Can I specify this certificate during commissioning to achieve commissioning? 
But I tried to download one of the paa certificates in DCL, and I saw that the format is PEM format, but in the `\connectedhomeip\credentials\production\ PAa-root-certs` certificate file, I saw that there are two files. In addition to PEM, there is also a DER file
Can I see who signed the PAA certificate returned by the device? Could you please tell me how to check this? Please ~

bzbarsky-apple, 2023-08-08:
> If I can know which company signed the PAA certificate actually returned by the device,

You want to know which company _controls_ the PAA certificate returned by the device, not just which one signed it...

> Can I see who signed the PAA certificate returned by the device?

If you look at the command payloads logs on your end, the device is returning its PAA certificate to you.  You should be able to use the chip-cert tool to examine it, but I don't have the syntax offhand; would have to read its help.

chuckrui, 2023-08-08:
Thank you very much for your help~

bzbarsky-apple, 2023-08-08:
> If you look at the command payloads logs on your end, the device is returning its PAA certificate to you.

Actually, I was wrong here, sorry.  You just get the PAI from the device.  You could then look at the issuer DN of that PAI to get some idea of what PAA signed it...

chuckrui, 2023-08-09:
> Actually, I was wrong here, sorry. You just get the PAI from the device. You could then look at the issuer DN of that PAI to get some idea of what PAA signed it...

Can you tell me how to get the PAI of the device for a product that has been certified by matter, I only have the QR Code of the device at present.

bzbarsky-apple, 2023-08-09:
@chuckrui You have to connect it using that QR code (i.e. create a PASE session) and then you can use the CertificateChainRequest command on the Operational Credentials cluster to get that information.

Again, if you are trying to commission it already, that happens as part of commissioning, so if you log received payloads on your side you will be able to see this data...

bzbarsky-apple, 2023-08-09:
You could also try `--paa-trust-store-path credentials/development/paa-root-certs`, by the way.  That seems to have more certs than the production dir does....  If that works, then it's one of the certs that's in the development dir but not the production dir that's the relevant PAA.

chuckrui, 2023-08-09:
emm, this seems to go back to the original discussion, because I made an error in the commissioning process via QR Code, it's like you said, "you did not point it to the set of PAA certificates you trust," but now I have to connect with the QR code and get the PAI certificate and see who signed it, Then go to the DCL to download the corresponding PAA, and finally formulate the PAA certificate in commissioning, which does not seem to work.

bzbarsky-apple, 2023-08-09:
Look, it depends on what the goal here is.

Is the goal "figure out what specific PAA signed this specific device, just as a test"?  Then you can start commissioning, get the PAI, stop commissioning, go figure out the Issuer of the PAI, etc.

Is the goal "be able to commission all Matter devices no matter what"?  Then you just want to download _all_ the PAA certs from the DCL and point the Matter stack at the full set.

Is the goal "I want to control which PAAs I trust, because some of these look fishy"?  Then you need to manually curate the set of PAAs that you trust, based on whatever your trust/threat model is.

chuckrui, 2023-08-09:
> Is the goal "figure out what specific PAA signed this specific device, just as a test"?

`./chip-tool-debug pairing ble-wifi 1111 <ssid> <password> 2625296 1266 --paa-trust-store-path credentials/production/paa-root-certs
./chip-tool-debug pairing ble-wifi 1111 <ssid> <password> 2625296 1266 --paa-trust-store-path credentials/development/paa-root-certs`

`I just want to test first, that is, I need to get the PAA. I executed these two network instructions in the command line, but both showed 101 errors`

`[1691552053.649500][4701:4701] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1691552053.649694][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_kvs-n49JAD)
[1691552053.649892][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
[1691552053.650193][4701:4701] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1691552053.650306][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-kjfsGD)
[1691552053.650419][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1691552053.650501][4701:4701] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1691552053.650588][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_config.ini-rlVy6D)
[1691552053.650688][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1691552053.650754][4701:4701] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1691552053.650834][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-dhpnlE)
[1691552053.650930][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1691552053.651096][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-S6RmgD)
[1691552053.651487][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1691552053.651527][4701:4701] CHIP:DL: NVS set: chip-factory/unique-id = "5D0A700BD68020C2"
[1691552053.651630][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-9g7TTF)
[1691552053.654189][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1691552053.654237][4701:4701] CHIP:DL: NVS set: chip-factory/vendor-id = 65521 (0xFFF1)
[1691552053.654348][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-dYEJoG)
[1691552053.655100][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
[1691552053.655145][4701:4701] CHIP:DL: NVS set: chip-factory/product-id = 32769 (0x8001)
[1691552053.655255][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-d92A4D)
[1691552053.655520][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1691552053.655554][4701:4701] CHIP:DL: NVS set: chip-counters/reboot-count = 1 (0x1)
[1691552053.655642][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-dZcXsF)
[1691552053.656891][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1691552053.656979][4701:4701] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1691552053.657097][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-PZ8zCG)
[1691552053.657695][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1691552053.657761][4701:4701] CHIP:DL: NVS set: chip-counters/boot-reason = 0 (0x0)
[1691552053.657866][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_config.ini-T7F6pG)
[1691552053.658124][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1691552053.658159][4701:4701] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)
[1691552053.658244][4701:4701] CHIP:DL: writing settings to file (/tmp/chip_config.ini-vYchaG)
[1691552053.659348][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1691552053.659392][4701:4701] CHIP:DL: NVS set: chip-config/location-capability = 2 (0x2)
[1691552053.659870][4701:4701] CHIP:DL: Got Ethernet interface: eth0
[1691552053.660168][4701:4701] CHIP:DL: Found the primary Ethernet interface:eth0
[1691552053.660480][4701:4701] CHIP:DL: Got WiFi interface: wlan0
[1691552053.663024][4701:4701] CHIP:DL: Found the primary WiFi interface:wlan0
[1691552053.663106][4701:4701] CHIP:IN: UDP::Init bind&listen port=0
[1691552053.663224][4701:4701] CHIP:IN: UDP::Init bound to port=58643
[1691552053.663259][4701:4701] CHIP:IN: UDP::Init bind&listen port=0
[1691552053.663357][4701:4701] CHIP:IN: UDP::Init bound to port=54099
[1691552053.663385][4701:4701] CHIP:IN: BLEBase::Init - setting/overriding transport
[1691552053.663409][4701:4701] CHIP:IN: TransportMgr initialized
[1691552053.663472][4701:4701] CHIP:FP: Initializing FabricTable from persistent storage
[1691552053.663529][4701:4701] CHIP:TS: Last Known Good Time: [unknown]
[1691552053.663564][4701:4701] CHIP:TS: Setting Last Known Good Time to firmware build time 2023-03-02T09:31:36
[1691552053.666055][4701:4701] CHIP:ZCL: Using ZAP configuration...
[1691552053.671734][4701:4701] CHIP:DL: Avahi client registered
[1691552053.673404][4701:4701] CHIP:CTL: System State Initialized...
[1691552053.676114][4701:4701] CHIP:CR: mbedTLS error: ASN1 - ASN1 tag was of an unexpected value
[1691552053.676311][4701:4701] CHIP:CR: mbedTLS error: ASN1 - ASN1 tag was of an unexpected value
[1691552053.676509][4701:4701] CHIP:CR: mbedTLS error: ASN1 - ASN1 tag was of an unexpected value
[1691552053.677566][4701:4701] CHIP:CR: mbedTLS error: ASN1 - ASN1 tag was of an unexpected value
[... cut ...]
[1691552062.665008][4701:4701] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1691552062.665079][4701:4701] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1691552062.665102][4701:4701] CHIP:DL: Inet Layer shutdown
[1691552062.665119][4701:4701] CHIP:DL: BLE shutdown
[1691552062.668417][4701:4701] CHIP:DL: System Layer shutdown
[1691552062.850489][4701:4701] CHIP:TOO: Run command failure: ../../src/controller/CHIPDeviceController.cpp:1075: CHIP Error 0x000000AC: Internal error
`

Please help me~Thanks~

bzbarsky-apple, 2023-08-09:
@chuckrui Attaching logs, not pasting, would be a lot more readable, for future reference.

So this is showing the relevant bits, but the logs are cut off.  That said, I grabbed the hex for the PAI that is shown, put it into https://lapo.it/asn1js/, fixed up some offsets near the front to account for truncation, and got [something sort of useful](https://lapo.it/asn1js/#MIIBFzCCAROgAwIBAgIRAJZJC8HllYamdLYh1VzblZ8wCgYIKoZIzj0EAwIwOjEiMCAGA1UEAwwZTWF0dGVyIFVhc2NlbnQgUEFBIDB4MTQwMDEUMBIGCisGAQQBgqJ8AgEMBDE0MDAwIBcNMjMwMjI0MDIxNDA2WhgPMjEyMjAxMzEwMzE0MDZaMDQxHDAaBgNVBAMME01hdHRlciBVYXNjZW50ICBQQUkxFDASBgorBgEEAYKifAIBDAQxNDAwMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEYnBl8HbVtNOFs-tDDu6fmxUKw9nyY23CJtTt5Fhxg7_hyqIa6AJ6T-G4yBXlEkBqnp62kSP_5HnQOpn7XLUCaA).

Looks like the issuer here is `Matter Uascent PAA 0x1400`.  This seems to be in the DCL; if I load https://webui.dcl.csa-iot.org/ and click on "PKI" on the left, then search for "Uascent" in the search field, it's found.  But it also seems to be present, for me at least, under `credentials/production/paa-root-certs`.  It's been there since early June...

I have to ask: when you did the `chip-tool-debug pairing ble-wifi 1111 <ssid> <password> 2625296 1266 --paa-trust-store-path credentials/production/paa-root-certs` bit above:

1. Are you running that in the root of a Matter checkout, so that the relative path `credentials/production/paa-root-certs` points to a directory with certs?
2. If the answer to question 1 is "yes", what SHA is that checkout at?

chuckrui, 2023-08-09:
> Attaching logs, not pasting, would be a lot more readable, for future reference.

I used to add logs in "add code" before, but when I clicked "comment", I found that the format was completely confused and not line by line. I was afraid that you might feel uncomfortable, so I posted it directly

>It's been there since early June...

Ooooh，but  I am curious about how you found the PAA issued by "Uascent", as I will need to use this step later.If it's not too complicated and you have time, you can try to explain it to me~

> Are you running that in the root of a Matter checkout, so that the relative path credentials/production/paa-root-certs points to a directory with certs?

My chip-tool is downloaded from here,[https://github.com/nrfconnect/sdk-connectedhomeip/releases](url) it is the version of v2.3.0, running on Raspberry PI. I git the source code of matter to windows. Will this chip-tool version affect commissioning? I don't understand what you mean by SHA in the second point. Can you just give me a link or help document to download a more general version of chip-tool?

[image]

I moved the whole "\connectedhomeip\credentials" folder from the matter source code into the Raspberry PI (I also git pulled it today) and used the v2.3.0 version of chip-tool in the link above.

bzbarsky-apple, 2023-08-09:
> I used to add logs in "add code" before,

No, just use the "Attach files" bit right below the comment box.

> but I am curious about how you found the PAA issued by "Uascent"

I took the part of your log that's a bit after `Performing next commissioning step 'SendPAICertificateRequest'` where it has the long hex dump (search for `474 bytes`; it's right before that).

Then I replaced all the `, 0x` with empty string, took the hex string that results and pasted it into the DER decoder I linked to.  Then fixed some sequence lengths near the beginning to account for it being truncated.  Did you try following the links in my comment above?

>  I git the source code of matter to windows

OK, what version?

In any case, the path to the PAA dir is relative to the working directory.  It lools like your working directory is not the parent directory of `credentials`, per your last screenshot, so you need to adjust paths as needed so that you are pointing at the right spot.

I highly recommend the following as a simple test:

1. Check out a recent connectedhomeup via git.
2. Go to the root of that repo.
3. Run chip-tool from wherever you have it, without changing the working dir (e.g. give a full path to chip-tool).
4. Pass `--paa-trust-store-path credentials/production/paa-root-certs`.  Again, this will only work if that is the right path relative to the working directory!  If you pass a path that chip-tool can't find, it will happily and silently ignore your entire paa-trust-store-path argument and just use the one test PAA.

The version of chip-tool should not matter here.  What matters is what your PAA directory looks like.

chuckrui, 2023-08-09:
> No, just use the "Attach files" bit right below the comment box.

[image]
this one?

> OK, what version?

[image]
emm，maybe V1.0？

bzbarsky-apple, 2023-08-09:
>  this one?

No, that's _above_ the comment box.  I am talking about the place right below the comment box where it says "Attach files by dragging & dropping, selecting or pasting them".  If you click there you get a filepicker.

>  emm，maybe V1.0？

The remote is not the part that matters.  What matters is the SHA you are on.  `git log -1` and look for the commit SHA.

> If I follow the steps you suggested correctly,

OK, so in that last case, when you are in "credentials", do the files `production/paa-root-certs/dcld_mirror_CN_Matter_Uascent_PAA_0x1400_vid_0x1400.der` and `production/paa-root-certs/dcld_mirror_CN_Matter_Uascent_PAA_0x1400_vid_0x1400.pem` exist?

chuckrui, 2023-08-09:
> No, that's above the comment box. I am talking about the place right below the comment box where it says "Attach files by dragging & dropping, selecting or pasting them". If you click there you get a filepicker.

Yeah，i know, thank you ~

> The remote is not the part that matters. What matters is the SHA you are on. git log -1 and look for the commit SHA.

[image]

> OK, so in that last case, when you are in "credentials"

Yes, I have confirmed that both files are present. I am also curious about the fact that only pem file is downloaded in DCL, while there is also a der file in the certificate directory besides pem file. Why is that? Does the attestation process need to verify these two files?

[image]

chuckrui, 2023-08-14:
> I don't think anyone running it normally is running it on Windows).
It's worth clarifying that I'm not running under windows, I'm running chip-tool on Raspberry PI 4B, and the path to the certificate is also inside the Raspberry PI

bzbarsky-apple, 2023-08-28:
>  But I do git submodule update and nothing seems to happen.

You want `git submodule update --init`.

For the rest, you closed this issue: does that mean you figured out what the issue was @chuckrui ? I'm really curious what it was, if you did.

chuckrui, 2023-08-30:
@bzbarsky-apple Yes, I have solved this problem, I re-installed the compilation environment, it is estimated that there is a network problem when git; As for the certificate, I added a parameter (--bypass-attestation-verifier true) when I configured the network. After adding this parameter, it seems that it can bypass the attestation process directly, so as not to verify the DAC and PAI of the device. That is, do not go to the DCL to find the PAA certificate of the corresponding device? Is that right? Or are there any other effects of adding this parameter?

bzbarsky-apple, 2023-08-30:
> Or is there any other effect of adding this parameter?

Well, it skips device attestation verification, so it will allow malicious non-certified devices, etc.  So it really depends on what the goals are here.

bzbarsky-apple, 2023-08-30:
> Oh, and by the way, if the problem was solved, do I need to close this issue？

Closing the issue if the actual issue is resolved, either due to you figuring out you are doing something wrong (and ideally describing here what needed to be changed to make things work) or if there is a code change that fixed things makes sense.

Closing because the problem was worked around while leaving the underlying problem around just means we don't fix the underlying proble, whatever it is, and then the software does not improve....

closed 2023-08-31 by chuckrui