project-chip/connectedhomeip #28493: [Feature] Support cross-complier for raspberry pi(aarch64)

resghst, 2023-08-03:
### Feature description

The raspberry pi builds chip-tool and host, which is very slow and limited memory resources.
If the connectedhomeip supports aarch64 cross-complier, everyone can build chip-tool and host on a desktop or laptop.
Currently, the raspberry pi needs a 64GB SDCard for matter env and source code. If supports aarch64 cross-complier, everyone can use a small SDCard for setup chip-tool and host on Raspberry Pi.

### Platform

raspi

bzbarsky-apple, 2023-08-03:
@resghst Is the issue that we just need better "here is how you cross-compile for Pi" documentation?

resghst, 2023-08-04:
@bzbarsky-apple amazing, I will try that 
Thanks you.

resghst, 2023-08-23:
Hi @bzbarsky-apple,
I successfully set up the docker image but don't know how to use it.

bzbarsky-apple, 2023-08-28:
@resghst Sorry, I don't know that either...

andy31415, 2023-08-28:
To cross compile, use for example:

```
./scripts/build/build_examples.py --target linux-arm64-all-clusters-clang build
./scripts/build/build_examples.py --target linux-arm64-chip-tool-clang build
```

Only clang versions can cross compile. This should work in the vscode image or crosscompile image in docker or you can setup your own sysroot (or download the existing one from CIPD).

Our CI for cross compile is defined in https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/examples-linux-arm.yaml

andy31415, 2023-08-28:
If you are wondering about the manual build commands:

```
# ./scripts/build/build_examples.py --target linux-arm64-chip-tool-clang --dry-run --log-level warn build
# Commands will be run in CHIP project root.
cd "."

# Generating linux-arm64-chip-tool-clang
bash -c '
PKG_CONFIG_PATH="/opt/ubuntu-22.04.1-aarch64-sysroot/lib/aarch64-linux-gnu/pkgconfig" \
 gn gen --check --fail-on-unused-args --export-compile-commands --root=/workspace/examples/chip-tool '"'"'--args=is_clang=true target_cpu="arm64" sysroot="/opt/ubuntu-22.04.1-aarch64-sysroot"'"'"' /workspace/out/linux-arm64-chip-tool-clang'

# Setting up Java deps
third_party/java_deps/set_up_java_deps.sh

# Building linux-arm64-chip-tool-clang
bash -c '
PKG_CONFIG_PATH="/opt/ubuntu-22.04.1-aarch64-sysroot/lib/aarch64-linux-gnu/pkgconfig" \
 ninja -C /workspace/out/linux-arm64-chip-tool-clang'
 ```

you can probably skip the java deps (why do we even have those here? looks like a bug to me ...) and generally it looks that we just make sure PKG_CONFIG_PATH is set and we setup the sysroot and target_cpu

closed 2024-07-11 by resghst