project-chip/connectedhomeip #28516: [Feature] Is matter support OOB(Out Of Bound) commission process?

dipVyas92, 2023-08-04:
### Feature description

Hi,

We have one query regarding the commission process for matter devices over BLE.

We want to try commission matter end device over the BLE using the OOB(Out Of Bound) method.

OOB method means:- Commission tool has all the matter dataset details. Using that commissioning tool will write all datasets to the matter end device and the device will read all datasets and the device will connect to the matter network over an openthread.

Based on this feature the end device does not need to be in range with any IoT gateway or Matter Host.

Can Matter end device support this kind of OOB method to commission?

### Platform

k32w

### Platform Version(s)

Matter-v1.1

dipVyas92, 2023-08-08:
@bzbarsky-apple can you please help me on this query?

bzbarsky-apple, 2023-08-08:
@dipVyas92 I did look at this issue, but I don't understand what you are asking, exactly.  If you have some method of writing networking configuration to a device not using Matter, then that devices advertises commissionable via DNS-SD on that network, then commissioners can commission it.  If that's not the question, what is the question?

dipVyas92, 2023-08-09:
Dear @bzbarsky-apple,

Thank you for getting back to me.

I want to use commissioning method to ensure the best possible results for end device. I suggest utilizing a mobile device as commissioning tool, which will seamlessly send all necessary Matter connection data to matter device. This will allow it to connect to OpenThread without any interference from the OpenThread Host, ensuring a smooth and reliable experience.

Please let me know if need more information regarding this.

I have attached two images for more clarification 

[image]

[image]

Thanks in advance.

bzbarsky-apple, 2023-08-09:
@dipVyas92 Yes, that's exactly how Matter commissioning works: you connect to the devices over BLE and provision them with the Thread credentials.

dipVyas92, 2023-08-10:
@bzbarsky-apple 

Thanks for the response

Is it necessary for the end device to be within range of the IoT gateway or Openthread Host device for this commissioning process?

am I understanding correctly?

bzbarsky-apple, 2023-08-10:
> Is it necessary for the end device to be within range of the IoT gateway or Openthread Host device for this commissioning process?

Well, commissioning is expected to put it on the Thread network _and then contact it over the Thread network to verify that worked_, so yes.

pramodhrachuri, 2025-07-22:
Thanks for the response. I have a follow-up question.

Openthread's documentation says, "Only the Thread Network Key is needed to attach to a network." [[reference](https://openthread.io/reference/cli/commands#dataset_set_activepending:~:text=Only%20the%20Thread%20Network%20Key%20is%20needed%20to%20attach%20to%20a%20network.)]
Therefore, I can code a device (such as an ESP32c6) to receive the network key (+ other parts of the dataset) via UART and initiate the attachment process to join a Thread network. Correct?
Now that the device is on the Thread network and DNS-SD has started, we can initiate the matter commissioning process (like explained [here](https://docs.silabs.com/matter/1.0.3/matter-overview-guides/commissioning)). 

I'm thinking of two different paths after this.
1. If the commissioner (like a mobile) is already in the thread network, it should be able to handle the onboarding. Correct? Or does it still need a Bluetooth communication link?
2. Once the device is attached to Thread, can software like Home Assistant handle the commissioning process? This way, we can eliminate the need for a separate commissioner device.

Please correct me if any of my assumptions are wrong or my hypotheses are incorrect.