project-chip/connectedhomeip #28521: [Scenes] Fix SceneTableSize

lpbeliveau-silabs, 2023-08-04:


bzbarsky-apple, 2023-08-04:
If we want this defined via ZAP, why are we doing it via AttributeAccessInterface at all?

lpbeliveau-silabs, 2023-08-04:
Currently we are only getting the SceneCount and RemainingCapacity through the AttributeAccessInterface, SceneTableSize should not rely on it.

bzbarsky-apple, 2023-08-05:
Then I am confused.  We're not using CHIP_CONFIG_MAX_SCENES_PER_ENDPOINT for the attribute value... Is this issue then about setting CHIP_CONFIG_MAX_SCENES_PER_ENDPOINT or some equivalent automatically to the largest SceneTableSize set in ZAP across all endpoints?

lpbeliveau-silabs, 2023-08-08:
Yes, that is the actual problem. CHIP_CONFIG_MAX_SCENES_PER_ENDPOINT should be equivalent to the zap parameter, otherwise someone trying to configure a device with a SceneTableSize > CHIP_CONFIG_MAX_SCENES_PER_ENDPOINT would face unexpected Resource exhausted errors.

closed 2023-10-31 by mergify[bot]