project-chip/connectedhomeip #28539: build chip-tool on ubuntu 22.04.1 x64 for arm64, but some files can not find

githubworksonline, 2023-08-05:
### Test issue(s)

**ubuntu 22.04.1   connectedhomeip-main branch 

use default build options is perfect, like this**
./scripts/examples/gn_build_example.sh examples/chip-tool out/test

**i want build for arm64:**
./scripts/examples/gn_build_example.sh examples/chip-tool out/armbian 'target_cpu="arm64" target_os="linux"'

**build is done, but FAILED occurred：**

Done. Made 268 targets from 231 files in 400ms
+ ninja -C out/armbian
ninja: Entering directory `out/armbian'
[1/2] ld ./chip-tool
FAILED: chip-tool chip-tool.map 
aarch64-linux-gnu-g++ -O0 -fPIC -Werror -Wl,--fatal-warnings -Wl,-z,defs -fdiagnostics-color -Wl,--gc-sections -pie liblogging-stdio.a -Wl,-Map,./chip-tool.map -Wl,--cref @./chip-tool.rsp -o ./chip-tool
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lssl: No such file or directory
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lcrypto: No such file or directory
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lgio-2.0: No such file or directory
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lgobject-2.0: No such file or directory
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: cannot find -lglib-2.0: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

**i tried many ways, but failed, so have to ask for advice。
thank u.**

bzbarsky-apple, 2023-08-05:
@githubworksonline Are those libraries installed for your cross-compiler's target architecture?

githubworksonline, 2023-08-12:
> @githubworksonline Are those libraries installed for your cross-compiler's target architecture?

sorry for reply so late. 
take ‘libssl.so’ as an eaxmple, this is some tests:

**$ cd /usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin
$ ls
ar  as  ld  ld.bfd  ld.gold  libssl.so  nm  objcopy  objdump  ranlib  readelf  strip

$ ls -l libssl.so 
lrwxrwxrwx 1 root root 44 Aug  6 16:58 libssl.so -> /usr/aarch64-linux-gnu/lib/lib/libssl.so.1.1

$ cd /usr/aarch64-linux-gnu/lib/lib
$ ls
engines-1.1  libcrypto.a  libcrypto.so  libcrypto.so.1.1  libssl.a  libssl.so  libssl.so.1.1  pkgconfig**

the target is ‘libssl.so’ or ‘libssl.so.1.1’，the fail always occurred.

bzbarsky-apple, 2023-08-28:
Is the compiler actually looking for the libraries at those paths?  Because clearly it's not finding them.....

taigila, 2023-09-19:
@bzbarsky-apple hi sir, i have been trying to redirect GN to look for library in a custom path (because i personally compile them to a specific path). i have used --system_libdir and when i wanna compile using ninja - C, they still look for the libraries on the default ubuntu 22.04 libraries. How can i change that? thank you

bzbarsky-apple, 2023-09-19:
@taigila I'm sorry, I don't know.

taigila, 2023-09-20:
@bzbarsky-apple thank you for your reply. No problem. Thank you for your contribution so far :) .

closed 2023-10-24 by woody-apple