project-chip/connectedhomeip #28544: [BUG] Modeselect without semantic tags, TLV encoder failure GPF

jonsmirl, 2023-08-06:
### Reproduction steps

https://github.com/espressif/esp-matter/issues/558

The TLV encoder is not handling the case when Semantic Tag Struct is empty (zero length).

It does work properly when I include a bogus semantic tag.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

core

bzbarsky-apple, 2023-08-06:
> when Semantic Tag Struct is empty (zero length)

What does that mean, exactly?  What does your ModeOptionsProvider implementation look like?  What is it returning?

The stacks in https://github.com/espressif/esp-matter/issues/558 seem to have both `TLVWriter::WriteData`  bits and `emberAfLongStringLength` bits.. and if it's the `WriteData` call the issue is `len=12, p=0x0`.

So chances are, the data being passed to the TLV encoder is garbage, but it's really hard to say without knowing what that data is.  Please provide that information.

jonsmirl, 2023-08-06:
I asked the Espressif people to look at in their issue.

In the ESP Platform code, (examples/platform/esp32/mode-support/static-supported-modes-manager.cpp maybe?) when there are no semantic tags, do you build a zero length array and initialize it? If it is not initialized you'd see this error.

bzbarsky-apple, 2023-08-07:
`examples/platform/esp32/mode-support/static-supported-modes-manager.cpp` should only be getting used for the in-tree example apps, I would think....

That said, I have no idea how exactly you are configuing your "no semantic tags" situation or even exactly what code you are running, so...

> Is line 133 right for no semantic tags?

Again: is this code even running in your situation?  That said, if it is line 133 is what runs on allocation failure... assuming your `new` can actually fail without trying to throw an exception.  And returning `ModeOptionsProvider(nullptr, nullptr)` should just lead to an empty mode list.

If this code _is_ running in your setting, what is `semanticTagCount` ending up as?  If nonzero, what are the outputs of the `ReadConfigValue` calls in the the loop?

dhrishi, 2023-08-07:
@jadhavrohit924 Please check

jadhavrohit924, 2023-08-08:
@jonsmirl From https://github.com/espressif/esp-matter/issues/558#issue-1838022033 your `SupportedModes` array is 
`[ { mode: 0, label: "Switch", semanticTag: [] } ,
    { mode: 1, label: "Dimmer", semanticTag: [] } ,
    { mode: 2, label: "Smartbulb", semanticTag: [] } ]` right?
  
And you need empty semanticTags, so the input for the mfg_tool should be in format `--supported-modes 0/Switch/1 1/Dimmer/1 2/Smartbulb/1` as mentioned in documents https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#usage

jonsmirl, 2023-08-08:
I am using this:
~~~
./mfg_tool.py -cn "Test" -v 0xFFF2 -p 0x8001 --pai \
-k ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
-c ../../connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd ../../connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
--vendor-name "Lowpan" --product-name "Dimmer" --serial-num "12345678" \
--hw-ver-str "1.0.0" \
--passcode 20202021 --discriminator 3840 \
--fixed-labels "1/name/Main" "2/name/BTN-2" "3/name/BTN-3" "4/name/BTN-4" "5/name/BTN-5" "6/name/BTN-6" "7/name/BTN-7"  "8/name/BTN-8" \
--supported-modes "0/Switch/1" "1/Dimmer/1" "2/Smartbulb/1"
~~~

jadhavrohit924, 2023-08-08:
Could you please try using for empty semanticTags? Don’t use `""`. It should work.
```
--supported-modes 0/Switch/1 1/Dimmer/1 2/Smartbulb/1
```

jonsmirl, 2023-08-08:
I have changed the script to remove the quotes and rebuilt the partition. But.. Google broke commissioning for me yesterday (attestation failure with the demo certificates again) and I'm trying to fix that right now. Once I fix I can commission the device and test mode setting.

If those quotes is what caused the error, the python code should error out instead of making an invalid partition.

You can try with chip-tool, just do a wild card subscribe from the device. When the wild card gets to mode setting, it faulted.

jonsmirl, 2023-08-08:
First, I take back my complaint against Google. Their code was working this time and it unmasked a bug in my code which is now fixed. 

Removing the quotes did not change anything.

@jadhavrohit924  @dhrishi  Add the modselect without the semantic tags to a device. Then use chip-tool to create a wildcard subscription to read everything from the device.  You will hit this every time.

~~~
I (68639) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000050's Attribute 0x0000FFFC is 0 **********
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x400588a2  RA      : 0x4208b0e4  SP      : 0x3fcc11d0  GP      : 0x3fc95000  
Stack dump detected
0x400588a2: _data_end_btdm_rom in ROM

0x4208b0e4: chip::TLV::TLVWriter::WriteData(unsigned char const*, unsigned long) at /home/jonsmirl/lowpanC3/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:786
 (inlined by) chip::TLV::TLVWriter::WriteData(unsigned char const*, unsigned long) at /home/jonsmirl/lowpanC3/build/esp-idf/chip/../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:760

TP      : 0x3fc7a070  T0      : 0x00003f9d  T1      : 0x00000389  T2      : 0xffffffff  
S0/FP   : 0x3fcc1430  S1      : 0x0000000c  A0      : 0x3fcd6171  A1      : 0x00000000  
A2      : 0x0000000c  A3      : 0x0000000c  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0xfffffffa  S2      : 0x0000000c  S3      : 0x00000000  
S4      : 0x3fcc1444  S5      : 0x3fcc143c  S6      : 0x0000ffff  S7      : 0x00000000  
S8      : 0x00000002  S9      : 0x0000ffff  S10     : 0x3c14d000  S11     : 0x42022000  
0x42022000: emberAfLongStringLength(unsigned char const*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/third_party/nlio/repo/include/nlio-base.h:151
 (inlined by) GetUnaligned16 at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/third_party/nlio/repo/include/nlio-base.hpp:145
 (inlined by) GetUnaligned16 at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/third_party/nlio/repo/include/nlio-byteorder-little.hpp:653
 (inlined by) chip::Encoding::LittleEndian::Get16(unsigned char const*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/lib/core/CHIPEncoding.h:312
 (inlined by) emberAfLongStringLength(unsigned char const*) at /home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/src/app/util/message.cpp:35

T3      : 0x0000005a  T4      : 0x3fcd616d  T5      : 0x3fcd5de4  T6      : 0x3fcc145c  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x400588a2 in ?? ()
#0  0x400588a2 in ?? ()
#1  0x4208b0e4 in chip::TLV::TLVWriter::WriteData (len=12, p=0x0, this=<optimized out>) at ../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:785
#2  chip::TLV::TLVWriter::WriteData (this=0x3fcc1430, p=<optimized out>, len=<optimized out>) at ../../../../aosp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/core/TLVWriter.cpp:760
#3  0xfffffff9 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
ELF file SHA256: 9e2b9fcdf5f481b9

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
0x40048b82: ets_secure_boot_verify_bootloader_with_keys in ROM

~~~

jonsmirl, 2023-08-08:
I have verified with chip-tool that the fault is on reading attribute 0x2.

The attribute is getting encoded here:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/mode-select-server/mode-select-server.cpp#L87
It is failing in the call to ReturnErrorOnFailure(encoder.Encode(modeOption));  on the first mode definition.

Is this modeOption variable being passed into Encode() what Encode is expecting?

This code is building the mode list:
https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/esp32/mode-support/static-supported-modes-manager.cpp#L61

jadhavrohit924, 2023-08-09:
I have tried as you asked. My device was never crashed. Here are the steps that I follow.

1. Generate factory partition
```
./tools/mfg_tool/mfg_tool.py -cn "My bulb" --vendor-id 0xFFF2 -p 0x8001 --discriminator 3434 --passcode 99663300 --pai \
    -k connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
    -c connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
    -cd connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der --supported-modes 0/Switch/1 1/Dimmer/1 2/Smartbulb/1
```
2. Add mode select as described in https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#build-example
3. Commission device and read and subscribe to mode-select
```
./out/host/chip-tool any read-all 0x50,0x50 0xFFFFFFFF 0xFFFFFFFF 1234 1,1

./out/host/chip-tool any subscribe-all 0x50,0x50 0xFFFFFFFF 0xFFFFFFFF 5 10 1234 1,1
```

Do let me know if I'm missing something. Also provide us what steps you are following.

jonsmirl, 2023-08-09:
I found the problem, i was doing InitEndpointArray() at the wrong place in my code. @dhrishi 

I find this implementation to be very fragile. 
1) it is pulling in code from examples/platform/esp32  -- shouldn't that code be in the esp-matter project?
2) This code that I had to add should be hidden in the esp-matter library.
~~~
        ModeSelect::StaticSupportedModesManager obj;
        obj.InitEndpointArray(get_count(node));
~~~
For example as static initialization on the modeselect cluster.  In my code is it ambiguous when to call it, before matter start? after?  make it a static?

closed 2023-08-09 by jonsmirl