project-chip/connectedhomeip #28546: [BUG] Python chip-device-ctrl module 'chip' has no attribute 'CertificateAuthority'

WladMl, 2023-08-06:
### Reproduction steps

chip-device-ctrl fails with the error
 2023-08-06 15:02:02 WRO1-LDL-P90374 chip.DL[74749] INFO PublishService D460C5A7D93C7AF4._matterd._udp5550.-1
2023-08-06 15:02:02 WRO1-LDL-P90374 chip.DIS[74749] INFO mDNS service published: _matterd._udp; instance name: D460C5A7D93C7AF4
module 'chip' has no attribute 'CertificateAuthority'
Failed to bringup CHIPDeviceController CLI
Aborted (core dumped)

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

c4d95d0bef9295daf8652348b72eec9778cac944

### Platform

python, other

mhanuel26, 2024-01-04:
I am getting this issue on my x86 platform, does this has a workaround or solution?

[image]

Chapoly1305, 2024-02-29:
Same, need some help here.