project-chip/connectedhomeip #28549: [Darwin] Swift support for CHIPTOOL on iOS?

twlawrenceko, 2023-08-07:
### Reproduction steps / Feature

Hello,

The current https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#using-chip-tool-on-macos-or-chip-tool-on-ios CHIPTOOL example on iOS is based on Objective-C. Is there any way to implement this on top of the existing Swift iOS project?

Thank you.

### Platform

darwin

### Type

Platform validated

bzbarsky-apple, 2023-08-07:
@twlawrenceko What is the exact ask here?  The example app is an example.  Matter.framework can be used from Swift.  There are no current plans to rewrite the example in Swift, but obviously if someone contributes a Swift example that will be gladly accepted.

twlawrenceko, 2023-08-14:
@bzbarsky-apple We are trying to get mDNS details after Matter device commissioning is complete. We do see the list of information print inside the Xcode console, but not sure where this code is located. Can you point out where this information is stored or how to retrieve it?

```
2023-08-14 22:15:48.243668+0800 CHIPTool[6201:4120210] [Controller] Stopping commissioning discovery over DNS-SD
2023-08-14 22:15:48.243681+0800 CHIPTool[6201:4120210] [Controller] Starting commissioning discovery over BLE
2023-08-14 22:15:48.243718+0800 CHIPTool[6201:4120210] [Ble] NewConnection
2023-08-14 22:15:48.243758+0800 CHIPTool[6201:4120210] [Controller] Starting commissioning discovery over DNS-SD
2023-08-14 22:15:48.243845+0800 CHIPTool[6201:4120210] [Discovery] Browsing for: _matterc._udp,_L3840
2023-08-14 22:15:48.247236+0800 CHIPTool[6201:4120640] [Discovery] Mdns: OnBrowseAdd  name: 5BB22D9469A796BF, type: _matterc._udp., domain: local., interface: 18
2023-08-14 22:15:48.247318+0800 CHIPTool[6201:4120640] [Discovery] Resolve type=_matterc._udp name=5BB22D9469A796BF interface=18
2023-08-14 22:15:48.250339+0800 CHIPTool[6201:4120736] [Discovery] Mdns: OnResolve flags: 0, interface: 18, fullname: 5BB22D9469A796BF._matterc._udp.local., hostname: 4C7525E49E24.local., port: 42005
2023-08-14 22:15:48.250407+0800 CHIPTool[6201:4120736] [Discovery] Mdns : OnNewInterface hostname:4C7525E49E24.local. fullname:5BB22D9469A796BF._matterc._udp.local. interface: 18 port: 5540 TXT:"VP=65521+32768,T=1,D=3840,CM=1,PH=33,PI="
2023-08-14 22:15:48.253458+0800 CHIPTool[6201:4120741] [Discovery] Mdns: OnGetAddrInfo flags: 1073741827, interface: 18, hostname: 4C7525E49E24.local.
2023-08-14 22:15:48.253567+0800 CHIPTool[6201:4120741] [Discovery] Mdns: OnNewAddress interface: 18 ip:fe80::4e75:25ff:fee4:9e24
2023-08-14 22:15:48.253610+0800 CHIPTool[6201:4120741] [Discovery] Mdns: OnGetAddrInfo flags: 1073741826, interface: 18, hostname: 4C7525E49E24.local.
2023-08-14 22:15:48.253638+0800 CHIPTool[6201:4120741] [Discovery] Mdns: OnNewAddress interface: 18 ip:192.168.2.202
2023-08-14 22:15:48.253666+0800 CHIPTool[6201:4120741] [Discovery] Mdns: Resolve success on interface 18
2023-08-14 22:15:48.253738+0800 CHIPTool[6201:4120741] [Discovery] Discovered node:
2023-08-14 22:15:48.253980+0800 CHIPTool[6201:4120741] [Discovery] 	Hostname: 4C7525E49E24
2023-08-14 22:15:48.253996+0800 CHIPTool[6201:4120741] [Discovery] 	IP Address #1: fe80::4e75:25ff:fee4:9e24
2023-08-14 22:15:48.254005+0800 CHIPTool[6201:4120741] [Discovery] 	IP Address #2: 192.168.2.202
2023-08-14 22:15:48.254011+0800 CHIPTool[6201:4120741] [Discovery] 	Port: 5540
2023-08-14 22:15:48.254045+0800 CHIPTool[6201:4120741] [Discovery] 	Mrp Interval idle: not present
2023-08-14 22:15:48.254050+0800 CHIPTool[6201:4120741] [Discovery] 	Mrp Interval active: not present
2023-08-14 22:15:48.254055+0800 CHIPTool[6201:4120741] [Discovery] 	TCP Supported: 1
2023-08-14 22:15:48.254061+0800 CHIPTool[6201:4120741] [Discovery] 	Vendor ID: 65521
2023-08-14 22:15:48.254066+0800 CHIPTool[6201:4120741] [Discovery] 	Product ID: 32768
2023-08-14 22:15:48.254070+0800 CHIPTool[6201:4120741] [Discovery] 	Long Discriminator: 3840
2023-08-14 22:15:48.254075+0800 CHIPTool[6201:4120741] [Discovery] 	Pairing Hint: 33
2023-08-14 22:15:48.254513+0800 CHIPTool[6201:4120741] [Discovery] 	Instance Name: 5BB22D9469A796BF
2023-08-14 22:15:48.254528+0800 CHIPTool[6201:4120741] [Discovery] 	Commissioning Mode: 1
2023-08-14 22:15:48.254567+0800 CHIPTool[6201:4120741] [Controller] Discovered device to be commissioned over DNS-SD
2023-08-14 22:15:48.254746+0800 CHIPTool[6201:4120741] [Controller] Attempting PASE connection to UDP:[fe80::4e75:25ff:fee4:9e24%en2]:5540
```

Thank you

bzbarsky-apple, 2023-08-25:
@twlawrenceko Sorry for the lag; I was on vacation for a bit.

The information you see logged there is not stored anywhere on any sort of persistent basis.  It's received, then acted on.  The relevant code is either Darwin-specific code (the things with the "Mdns:" prefixes on the log messages) in https://github.com/project-chip/connectedhomeip/blob/ed3570daeb09b796a6159570cbdcd75453463609/src/platform/Darwin/DnssdContexts.cpp and https://github.com/project-chip/connectedhomeip/blob/ed3570daeb09b796a6159570cbdcd75453463609/src/platform/Darwin/DnssdImpl.cpp or platform-independent code (the "Discovered node:" bit) in https://github.com/project-chip/connectedhomeip/blob/ed3570daeb09b796a6159570cbdcd75453463609/src/lib/dnssd/Discovery_ImplPlatform.cpp where it calles `LogDetail` in `HandleNodeResolve`.

Note that what you have logged there is the commissionable advertisement.  If you are looking at "mDNS details after Matter device commissioning is complete" are you actually looking for the operational advertisement?  And what information are you actually looking for from it?

law-ko, 2023-09-02:
@bzbarsky-apple We are looking for the hostname and IP address so we can connect to the device using other protocols. If we are using the `import Matter` framework directly in Swift, is there an easy way to get these information?

bzbarsky-apple, 2023-09-02:
> We are looking for the hostname and IP address so we can connect to the device using other protocols.

Matter.framework does not expose APIs for this, since it's not needed for normal Matter interactions.

That said, you should be able to use APIs like https://developer.apple.com/documentation/dnssd/1804744-dnsserviceresolve to look up that sort of information, if you know the instance name.  For operational advertisements, the instance name is just the concatenation of the compressed fabric id (which you can compute from your root public key and fabric id) and the node id of the node.

closed 2023-10-19 by woody-apple

huynguyen-df, 2023-10-20:
> > We are looking for the hostname and IP address so we can connect to the device using other protocols.
> 
> Matter.framework does not expose APIs for this, since it's not needed for normal Matter interactions.
> 
> That said, you should be able to use APIs like https://developer.apple.com/documentation/dnssd/1804744-dnsserviceresolve to look up that sort of information, if you know the instance name. For operational advertisements, the instance name is just the concatenation of the compressed fabric id (which you can compute from your root public key and fabric id), a dash, and the node id of the node. Both ids are formatted as 16-char uppercase hex strings.

How can I calculate `the instance name`? Do you have a formula for it, @woody-apple ? I hope you will be able to provide the information.

bzbarsky-apple, 2023-10-21:
> How can I calculate the instance name?

I assume you know the fabric's public key, fabric id, and node id?  The instance name is XXXXXXXX-YYYYYYYY where XXXXXXXXX is the compressed fabric ID and YYYYYYYY is the node id.  Both are 16 ASCII chars.

The compressed fabric ID is computed by using the Matter-spec-defined KDF (see below) with the root public key as the key, the fabric ID (as a 64-bit unsigned integer scalar in big-endian byte order) as the salt, "CompressedFabric" as the "info" passed to the KDF.  Then you encode it as all-uppercase hex to get a 16-char string.

The node id is just encoded as a 16-char hex string in big-endian order (it's an 8-byte value to start with).

The KDF involved is `HKDF-Expand(HKDF-Extract(salt, inputKey), info, 8)` where HKDF-Expand and HKDF-Extract come from  RFC 5869 section 2.3 and 2.2 respectively, both using HMAC-SHA256 as the auxiliary HMAC-Hash function.

So a little annoying.

More generally, all this is defined in the Matter specification...