project-chip/connectedhomeip #28557: How can I bind the Matter Light to the Light Switch on Zigbee to Matter Bridge?

LongND0105, 2023-08-07:
I have a Matter Light application is running on ESP32. And I have a Light Switch on Zigbee to Matter Bridge. Can I bind Light to Light Switch? I have tried bind but chip-tool return CONSTRAINT_ERROR.
I defined Light Switch end point belike: 
[image]

[image]

[image]

bzbarsky-apple, 2023-08-07:
> I have tried bind but chip-tool return CONSTRAINT_ERROR

What was the exact chip-tool command involved here @LongND0105 ?

That said, what you can do is bind Matter endpoints to other Matter endpoints.  In your picture above, there is no Matter endpoint representing the Zigbee light, right?

LongND0105, 2023-08-08:
> What was the exact chip-tool command involved here @LongND0105 ?

I used the command: binding write binding '[{"fabricIndex": 1, "node": <node id Light ESP32>, "endpoint": 1, "cluster": 6}]' <node id Zigbee to Matter bridge> 3
I think this error is not relate to chip-tool.
> That said, what you can do is bind Matter endpoints to other Matter endpoints. In your picture above, there is no Matter endpoint representing the Zigbee light, right?

Endpoint 3 is the Matter endpoint representing the Zigbee light in the bridge. In theory, can I bind light esp32 to matter endpoint in the bridge? If I can, the step to do this.

bzbarsky-apple, 2023-08-08:
> `binding write binding '[{"fabricIndex": 1, "node": , "endpoint": 1, "cluster": 6}]' 3`

That does not look like a valid binding value (missing a node id, no?).  And it's missing the endpoint for the write... so I doubt that's the actual command you ran.  But if you did in fact try to create a binding with no node id in the binding entry but an endpoint present, that's not allowed by the spec and you would in fact get CONSTRAINT_ERROR for that.

> Endpoint 3 is the Matter endpoint representing the Zigbee light in the bridge.

So you're trying to bind one endpoint on the bridge to another endpoint on the bridge?  That is likely to not work well; see https://github.com/project-chip/connectedhomeip/issues/21626

LongND0105, 2023-08-08:
> That does not look like a valid binding value (missing a node id, no?)

Sorry, it is the error typing. The actual command I ran: 
> binding write binding '[{"fabricIndex": 1, "node": 'node id Light ESP32', "endpoint": 1, "cluster": 6}]' 'node id Zigbee to Matter bridge' 'ep id'

> So you're trying to bind one endpoint on the bridge to another endpoint on the bridge? 

No. I am trying to bind one endpoint on the bridge to another endpoint, that is not on the bridge.
[image]

bzbarsky-apple, 2023-08-08:
> `binding write binding '[{"fabricIndex": 1, "node": 'node id Light ESP32', "endpoint": 1, "cluster": 6}]' 'node id Zigbee to Matter bridge' 'ep id'`

I assume those strings were actually numbers, right?

As far as I can tell, bindings only returns CONSTRAINT_ERROR if IsValidBinding returns false. The question is why it returns false in your case.

You have "node" and "endpoint" and no "group", so that's not a problem.  It's _possible_ `emberAfContainsClient` returns false for this cluster id on that endpoint.  You could test by not passing a "cluster" in the binding struct and seeing if that works.

I do see a client cluster declared in your cluster list bits above, but maybe something goes wrong with that...

LongND0105, 2023-08-08:
> I assume those strings were actually numbers, right? 

Yes

> You could test by not passing a "cluster" in the binding struct and seeing if that works.

I tested with not passing a cluster in the binding command and this is the result:
[image]

LongND0105, 2023-08-08:
> What does your code-generated `MATTER_PLUGINS_INIT` macro look like? It should be in the generated `PluginApplicationCallbacks.h` for your application.
This is PluginApplicationCallbacks.h file: 
[image]

So I actually do not understand code. Could you descript the arch of the project bridge-app? And I don't know this function `MatterBindingPluginServerInitCallback` that how it to do.

jonsmirl, 2023-08-08:
The ESP Zigbee bridge is only a demo, it is not a complete bridge. It is meant as a starting point for a customer who wants to build a complete bridge. It is likely that switch binding has not implement in the bridge code. AFAIK the only thing implemented is light bulb support.  @dhrishi 

If you implement the code for switch binding support, it should work.

bzbarsky-apple, 2023-08-08:
> Could you descript the arch of the project bridge-app?

That's a very broad question....

Your very concrete problem is that the example bridge-app has an endpoint (endpoint 2, I believe) where the ZAP config enables various application clusters that need to be exposed to codegen...  So if you enable Bindings on that endpoint, that should solve your problem with MatterBindingPluginServerInitCallback.

LongND0105, 2023-08-09:
> Your very concrete problem is that the example bridge-app has an endpoint (endpoint 2, I believe) where the ZAP config enables various application clusters that need to be exposed to codegen... So [...]

I declared binding cluster in dynamic endpoin but `PluginApplicationCallbacks.h` have not `MatterBindingPluginServerInitCallback`. Could you give me some idea to enable `MatterBindingPluginServerInitCallback`

[image]

bzbarsky-apple, 2023-08-09:
@LongND0105 Again: you need to go into the ZAP config of your application and enable that cluster on the non-dynamic endpoint you presumably have (and which you are then presumably disabling, if you are using bridge-app as a basis).

Or you could just call the function when bringing up the data model, as long as you make very sure it's called only once....

LongND0105, 2023-08-09:
It mean I have to enable in ZAP config? But I want to dynamic endpoint

bzbarsky-apple, 2023-08-09:
@LongND0105 Does your device have a ZAP config at all?  I expect yes.  What does that ZAP config look like?

LongND0105, 2023-08-09:
> Does your device have a ZAP config at all?

Yes.
> What does that ZAP config look like? 

I'm using `bridge-common.zap`

bzbarsky-apple, 2023-08-09:
You mean `examples/bridge-app/bridge-common/bridge-app.zap`?

If so, open that up in ZAP, go to endpoint 2, enable the Bindings server on that endpoint, save, quit.

LongND0105, 2023-08-09:
I'm trying to run zap but terminal output this error: 

[image]

bzbarsky-apple, 2023-08-09:
@LongND0105 Remove the `~/.zap` cache and try again.

LongND0105, 2023-08-10:
> @LongND0105 Remove the `~/.zap` cache and try again.

Thanks. I opened `examples/bridge-app/bridge-common/bridge-app.zap` in ZAP. I'm trying to edit:
- ep0: root
- ep1: aggregator (parent endpoint for matter endpoint representing zigbee device)
- ep2: aggregator (parent endpoint for matter endpoint representing BLE Mesh device)
- other endpoint is created dynamic
[image]

Now I'm have to enable binding client cluster in ep1 and ep2?

bzbarsky-apple, 2023-08-10:
What happened to the EP2 `examples/bridge-app/bridge-common/bridge-app.zap` used to have?  The one that was labeled "Matter Dimmable Light"?

The way the bridge-app example is structured, it has three static endpoints:
* EP0, for commissioning and whatnot
* EP1, aggregator
* EP2, dummy endpoint that just hosts server clusters for all the clusters the bridge knows how to bridge.  This endpoint is then disabled at https://github.com/project-chip/connectedhomeip/blob/96426f33700c18e01b577b6f79a13aa6cdf25cd7/examples/bridge-app/linux/main.cpp#L963-L965 with a comment explaining why it's there.

and then dynamic endpoints starting at EP3.  In your setup, if you have two aggregators, you could put the dummy endpoint on EP3 or something (or really, any place you want), turn on severs for all the clusters you want to bridge, then disable the endpoint at runtime like the example does.

LongND0105, 2023-08-10:
I tried, I'm getting this error. I do not enable switch cluster.
[image]

LongND0105, 2023-08-10:
I enabled binding cluster but `PluginApplicationCallbacks.h` have not `MatterBindingPluginServerInitCallback`

[image]

[image]

bzbarsky-apple, 2023-08-10:
Where is that PluginApplicationCallbacks coming from?

After you edited the .zap file, did you rerun codegen?  If so, how _exactly_ did you run codegen?

LongND0105, 2023-08-10:
> Where is that PluginApplicationCallbacks coming from?

I run this command to build: 
`$ gn gen out/debug`
`$ ninja -C out/debug`
`PluginApplicationCallbacks.h` in folder `out/debug/gen/third_party/connectedhomeip/examples/bridge-app/bridge-common/app/`
> After you edited the .zap file, did you rerun codegen? If so, how _exactly_ did you run codegen?

I use the above command to build. I don't know if I have rerun the codegen? Could you tell me what should I do?

bzbarsky-apple, 2023-08-10:
Ah, I see, this is inside `out`.  Then yes, this is the build-time codegen and this part should work.

Did you save the .zap file after editing in the UI?

Check the timestamp on that `PluginApplicationCallbacks.h` file: is it after you edited the .zap file?

LongND0105, 2023-08-11:
I checked, I saved .zap and `PluginApplicationCallbacks.h` file rebuilt

bzbarsky-apple, 2023-08-11:
Ah, I see.  PluginApplicationCallbacks is generated by jinja, not ZAP, and hence based on the .matter file, not the .zap file.

So after you save the .zap you need to `./scripts/tools/zap/generate.py path-to-your-bridge-app.zap`.  That should update the `.matter` file in the same directory, and then you should be able to build and get things updated properly.

This all needs to be _much_ better documented.  :(

LongND0105, 2023-08-11:
Yeah. I tried and sucess. I copied `binding-handler.h` and `binding-handler.cpp` from `examples/all-clusters-app/all-cluster-common` to my project. I wrote binding sucess. But when I toggle light switch, `BoundDeviceChangedHandler` function not called.

```c++
static void BoundDeviceChangedHandler(const EmberBindingTableEntry & binding, chip::OperationalDeviceProxy * peer_device, void * context)
{
    using namespace chip;
    using namespace chip::app;

    ChipLogError(NotSpecified, "\r\nBoundDeviceChangedHandler function called\r\n");

    if (binding.type == EMBER_MULTICAST_BINDING)
    {
        ChipLogError(NotSpecified, "Group binding is not supported now");
        return;
    }

    if (binding.type == EMBER_UNICAST_BINDING && binding.local == 4 &&
        (!binding.clusterId.HasValue() || binding.clusterId.Value() == Clusters::OnOff::Id))
    {
        auto onSuccess = [](const ConcreteCommandPath & commandPath, const StatusIB & status, const auto & dataResponse) {
            ChipLogProgress(NotSpecified, "OnOff command succeeds");
        };
        auto onFailure = [](CHIP_ERROR error) {
            ChipLogError(NotSpecified, "OnOff command failed: %" CHIP_ERROR_FORMAT, error.Format());
        };

        VerifyOrDie(peer_device != nullptr && peer_device->ConnectionReady());
        if (sSwitchOnOffState)
        {
            Clusters::OnOff::Commands::On::Type onCommand;
            Controller::InvokeCommandRequest(peer_device->GetExchangeManager(), peer_device->GetSecureSession().Value(),
                                             binding.remote, onCommand, onSuccess, onFailure);
        }
        else
        {
            Clusters::OnOff::Commands::Off::Type offCommand;
            Controller::InvokeCommandRequest(peer_device->GetExchangeManager(), peer_device->GetSecureSession().Value(),
                                             binding.remote, offCommand, onSuccess, onFailure);
        }
    }
}
```

LongND0105, 2023-08-11:
I bound done. Thanks for your support. You saved my day. Now I will close this issue.

closed 2023-08-11 by LongND0105