project-chip/connectedhomeip #28567: [CERT-TEST-FAILURE] IsUrgent flag is not getting propagated on TH scripts

jrhees-cae, 2023-08-07:
### Feature Area

Area:TimeSync

### Test Case

TC-TIMESYNC-2.10, 2.13, 2.13

### Reproduction steps

Pyton scripts for TC-TIMESYNC 2.10, 2.12, and 2.13 subscribe to events with a path tuple include 'urgent==1'. However, the DUT receives a subscribe request without the urgent flag set.

This causes said tests to fail on DUT's which increase the maxInterval during the subscription negotiation.

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

TH -Fall2023-Beta3

### Platform

python

jrhees-cae, 2023-08-07:
I suspect this affects other Python TC's as well.

raju-apple, 2023-08-07:
@cecille can you take a look when you get a chance please

PR #28793 (2023-08-23): python: Set urgent before early returns
Properly set isUrgent on python generated event subscriptions. 

test: subscribed with isUrgent, can now see it being set

Fixes: #28567
closed 2023-08-23 by mergify[bot]