project-chip/connectedhomeip #28568: Matter:Getting Mdns: Resolve failure (kDNSServiceErr_Timeout),

Nishant0585, 2023-08-07:
Hi
I have onboarded matter accessory using matterSupport and Commissioned it on iPhone using Matter iOS SDK API's
Now Again I want to open pairing window to other admin from my app?
or trying to read attribute 
getting error

Mdns: Resolve failure (kDNSServiceErr_Timeout)
2023-08-07 17:43:50.627045-0500 PlusHome[1410:726857] [Discovery] OperationalSessionSetup[1:0000000000000001]: operational discovery failed: ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
2023-08-07 17:43:50.627181-0500 PlusHome[1410:726857] [Controller] Failed to open pairing window on the device. Status ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
2023-08-07 17:43:50.627664-0500 PlusHome[1410:725521] [Controller] Creating NSError from ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
Error Domain=MTRErrorDomain Code=9 "Transaction timed out." UserInfo={NSLocalizedDescription=Transaction timed out.}

Code Used to Open pairing window 

lastPairDeviceId value is 1, which is used to commission this device with matterSuppot
let basedevice =  MTRBaseDevice(nodeID: lastPairDeviceId as NSNumber, controller: mtrCntrl)
let passcode = MTRSetupPayload.generateRandomSetupPasscode()
 basedevice.openCommissioningWindow(withSetupPasscode: passcode, discriminator: 3480, duration: 600, queue: DispatchQueue.main) { payload, error in
                    if let err = error {
                        print(err)
                    }
                    else{
                        print(payload as Any)
                    }
           }

or 
   basedevice.readAttributes(withEndpointID: nil, clusterID: nil, attributeID: nil, params: nil, queue: DispatchQueue.main) { data, error in
                    if let err = error {
                        print(err)
                    }
                    else{
                        print(data as Any)
                    }
                }

Getting Mdns: Resolve failure (kDNSServiceErr_Timeout)

Node Id Used same which used in commission i.e 1

Nishant0585, 2023-08-07:
@bzbarsky-apple

bzbarsky-apple, 2023-08-08:
@Nishant0585 I asked you to create a feedback assistant ticket and attach logs.  Full logs, not snippets that exclude all the pieces leading up to the failure.

- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

bzbarsky-apple, 2023-08-08:
@Nishant0585 In particular, the log pasted above does not show what operational ID was being looked up.

Nishant0585, 2023-08-09:
> @Nishant0585 In particular, the log pasted above does not show what operational ID was being looked up.

what is operational ID  ?
@bzbarsky-apple 
After commissioning using Matter Support, I should be able to read the attributes of the Accessory ?
 I can see that Accessory  after commissioning listed in Setting -> Matter Accessory  App name come under connected  Services
 
 But I tried to fetch accessory Attribute. 
let basedevice = MTRBaseDevice(nodeID: 1 as NSNumber, controller: MTRController)
basedevice.readAttributes(withEndpointID: nil, clusterID: nil, attributeID: nil, params: nil, queue: DispatchQueue.main) { data, error in
if let err = error {
print(err)
}
else{
print(data as Any)
}
}

Mdns: Resolve failure (kDNSServiceErr_Timeout)

bzbarsky-apple, 2023-08-09:
> what is operational ID ?

That's the thing like  E13D03B5DD7EB17D-0000000000000001 that you saw in your dns-sd browse.

> After commissioning using Matter Support, I should be able to read the attributes of the Accessory ?

Not until you commission it into your fabric.

> Mdns: Resolve failure (kDNSServiceErr_Timeout)

Sure.  Again, that tells me nothing about what you were actually trying to resolve, so I can't tell you why the resolve failed.

Nishant0585, 2023-08-17:
@bzbarsky-apple 
I have commissioned the app using MatterSupport Framework 
In Iphone Setting -> Matter Accessories -> {Accessories name} -> Connected Service name
In my case it always show" Matter Test", how to change this on app name ?
Any idea?

qawitherev, 2023-08-18:
@Nishant0585 i am stuck at implementing matterSupport framework especially [this part](https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem#Set-the-principal-class). How exactly do I do that in XCode 14.3.

Nishant0585, 2023-08-18:
> @Nishant0585 i am stuck at implementing matterSupport framework especially [this part](https://developer.apple.com/documentation/mattersupport/adding-matter-support-to-your-ecosystem#Set-the-principal-class). How exactly do I do that in XCode 14.3.

HI @qawitherev 
First create Matter Extension
add bonjour service
_matter._tcp
_matterc._udp
_matterd._udp

implement all the methods  mention in the link
Ios Will invoke below method, which has onboardingPayload , this setup payload you can use to commission the device into your fabric
func commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID)

qawitherev, 2023-08-18:
> HI @qawitherev First create Matter Extension add bonjour service _matter._tcp _matterc._udp _matterd._udp
>
> implement all the methods mention in the link Ios Will invoke below method, which has [...]

very cool man, now I can get my Matter device into my IP network. You mentioned that I should use the `commissionDevice()` to commission the device onto a fabric. Is the flow the same as [CHIPTool](https://github.com/project-chip/connectedhomeip/tree/master/src/darwin/CHIPTool) where I use `MTRDeviceController.commissionNode(nodeID, commissionParams)` to commission the device (assume all the prerequisite things e.g. startupparams, MTRStorage, etc) has been set up? 

Also, @bzbarsky-apple the whole certificate thing is confusing. Do you have a ref/issue where I can read more on this?

Nishant0585, 2023-08-18:
> very cool man, now I can get my Matter device into my IP network. You mentioned that I should use the `commissionDevice()` to commission the device onto a fabric. Is the flow the same as [...]

Hi @qawitherev 
This method is commissionDevice(in home: MatterAddDeviceRequest.Home?, onboardingPayload: String, commissioningID: UUID) sort of open pairing window method of chip tool

Nishant0585, 2023-08-18:
Hi @qawitherev 
Can you see the Matter Accessories in the App setting ?

qawitherev, 2023-08-21:
> Hi @qawitherev Can you see the Matter Accessories in the App setting , with connected service name?

no i don't see my accessory in the app setting. is it supposed to be doing that? 
i do see in the esp log that i have sigma3 message, and commissioning window opened. what do i use inside the `commissionDevice` inside the handler extension to start commission device?

bzbarsky-apple, 2023-08-28:
> In my case it always show" Matter Test", how to change this on app name ?

@Nishant0585 I have no idea where that UI is getting that string.  Please file a feedback ticket following the instructions at https://github.com/project-chip/connectedhomeip/issues/23566#issuecomment-1312351343 and let me know what the feedback ticket ID is, and I will try to route it to the right people.

bzbarsky-apple, 2023-08-28:
> Also, @bzbarsky-apple the whole certificate thing is confusing. Do you have a ref/issue where I can read more on this?

@qawitherev I don't know what you are trying to do, so not sure which part is confusing....  More step-by-step commissioning documentation is being worked on, but doesn't exist yet.  In the meantime, more details needed to answer the question.

bzbarsky-apple, 2023-08-28:
@Nishant0585 Also, is the original issue here still an issue?  If not, please resolve and open new issues for other problems as needed.

qawitherev, 2023-08-29:
> I don't know what you are trying to do, so not sure which part is confusing....

@bzbarsky-apple 
Exactly lol. I was able to commission my ESP device using MatterSupport and Matter framework. However, some of the code is not mine, it is from [CHIPTool](https://github.com/project-chip/connectedhomeip/tree/master/src/darwin/CHIPTool). 

This is what I understand, correct me if I am wrong: 
1. To make a controller, we need `MTRKeypair` passed to the `MTRStartupParam`.
2. The function of this protocol is to have a private and public key to sign [whatever] stuff that need to be signed (is it NOC?)
3. Then we also need Identity Protection Key (IPK), which I assume is like the secure identifier for controller that exist.
4. In CHIPTool, this (IPK) is generated by the app itself. 
5. Then we use `createController` passing in the `MTRStartupParam` with the `ipk` and `MTRKeypair` to create a controller. 
6. I am guessing, since we don't pass in the `operationalCertificate`, `intermediateCertificate` and `rootCertificate`, the framework will use the default one built inside it.

bzbarsky-apple, 2023-08-29:
> The function of this protocol is to have a private and public key to sign [whatever] stuff that need to be signed (is it NOC?)

Correct.  What's being signed depends on whether you are initializing the controller with a certificate-signing keypair or an operational keypair....

> Then we also need Identity Protection Key (IPK), which I assume is like the secure identifier for controller that exist.

The IPK is the IPK; it's part of the Matter specification.  In the simplest case, it's a fabric-wide 16-byte random constant value.  You do need to make sure you use the same IPK every time with the same fabric.

> I am guessing, since we don't pass in the operationalCertificate, intermediateCertificate and rootCertificate, the framework will use the default one built inside it.

If you are using the "initialize with a root keypair" version of things, as you are, the framework will use that root keypair to self-sign a root certificate and use it to sign an operational certificate.

qawitherev, 2023-08-30:
> In the simplest case, it's a fabric-wide 16-byte random constant value. You do need to make sure you use the same IPK every time with the same fabric.

@bzbarsky-apple 
1 fabric = 1 ipk, got it. 
about the fabric thing, in CHIPTool implementation, when instantiating `MTRDeviceControllerStartupParams`, a hard coded id (1: NSNumber) is passed. I assume, we can assign any non-zero NSNumber to the fabric id. However, in the documentation, it says 

> [Fabric id] is scoped by the root public key

Which in my case would be the public key of the passed `nocSigner`. What "scoped by root public key" means?

bzbarsky-apple, 2023-08-31:
> we can assign any non-zero NSNumber to the fabric id

Correct.

> What exactly "scoped by root public key" means?

The actual fabric identity is the pair (root public key, fabric id).

So you can have two different fabrics with different root public keys but the same "fabric id" value.

Or you can have two different fabrics with the same root public key but different "fabric id" values.

Or you can have two different fabrics which have different root public keys _and_ different "fabric id" values.

iVoGia, 2024-06-13:
HI @Nishant0585 , 
Did you readed attributes of Matter device ? 
I have a issue with matter device, pls take a look at : https://developer.apple.com/forums/thread/756870
Thanks .