project-chip/connectedhomeip #28585: [1.2] Mode Base + derivations - invalid StartUpMode test (TC 3.2)

plauric, 2023-08-08:
### Reproduction steps

Step 10 of Mode Base (and derived cluster) test plan expects the device's current mode to be equal to that of the startup mode attribute, after a device power cycle. This is incorrect if OnMode is not null, because in that case it overrides StartUp Mode

"StartUp Mode - If this attribute is not null, the CurrentMode attribute SHALL be set to the StartUpMode value, when the server is powered up, **except in the case when the OnMode attribute overrides the StartUpMode attribute** "

"6.5.1. OnMode with Power Up

If the On/Off feature is supported and the On/Off cluster attribute [StartUpOnOff] is present, with a value of On (turn on at power up), then the CurrentMode attribute SHALL be set to the OnMode attribute value when the server is supplied with power, except if the OnMode attribute is null."

The TC must include a precondition so that the StartUpMode will not be overridden by the OnMode attribute. 

Note: the same issue appears to affect the ModeSelect TP. 

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

732f797

### Platform

core

### Type

Test Improvement

bzbarsky-apple, 2023-08-09:
@plauric Generally test plan issues should get filed in https://github.com/CHIP-Specifications/chip-test-plans/

plauric, 2023-08-09:
@bzbarsky-apple Indeed. I'll close this one and will track this issue in the TP repo: https://github.com/CHIP-Specifications/chip-test-plans/issues/3296

closed 2023-08-09 by plauric