project-chip/connectedhomeip #28596: [BUG] TE2 TC-ACL-2.6 Didn't Getting Expected Result of  access-control-entry-changed event of Access Control Cluster

VaishaliAvhale, 2023-08-09:
### Reproduction steps

1. Commission a device 
2. Execute TC-ACL-2.6
steps :
1  ./chip-tool accesscontrol  read-event access-control-entry-changed 1 0

2  ./chip-tool accesscontrol write acl '[{ "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, { "privilege": 3, "authMode": 3, "subjects": null, "targets":null}]' 1 0

3  ./chip-tool  accesscontrol  read-event  access-control-entry-changed  1 0

4   ./chip-tool accesscontrol write acl '[{ "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, { "privilege": 3, "authMode": 3, "subjects": [0], "targets":null}]' 1 0

5. /chip-tool accesscontrol read-event access-control-entry-changed 1 0
Verify the AccessControl cluster AccessControlEntryChanged event containing at least 3 new elements and MUST NOT contain an added event for second entry written at step 6.
[ACL-2-6-bug.log](https://github.com/project-chip/connectedhomeip/files/12298860/ACL-2-6-bug.log)

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

5b4f8004662d00bdb111367fec7d3ea978c23372

### Platform

core

### Platform Version(s)

v1.2

bzbarsky-apple, 2023-08-09:
> and MUST NOT contain an added event for second entry written at step 6.

That's exactly what the log attached shows.  What _exactly_ is the "Didn't Getting Expected Result" bit?

Please clearly list:

1. The result you were expecting.
2. The result you got instead.

Because as far as I can tell, the log is showing the expected result.

VaishaliAvhale, 2023-08-10:
Expected Result: Event entries before and after constraint error must be same
But in my case i'm getting 3 more entries after CONSTRAINT ERROR.

closed 2023-08-10 by bzbarsky-apple