project-chip/connectedhomeip #28598: [Build] [chip-tool] Chip Tool compilation fails due to the use of variables in eventNumber

ericzijian1994, 2023-08-09:
### Build issue(s)

This issue came up in TC-SMOKECO and I modified TestEvent.yaml to make it easier to find the issue.

```yaml
config:
    nodeId: 0x12344321
    cluster: "Unit Testing"
    endpoint: 1
    prevEventNumber:
        type: int32s
        defaultValue: 0

tests:
    - label: "Read the event with prevEventNumber set to the event value"
      command: "readEvent"
      event: "TestEvent"
      eventNumber: prevEventNumber
```

I added prevEventNumber to `config` and then used it in `eventNumber` of `readEvent`.

Compile chip-tool: `scripts/examples/gn_build_example.sh examples/chip-tool out/chip-tool`

Return error: 

```sh
FAILED: obj/chip-tool.main.cpp.o 
g++ -MMD -MF obj/chip-tool.main.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -Wconversion -Wno-format-nonliteral -Wno-format-nonliteral -fmacro-prefix-map=../../examples/chip-tool/third_party/connectedhomeip/= -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -DCONFIG_USE_LOCAL_STORAGE=true -DCONFIG_USE_SEPARATE_EVENTLOOP=true -DCONFIG_USE_INTERACTIVE_MODE=true -DCONFIG_ENABLE_YAML_TESTS=true -DCHIP_ADDRESS_RESOLVE_IMPL_INCLUDE_HEADER=\<lib/address_resolve/AddressResolve_DefaultImpl.h\> -DCHIP_HAVE_CONFIG_H=1 -DCHIP_MINMDNS_USE_EPHEMERAL_UNICAST_PORT=1 -DCHIP_MINMDNS_HIGH_VERBOSITY=0 -DCHIP_MINMDNS_DEFAULT_POLICY=1 -I../../examples/chip-tool -I../../examples/chip-tool/third_party/connectedhomeip/examples/common -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common/app-common -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool -I../../examples/chip-tool/third_party/connectedhomeip/src/lib -I../../examples/chip-tool/third_party/connectedhomeip/examples/common/tracing -I../../examples/chip-tool/third_party/connectedhomeip/src/tracing/perfetto/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/perfetto/repo/sdk -I../../examples/chip-tool/third_party/connectedhomeip/src/include -I../../examples/chip-tool/third_party/connectedhomeip/src -Igen/include -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/app-common -I../../examples/chip-tool/third_party/connectedhomeip/examples/chip-tool/include -I../../examples/chip-tool/third_party/connectedhomeip/config/standalone -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlassert/repo/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlio/repo/include -I../../examples/chip-tool/third_party/connectedhomeip/third_party/inipp/repo/inipp -I../../examples/chip-tool/third_party/connectedhomeip/third_party/nlunit-test/repo/src -I../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/controller-clusters -Igen/third_party/connectedhomeip/src/controller/data_model -Igen/third_party/connectedhomeip/src/controller/data_model/zapgen -I../../examples/chip-tool/third_party/connectedhomeip/third_party/jsoncpp/repo/include -c ../../examples/chip-tool/main.cpp -o obj/chip-tool.main.cpp.o
In file included from ../../examples/chip-tool/main.cpp:33:
../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/test/Commands.h: In member function ‘virtual CHIP_ERROR TestEventsSuite::DoTestStep(uint16_t)’:
../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/test/Commands.h:90051:64: error: ‘prevEventNumber’ was not declared in this scope; did you mean ‘mPrevEventNumber’?
90051 |                              chip::Optional<chip::EventNumber>(prevEventNumber));
      |                                                                ^~~~~~~~~~~~~~~
      |                                                                mPrevEventNumber
../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/test/Commands.h:90051:64: note: maximum limit of 1000 namespaces searched for ‘prevEventNumber’
../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/test/Commands.h:90049:31: error: this statement may fall through [-Werror=implicit-fallthrough=]
90049 |             mTestSubStepCount = 0;
      |             ~~~~~~~~~~~~~~~~~~^~~
../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/test/Commands.h:90053:9: note: here
90053 |         case 6: {
      |         ^~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

### Platform

all

adminy, 2023-08-11:
I'm getting this on darwin:
```
In file included from ../../../matter/src/app/DefaultAttributePersistenceProvider.h:19,
                 from ../../../matter/src/app/DefaultAttributePersistenceProvider.cpp:17:
../../../matter/src/app/SafeAttributePersistenceProvider.h: In member function 'CHIP_ERROR chip::app::SafeAttributePersistenceProvider::WriteScalarValue(const chip::app::ConcreteAttributePath&, T&)':
../../../matter/src/app/SafeAttributePersistenceProvider.h:57:12: error: 'if constexpr' only available with '-std=c++17' or '-std=gnu++17' [-Werror=c++17-extensions]
   57 |         if constexpr (std::is_signed_v<T>)
      |            ^~~~~~~~~
../../../matter/src/app/SafeAttributePersistenceProvider.h:57:28: error: 'is_signed_v' is not a member of 'std'; did you mean 'is_signed'?
   57 |         if constexpr (std::is_signed_v<T>)
      |                            ^~~~~~~~~~~
      |                            is_signed
../../../matter/src/app/SafeAttributePersistenceProvider.h:57:41: error: expected primary-expression before '>' token
   57 |         if constexpr (std::is_signed_v<T>)
      |                                         ^
../../../matter/src/app/SafeAttributePersistenceProvider.h:57:42: error: expected primary-expression before ')' token
   57 |         if constexpr (std::is_signed_v<T>)
      |                                          ^
At global scope:
cc1plus: note: unrecognized command-line option '-Wno-unknown-warning-option' may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
[226/367] c++ obj/src/app/libCHIPDataModel.InteractionModelEngine.cpp.o
ninja: build stopped: subcommand failed.
```

ericzijian1994, 2023-08-24:
Since codegen tests were removed, this issue is resolved.

closed 2023-08-24 by ericzijian1994