project-chip/connectedhomeip #28599: [TE2-v1.2] Scenes Cluster's Explicit, TableSize, FabricScenes features implementation is missing

PSONALl, 2023-08-09:
### Reproduction steps / Feature

- Scenes Cluster's Explicit, TableSize, FabricScenes features are mandatory in the specifications
- But it's implementation is missing in the SDK 
 

### Platform

core (please add to version below)

### Platform Version(s)

v1.2

### Type

Manually tested with SDK

PSONALl, 2023-08-10:
> @PSONALl Is this a duplicate of #28217 ?

Closing this issue as it is a duplicate!!

closed 2023-08-10 by PSONALl

dhrishi, 2023-08-10:
@PSONALl Please include the original issue in the results of the test case(s) that failed