project-chip/connectedhomeip #28602: [1.1.0.0] [Android] CHIPDeviceController.cpp:829: CHIP Error 0x00000003: Incorrect state

Allen-Guof, 2023-08-09:
### Reproduction steps

Pair device always failed. It must restart the App. Then the issue gone.

### Bug prevalence

Maybe a day or a few days

### GitHub hash of the SDK that was being used

5a21d17fd2bb0a48f4a356937f4741f6c9dc7975

### Platform

android

### Type

Platform Issue

### Anything else?

08-09 16:39:34.418 31192 10921 D chip.BluetoothManager: null.onMtuChanged: connecting to CHIP device
08-09 16:39:34.418 31192 31192 D ChipDeviceController: Bluetooth connection added with ID: 2
08-09 16:39:34.419 31192 31192 D ChipDeviceController: Pairing device with ID: 39988437302833165
08-09 16:39:34.419 31192 31192 D CTL     : pairDevice() called with device ID, connection object, and pincode
08-09 16:39:34.419 31192 31192 E CTL     : Found unconnected device, removing
08-09 16:39:34.419 31192 31192 D DIS     : Closing all BLE connections
08-09 16:39:34.419 31192 31192 D IN      : SecureSession[0xb400007c8bde2f38]: MarkForEviction Type:1 LSID:54808
08-09 16:39:34.419 31192 31192 D SC      : SecureSession[0xb400007c8bde2f38]: Moving from state 'kActive' --> 'kPendingEviction'
08-09 16:39:34.419 31192 31192 D IN      : SecureSession[0xb400007c8bde2f38]: Released - Type:1 LSID:54808
08-09 16:39:34.419 31192 31192 D IN      : BleConnectionComplete: endPoint 0x7c723f5e88
08-09 16:39:34.419 31192 31192 D DL      : Received GetMTU
08-09 16:39:34.419 31192 31192 D AndroidBleManager: Android Manufacturer: (Xiaomi)
08-09 16:39:34.419 31192 31192 D AndroidBleManager: Android Model: (M2011J18C)
08-09 16:39:34.419 31192 31192 D DL      : Received SendWriteRequest
08-09 16:39:34.420 31192 31192 D IN      : SecureSession[0xb400007c8bde2f38]: Allocated Type:1 LSID:54809
08-09 16:39:34.420 31192 31192 D SC      : Assigned local session key ID 54809
08-09 16:39:34.420 31192 31192 D EM      : <<< [E:27863i S:0 M:134538531] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
08-09 16:39:34.420 31192 31192 D IN      : (U) Sending msg 134538531 to IP address 'BLE'
08-09 16:39:34.420 31192 31192 D IN      : Message appended to BLE send queue
08-09 16:39:34.420 31192 31192 D SC      : Sent PBKDF param request
08-09 16:39:34.420 31192 31192 D CTL     : Setting wifi credentials from parameters
08-09 16:39:34.420 31192 31192 D CTL     : Setting attestation nonce from parameters
08-09 16:39:34.420 31192 31192 D CTL     : Setting CSR nonce from parameters
08-09 16:39:34.420 31192 31192 E CTL     : Commissioning already in progress - not restarting
08-09 16:39:34.420 31192 31192 E CTL     : Failed to pair the device.
08-09 16:39:34.420 31192 31192 W System.err: chip.devicecontroller.ChipDeviceControllerException: src/controller/CHIPDeviceController.cpp:829: CHIP Error 0x00000003: Incorrect state
08-09 16:39:34.420 31192 31192 W System.err: 	at chip.devicecontroller.ChipDeviceController.pairDevice(Native Method)
08-09 16:39:34.420 31192 31192 W System.err: 	at chip.devicecontroller.ChipDeviceController.pairDevice(Unknown Source:84)
08-09 16:39:34.420 31192 31192 W System.err: 	at chip.devicecontroller.ChipDeviceController.pairDevice(Unknown Source:8)
08-09 16:39:34.420 31192 31192 W System.err: 	at com.yeelight.yeelight_fluid.matter.ChipControllerConnectProxy.connectBleDevice(Unknown Source:235)
08-09 16:39:34.420 31192 31192 W System.err: 	at com.yeelight.yeelight_fluid.matter.ChipControllerConnectProxy.access$connectBleDevice(Unknown Source:0)
08-09 16:39:34.420 31192 31192 W System.err: 	at com.yeelight.yeelight_fluid.matter.ChipControllerConnectProxy$connectBleDevice$2.invokeSuspend(Unknown Source:21)
08-09 16:39:34.420 31192 31192 W System.err: 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(Unknown Source:11)
08-09 16:39:34.420 31192 31192 W System.err: 	at kotlinx.coroutines.DispatchedTask.run(Unknown Source:88)
08-09 16:39:34.420 31192 31192 W System.err: 	at android.os.Handler.handleCallback(Handler.java:938)
08-09 16:39:34.420 31192 31192 W System.err: 	at android.os.Handler.dispatchMessage(Handler.java:99)
08-09 16:39:34.420 31192 31192 W System.err: 	at android.os.Looper.loop(Looper.java:233)
08-09 16:39:34.420 31192 31192 W System.err: 	at android.app.ActivityThread.main(ActivityThread.java:8272)
08-09 16:39:34.420 31192 31192 W System.err: 	at java.lang.reflect.Method.invoke(Native Method)
08-09 16:39:34.420 31192 31192 W System.err: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:656)
08-09 16:39:34.420 31192 31192 W System.err: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:967)

xy371661665, 2023-08-23:
same problem!!

maurycyw, 2024-03-27:
If you are using the `ChipClient.kt` from android `chip-tool`, I found that after calling `ChipDeviceController#close` and `ChipDeviceController#shutdownCommissioning` post commissioning, creating a new controller was the only way around this (as well as a similar ble memory error). To accomplish this I replaced the following (within `ChipClient.kt`) although this is not a great solution without first verifying that all (global) refs are cleared. 
```
//        if (!this::chipDeviceController.isInitialized) {
//            chipDeviceController = ChipDeviceController(controllerParams)
//        }

        chipDeviceController = ChipDeviceController(controllerParams)
```